apache/nuttx #11670: xtensa/esp32s3-devkit mcuboot

maxikrie, 2024-02-07:
I am trying to get mcuboot running on the esp32s3-devkit. For this I have modified the wifi config  based on the description in this video https://www.youtube.com/watch?v=Vzy0rl-ixbc. 

The resulting config is attached [config.txt](https://github.com/apache/nuttx/files/14202647/config.txt). However after flashing, the NSH doesn't start but the chip seems to be continuously rebooting  with the following output:

> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdac2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

Can anybody help me with this?

acassis, 2024-02-08:
ping @tmedicci @gustavonihei

rftafas, 2024-02-09:
Which MCUboot are you using? There are two possible flavors, the Espressif Port and NuttX port. Can you also run the diagnostic tool without the MCUboot for sanity reasons?

maxikrie, 2024-02-23:
@tmedicci Thank you for pointing me to this config. I somehow missed it! Also the detailed description on how to compile helped a lot. It is all there in the documentation, but without knowing how it is all related, it is a bit confusing.

I can confirm that this config works. I get the following output:
`
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd35a8,len:0x15e4
load:0x403b0000,len:0x1e40
load:0x403ba000,len:0x28f4
entry 0x403bc89c
[esp32s3] [INF] *** Booting MCUboot build v2.0.0-49-gca02c750 ***
[esp32s3] [INF] [boot] chip revision: v0.2
[esp32s3] [INF] [boot.esp32s3] Boot SPI Speed : 40MHz
[esp32s3] [INF] [boot.esp32s3] SPI Mode       : DIO
[esp32s3] [INF] [boot.esp32s3] SPI Flash Size : 4MB
[esp32s3] [INF] [boot] Enabling RNG early entropy source...
[esp32s3] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32s3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32s3] [INF] Boot source: primary slot
[esp32s3] [INF] Image index: 0, Swap type: none
[esp32s3] [INF] Disabling RNG early entropy source...
[esp32s3] [INF] br_image_off = 0x10000
[esp32s3] [INF] ih_hdr_size = 0x20
[esp32s3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s3] [INF] DRAM segment: start=0x148c0, size=0xa00, vaddr=0x3fc896d0
[esp32s3] [INF] IRAM segment: start=0x1279c, size=0x2124, vaddr=0x40374000
[esp32s3] [INF] start=0x403749e4

NuttShell (NSH) NuttX-12.3.0
nsh> 

`
I then tried to enable the mcuboot functionality in the esp32s3-devkit:wifi config to eventually enable the mcuboot update agent over WiFi as in the video. However the nsh never shows up and it this is the output

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd35a8,len:0x15c4
load:0x403b0000,len:0x1e40
load:0x403ba000,len:0x28f4
entry 0x403bc89c
[esp32s3] [INF] *** Booting MCUboot build v2.0.0-49-gca02c750 ***
[esp32s3] [INF] [boot] chip revision: v0.2
[esp32s3] [INF] [boot.esp32s3] Boot SPI Speed : 40MHz
[esp32s3] [INF] [boot.esp32s3] SPI Mode       : DIO
[esp32s3] [INF] [boot.esp32s3] SPI Flash Size : 4MB
[esp32s3] [INF] [boot] Enabling RNG early entropy source...
[esp32s3] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32s3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32s3] [INF] Boot source: primary slot
[esp32s3] [INF] Image index: 0, Swap type: none
[esp32s3] [INF] Disabling RNG early entropy source...
[esp32s3] [INF] br_image_off = 0x10000
[esp32s3] [INF] ih_hdr_size = 0x20
[esp32s3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s3] [INF] DRAM segment: start=0x2ab08, size=0x1684, vaddr=0x3fc96dd0
[esp32s3] [INF] IRAM segment: start=0x266dc, size=0x442c, vaddr=0x40374000
[esp32s3] [INF] start=0x40374a50
`
I am attaching the config for the latter case.
[config.txt](https://github.com/apache/nuttx/files/14390974/config.txt)

maxikrie, 2024-03-06:
@tmedicci Can you maybe send me working config file for the mcuboot update agent example?

tmedicci, 2024-03-25:
> @tmedicci Can you maybe send me working config file for the mcuboot update agent example?

I'm sorry, I completely missed your message (I receive notifications of all issues and PRs).

I don't have any config for MCUboot's update agent for ESP32-S3, but we do have that for ESP32. Can you please check the configs (suggestion, you can compare `nuttx/boards/xtensa/esp32/esp32-devkitc/configs/mcuboot_nsh/defconfig` with `nuttx/boards/xtensa/esp32/esp32-devkitc/configs/mcuboot_update_agent/defconfig` and apply the same changes for ESP32-S3). It should work flawlessly.

SmartIotMaker, 2024-10-15:
I encountered a different issue, using xtensa/esp32s3-devkit to enable MCUboot: ./tools/configure.sh -S -l esp32s3-devkit:blewifi, then enable MCUboot. Also configure: CONFIG_ESP32S3_SPIFLASH_SPIFFS=y CONFIG_ESP32S3_STORAGE_MTD_OFFSET=0x450000 CONFIG_ESP32S3_STORAGE_MTD_SIZE=0xa00000.

The system hangs after the setup_spiffs function, with no response. Upon debugging, it was found that the SPI flash functions from the ROM, including spi_flash_read and spi_flash_write, cannot be used.

Should I burn a partition table when using MCUboot? Currently, in the test, only the mcuboot-esp32s3.bin compiled with make bootloader has been flashed.

tmedicci, 2024-10-16:
> Should I burn a partition table when using MCUboot? Currently, in the test, only the mcuboot-esp32s3.bin compiled with make bootloader has been flashed.

No. Can you share your system information (GCC, host system) and your defconfig for testing?