apache/nuttx #11672: AES_MODE_ECB support removed with PR #6920

asmatlock, 2024-02-08:
Previous support for AES_MODE_ECB is still not returned since PR #6920 removed it. The arch types that did support it are left waiting and the crypto test all fail.

acassis, 2024-02-08:
Ping @anjiahao1 @xiaoxiang781216

xiaoxiang781216, 2024-02-16:
@ThePassionate please take a look.

ThePassionate, 2024-02-19:
> Previous support for AES_MODE_ECB is still not returned since PR #6920 removed it. The arch types that did support it are left waiting and the crypto test all fail.

AES_MODE_ECB can be regarded as a special AES_MODE_CBC mode with fixed length(16 bytes).
Please calculate according to the above method. If there is still a problem, please provide more information. We will improve the crypto module later.