apache/nuttx #11677: STM32H7/pwm: multi-channel issue

maxikrie, 2024-02-09:
I encountered an issue for the case where multiple channels are supposed to output a PWM signal. For the case where the first channel doesn't start with number 1, there is no output.

The issue is the following code in stm32_pwm.c:3317-3338:
```

      /* A value of zero means to skip this channel */

      if (channel != 0)
        {
          /* Find the channel */

          for (j = 0; j < priv->chan_num; j++)
            {
              if (priv->channels[j].channel == channel)
                {
                  break;
                }
            }

          /* Check range */

          if (j >= priv->chan_num)
            {
              pwmerr("ERROR: No such channel: %u\n", channel);
              ret = -EINVAL;
              goto errout;
            }
```
For the case where e.g. only channel 3 and 4 (priv->chan_num is 2 in this case) are enabled, the 'check range' fails and the function returns. I suggest to remove this check since it solves the problem and I am not sure if it adds any value.

Please comment if you think this causes any other problems.

acassis, 2024-02-15:
@maxikrie there is something strange, because "chan_num" is the number of channels enabled in each STM32_TIMx timer. So maybe you forgot to configure it correctly in the menuconfig (I'm supposing, not saying you missed to enable/disable CONFIG_STM32_TIMx_CHANNELx) or there is some error in the macro (I checked here and it seems fine).
  
@raiden00pl any idea why the "check range" is causing this error?

raiden00pl, 2024-02-15:
The channel checking logic looks good at first glance. We cannot remove it because it checks input from the user. Maybe something wrong with the configuration or `PWM_TIMx_NCHANNELS` macros in `stm32_pwm.h` ? I don't have any stm32 on the bench right now to check it quickly, maybe during the weekend

maxikrie, 2024-02-24:
@raiden00pl Thanks for looking into this! 

My bad, there was a naming confusion on my side. I didn't realize that channel description in `pwm_info_s info` always needs to start from `info.channels[0]`. I originally had timer channels 1-4 enabled and when I later disabled channel 1 and 2. I set
 ```
info.channels[0].channel=0
info.channels[1].channel=0
info.channels[2].channel=3
info.channels[3].channel=4
```
However, in this case the check described above would fail, which is correct, and I mistake on my side.

closed 2024-02-24 by maxikrie