apache/nuttx #11683: LVGL Example Crash

halyssonJr, 2024-02-10:
## Summary
When running an LVGL example, an exception occurred. Validating the operation of the same example in previous commits and using the bisect tool to track the problem.
PR
- https://github.com/apache/nuttx/pull/11147

 Searching, I found a list showing an explanation for this exception:  
- https://links2004.github.io/Arduino/dc/deb/md_esp8266_doc_exception_causes.html

## Test

Board: ESP32 Wrover

```
nsh> uname -a
NuttX 12.4.0 bc43c419f2 Feb 10 2024 11:26:02 xtensa esp32-wrover-kit
nsh>?
Builtin Apps:
    lvgldemo    nsh         sh          
nsh> lvgldemo
[LVGL] [Warn]	(157.800, +157800)	 lv_demo_widgets: LV_FONT_MONTSERRAT_18 is not enabled for the widgets demo. Using LV_FONT_DEFAULT instead. 	(in lv_demo_widgets.c line #138)
[LVGL] [Warn]	(157.800, +0)	 lv_demo_widgets: LV_FONT_MONTSERRAT_12 is not enabled for the widgets demo. Using LV_FONT_DEFAULT instead. 	(in lv_demo_widgets.c line #143)
xtensa_user_panic: User Exception: EXCCAUSE=0003 task: lvgldemo
_assert: Current Version: NuttX  12.4.0 bc43c419f2 Feb 10 2024 11:26:02 xtensa
_assert: Assertion failed user panic: at file: :0 task: lvgldemo process: lvgldemo 0x4010197c
up_dump_register:    PC: 40108f65    PS: 00060e30
up_dump_register:    A0: 8010b908    A1: 3ffe7c70    A2: 00000000    A3: 40001a98
up_dump_register:    A4: 00000000    A5: 3ff64000    A6: 00000280    A7: 3ff6401c
up_dump_register:    A8: 3f41e750    A9: 3ff64080   A10: 00000040   A11: 00000040
up_dump_register:   A12: 3ff64028   A13: 3ff6402c   A14: 00000051   A15: 00000004
up_dump_register:   SAR: 0000000b CAUSE: 00000003 VADDR: 40001a98
up_dump_register:  LBEG: 00000000  LEND: 00000000  LCNT: 00000000
dump_stack: User Stack:
dump_stack:   base: 0x3ffe7688
dump_stack:   size: 00002024
dump_stack:     sp: 0x3ffe7c70
stack_dump: 0x3ffe7c60: 80108431 3ffe7cd0 40108d04 40001a98 95511f6f 109935af 8ea274c0 e0eec522
stack_dump: 0x3ffe7c80: 5b7dc047 07a8cfa1 19a6e48c 5d45a44a 3ffb21c4 8fe0b31e b4164afa 9ab84b4c
stack_dump: 0x3ffe7ca0: 489ceb43 8ea80b42 4d071bec 15b5729a 836d4f82 e5db1e71 ad99b120 edc8c613
stack_dump: 0x3ffe7cc0: 801081ca 3ffe7cf0 4010b8e8 00000000 00000140 0000013f 00000008 3ffe4158
stack_dump: 0x3ffe7ce0: 800db750 3ffe7d20 401083c0 00003ac6 00000140 00000008 3ffb2208 3ffec9e0
stack_dump: 0x3ffe7d00: 00000000 40001a98 00000140 3ffb0ee0 800db8ee 3ffe7d60 3ffafb40 00001101
stack_dump: 0x3ffe7d20: 00000000 c6495685 13d05f97 00000000 00000140 00000000 00000000 4008073a
stack_dump: 0x3ffe7d40: 3ffafca0 3ffaf488 00000008 40001a98 801019a0 3ffe7dc0 00000000 00001101
stack_dump: 0x3ffe7d60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
stack_dump: 0x3ffe7d80: 00000000 00000000 3f0854f2 d5c2ffea 3ffe7df0 3ffe7dd0 00000008 00050020
stack_dump: 0x3ffe7da0: 3ffafca0 3ffaf4b0 0000030b 3ffaf8b0 800dfe6c 3ffe7e10 3ffafc30 3ffafcc0
stack_dump: 0x3ffe7dc0: 3ffe7df0 3ffe7dd0 00000008 3ffafb40 00060020 3ffb00d8 3ffafca0 3ffe41d0
stack_dump: 0x3ffe7de0: 00000000 00000001 3ffafc30 3ffafcc0 3ffafca0 3ffe41d0 00000000 00000001
stack_dump: 0x3ffe7e00: 80107c01 3ffe7e30 4010197c 3ffafc30 3ffafcb4 3ffe41f0 3ffafca0 00000000
stack_dump: 0x3ffe7e20: 00000000 3ffe7e50 3ffafcd0 00000000 3ffb08c0 3ffafbf0 00000020 00000046
stack_dump: 0x3ffe7e40: 00000000 3ffe7e70 00000000 00000000 00000000 00000000 00000000 00000000
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE   COMMAND
dump_task:       0     0   0 FIFO     Kthread N-- Ready              0000000000000000 0x3ffb1200      3056   Idle_Task
dump_task:       1     1 100 RR       Task    --- Waiting Semaphore  0000000000000000 0x3ffaec88      1992   nsh_main
dump_task:       2     2 100 RR       Task    --- Waiting Semaphore  0000000000000000 0x3ffe0490     16336   lvgldemo
dump_task:       3     2 100 RR       pthread --- Running            0000000000000000 0x3ffe7688      2024   lvgldemo 0x4010197c 0x3ffafc30
```

acassis, 2024-02-10:
@terry0012 @xiaoxiang781216 please take a look, probably many other boards are damaged as well.

@tmedicci ESP32-WROVER-KIT LVGL demo is broken, if you can, please test on your side. I did a git bisect to help Halysson to find the root causes and it was caused by that above mentioned commit.

terry0012, 2024-02-18:
The reason for this issue may be that the `stride` in `struct lcddev_area_s` has not been initialized.

@halyssonJr can you help to  add `memset(lcddev_obj, 0, sizeof(struct lcddev_obj_s))` after the code below and try to run `lvgldemo` again ?

https://github.com/apache/nuttx-apps/blob/49204e1389c07083679e66da498c08c669cb42e8/graphics/lvgl/port/lv_port_lcddev.c#L145C1-L149C6

closed 2024-03-23 by halyssonJr