apache/nuttx #11695: Board request: LilyGO T-Deck

mirkokral, 2024-02-14:
The T-Deck is an ESP32-S3 board in a semi-blackberry form factor with a BB keyboard, an (ST7789-compatible) display, power managment, optional LORA, i2s, microphone and no uart-to-usb adapter.

I could not get it into a nuttshell, but it was atleast running.
The only driver needed is probably for the keyboard, and that should be an easy task as the keyboard communicates thru the I2C bus.

Here is a keyboard_get_key arduino function taken from their repo:
```cpp
    Wire.requestFrom(0x55, 1);
    if (Wire.available() > 0) {
        return Wire.read();
    } else {
      return 0;
    }
```
And all the pins seem to be defined [here](https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/examples/LvglArduinoVNC_VGA/TDECK_PINS.h).

acassis, 2024-02-15:
@mirkokral I have a T-Deck here and I could help you to get NuttX running on it. To get access to the NuttShell I suggest you to following this same approach: https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32s3/boards/esp32s3-eye/index.html

mirkokral, 2024-02-16:
> @mirkokral I have a T-Deck here and I can help you to get NuttX running on it. To get access to the NuttShell I suggest you to following this same approach: https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32s3/boards/esp32s3-eye/index.html

I do see the tty, but when using minicom, picocom or screen i do not get anything.

znmeb, 2024-03-20:
I have a T-Deck too ... and an M5Stack Cardputer. Is there a blog post somewhere on how to do this?

acassis, 2024-03-21:
Hi @znmeb I do have a T-Deck and Cardputer here too, but didn't have the chance to test it yet. It is on my list. If you and/or @mirkokral want to start the board port I can help you guys!

coolcoder613eb, 2024-05-02:
I am interested in this.
I don't think I can help with the porting, not knowing C/C++, but I can help test.

mirkokral, 2025-05-04:
After revisiting this, I have managed to get a shell using ADB on the latest git nuttx. I'll try to get the display up and running, but due to my keyboard connector being ripped and trackball being broken because of a fall, I will not write such drivers. Additionaly, it seems like the screen driver is already there (ILI9341/ST7789, both worked), I2S is also already implemented but I didn't research lora for the time being.