apache/nuttx #11751: Build error in sixaxis_main.c

maurosollar, 2024-02-22:
After this steps, I have this build error:

./tools/configure.sh -l nrf52832-dk:nsh
menu configure
Application Configuration
  Examples
    (*) BMI160

make -j

CC:  fixedmath/lib_ubsqrt.c sixaxis_main.c: In function 'sixaxis_main':
sixaxis_main.c:50:26: error: storage size of 'data' isn't known
   50 |   struct accel_gyro_st_s data;
      |                          ^~~~
sixaxis_main.c:65:36: error: invalid application of 'sizeof' to incomplete type 'struct accel_gyro_st_s'
   65 |       ret = read(fd, &data, sizeof(struct accel_gyro_st_s));
      |                                    ^~~~~~
sixaxis_main.c:66:25: error: invalid application of 'sizeof' to incomplete type 'struct accel_gyro_st_s'
   66 |       if (ret != sizeof(struct accel_gyro_st_s))
      |                         ^~~~~~
sixaxis_main.c:50:26: warning: unused variable 'data' [-Wunused-variable]
   50 |   struct accel_gyro_st_s data;
      |                          ^~~~

acassis, 2024-02-22:
@fdcavalcanti maybe you also can help here, because you added support to BMI160 on tm4c1294-launchpad board :-)

maurosollar, 2024-02-22:
Thanks Alan, exactly what it says. It worked! I really like NuttX, but I still know very little about this RTOS, before I used Microchip with Assembly. I'm from Minas Gerais Uai, rs.

Thanks a lot @acassis

closed 2024-02-22 by maurosollar