apache/nuttx #11800: Can't create new folder on the dialog when vscode.window.showOpenDialog is called on WSL environment.

takayoshi-k, 2024-02-28:
Hi,

vscode.window.showOpenDialog method requests to open a file exproler dialog on the host system.
On the WSL environment, dialog is not opened and use command palet area to select a folder because of no system dialog on it.
In some cases, users want to create a new folder when they want to select a folder.
The system file exproler dialog can do it, but it can't in WSL.
Do you have any solution for it, or do you have a plan to support it on VSCode?

acassis, 2024-02-28:
@JorgeGzm @microhobby did you guys face this issue?

JorgeGzm, 2024-02-28:
Hi @takayoshi-k,

Check the microhobby  repository, maybe can solve your problem.

https://github.com/microhobby/vscode-nuttx

```
$ grep -R "showOpenDialog" ./
/extensions/microhobby.linuxkerneldev-0.6.1/out/Utils/ExtensionsUtils.js:            return yield vscode.window.showOpenDialog({
./extensions/microhobby.linuxkerneldev-0.6.1/out/DTSEngine.js:                    vscode.window.showOpenDialog(options).then(uris 
```

takayoshi-k, 2024-03-12:
Oh...
I am sorry...
I thought I make this issue on VSCode github...
But thank you for your reply. I will check it..

takayoshi-k, 2024-09-10:
Close

closed 2024-09-10 by takayoshi-k