apache/nuttx #11803: Debug Assertions asserting on `rv-virt:knsh32`

tmedicci, 2024-02-28:
## Summary

Debug assertions are asserting when enabled for `rv-virt:knsh32` defconfig:

```
ABC
NuttShell (NSH) NuttX-10.4.0
nsh> /system/bin/hello
Hello, World!!
nsh> [  372.381000] _assert: Current Version: NuttX  10.4.0 9f62947d40-dirty Feb 28 2024 09:59:14 risc-v
[  372.381000] _assert: Assertion failed (priv->gat[gatidx] & gatmask) == gatmask: at file: mm_gran/mm_granfree.c:128 task: lpwork process: Kernel 0x80002e4a
[  372.381000] up_dump_register: EPC: 80001c46
[  372.381000] up_dump_register: A0: 80405a30 A1: 00000080 A2: 8001c360 A3: 0000007e
[  372.381000] up_dump_register: A4: 80407a90 A5: 00000001 A6: 00000000 A7: 00000000
[  372.381000] up_dump_register: T0: 00000000 T1: 00000000 T2: 00000000 T3: 00000000
[  372.381000] up_dump_register: T4: 00000000 T5: 00000000 T6: 00000000
[  372.381000] up_dump_register: S0: 00000000 S1: 80407a90 S2: 80406000 S3: 00000000
[  372.381000] up_dump_register: S4: 8001c360 S5: 8001c32c S6: 00042022 S7: 80406000
[  372.381000] up_dump_register: S8: 00000080 S9: 00000000 S10: 00000000 S11: 00000000
[  372.381000] up_dump_register: SP: 80408660 FP: 00000000 TP: 00000000 RA: 80001c46
[  372.381000] dump_stack: User Stack:
[  372.381000] dump_stack:   base: 0x80408040
[  372.381000] dump_stack:   size: 00001984
[  372.381000] dump_stack:     sp: 0x80408660
[  372.381000] stack_dump: 0x80408640: 00000000 00000000 80408040 80405000 00000001 80407a90 80408660 80001df6
[  372.381000] stack_dump: 0x80408660: 80002e4a 00000000 00000000 00000000 00000000 00000000 80407a90 80405a30
[  372.381000] stack_dump: 0x80408680: 8001c360 8001c32c 00000080 7474754e 00000058 00000000 00000000 00000000
[  372.381000] stack_dump: 0x804086a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  372.381000] stack_dump: 0x804086c0: 2e303100 00302e34 00000000 00000000 00000000 66390000 34393236 30346437
[  372.381000] stack_dump: 0x804086e0: 7269642d 46207974 32206265 30322038 30203432 39353a39 0034313a 00000000
[  372.381000] stack_dump: 0x80408700: 00000000 00000000 73697200 00762d63 00000000 00000000 00000000 00000000
[  372.381000] stack_dump: 0x80408720: 00000000 00000000 00000000 00002000 00001000 fffff000 80896000 ffffff83
[  372.381000] stack_dump: 0x80408740: 00001000 80406ed8 000898a1 800070ba 00001000 80406ed8 0a0a1000 80009b9a
[  372.381000] stack_dump: 0x80408760: 00000000 00000000 00000000 00001004 00001000 80408bd8 80895800 8000a9d6
[  372.381000] stack_dump: 0x80408780: 00000000 80405e24 80407a90 00000000 00000000 00000002 80405e24 80405e38
[  372.381000] stack_dump: 0x804087a0: 80405e1c 80405e30 80408bd8 800014a2 80405e1c 80405e30 00042022 80002ea8
[  372.381000] stack_dump: 0x804087c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  372.381000] stack_dump: 0x804087e0: 00000000 00000000 00000000 80004224 00000000 00000000 00000000 00000000
[  372.381000] stack_dump: 0x80408800: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  372.381000] dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE   COMMAND
[  372.381000] dump_tasks:   ----   --- --- -------- ------- --- ------- ---------- ---------------- 0x80405000      2048   irq
[  372.381000] dump_task:       0     0   0 FIFO     Kthread - Ready              0000000000000000 0x80406170      3056   Idle_Task
[  372.381000] dump_task:       1     1 100 RR       Kthread - Running            0000000000000000 0x80408040      1984   lpwork 0x80405e1c 0x80405e30
[  372.381000] dump_task:       2     2 100 RR       Task    - Waiting Semaphore  0000000000000000 0xc0802030      3024   /system/bin/init
QEMU: Terminated
```

## How to reproduce

Followed the instructions in [`knsh32`](https://nuttx.apache.org/docs/latest/platforms/risc-v/qemu-rv/boards/rv-virt/index.html#knsh32) on the master branch of NuttX. Before building the firmware, the `DEBUG_ASSERTIONS` config was enabled:

```
--- a/boards/risc-v/qemu-rv/rv-virt/configs/knsh32/defconfig
+++ b/boards/risc-v/qemu-rv/rv-virt/configs/knsh32/defconfig
@@ -44,6 +44,9 @@ CONFIG_ARCH_USE_S_MODE=y
 CONFIG_BINFMT_ELF_EXECUTABLE=y
 CONFIG_BOARD_LOOPSPERMSEC=6366
 CONFIG_BUILD_KERNEL=y
+CONFIG_DEBUG_ASSERTIONS=y
+CONFIG_DEBUG_ASSERTIONS_EXPRESSION=y
+CONFIG_DEBUG_FEATURES=y
 CONFIG_DEV_ZERO=y
 CONFIG_ELF=y
 CONFIG_EXAMPLES_HELLO=m
@@ -65,6 +68,7 @@ CONFIG_LIBC_PERROR_STDOUT=y
 CONFIG_LIBC_STRERROR=y
 CONFIG_LIBM=y
 CONFIG_MM_PGALLOC=y
+CONFIG_NDEBUG=y
 CONFIG_NFILE_DESCRIPTORS_PER_BLOCK=6
 CONFIG_NSH_ARCHINIT=y
 CONFIG_NSH_FILEIOSIZE=512
```

**NuttX**:  9f62947d40cf44ab826368bd3679a6e22f96e1ad
**NuttX Apps**: 17539f13408fae08a635913b16f865db1e282e7e
**OS**: Manjaro
**Compiler**: gcc version 13.2.0 (xPack GNU RISC-V Embedded GCC x86_64)

acassis, 2024-02-28:
@yf13 I think you already used it, any idea about what is happening?

yf13, 2024-04-13:
This should be the same as #12123, which can be fixed by pull #12132.

closed 2024-04-14 by xiaoxiang781216