apache/nuttx #11819: Incident with Pull Requests and Actions

simbit18, 2024-03-01:
**githubstatus**

https://www.githubstatus.com/

simbit18, 2024-03-01:
Now github is ok. I think you need rebase or update from upstream/master

acassis, 2024-03-01:
Thank you @simbit18, but seems it still failing :'(

@xiaoxiang781216 what do you think? Is it better stop merging the PRs until the CI be fixed again?

xiaoxiang781216, 2024-03-02:
> Thank you @simbit18, but seems it still failing :'(
> 
> @xiaoxiang781216 what do you think? Is it better stop merging the PRs until the CI be fixed again?

Yes, but it's strange that https://github.com/apache/nuttx/pull/11676/checks pass all ci check.

simbit18, 2024-03-02:
With the **old cibuild.sh**  #11826 the problem still remains.

In my test repository instead seems ok [[nuttx-testing-ci](https://github.com/simbit18/nuttx-testing-ci/actions/runs/8121359403)]
with branch [nuttx -> simbit18-test](https://github.com/simbit18/nuttx/commits/simbit18-test/) without https://github.com/apache/nuttx/pull/11202

closed 2024-03-04 by simbit18