apache/nuttx #11870: CMake Support for The JH7110 Board

theoparis, 2024-03-08:
I need to add cmake support because Kconfig refuses to use LLVM/Clang for this board.

I am considering making a PR for this, however I ran into an issue with cmake as well. Removing this line https://github.com/apache/nuttx/blob/master/CMakeLists.txt#L401 let me use a custom clang executable for the toolchain. Without it, CMake tries to find riscv64-unknown-elf-gcc which I don't have. I'd rather not build gcc multiple times for different targets  - that's why I'm using LLVM.

theoparis, 2024-03-08:
I just ran into this https://github.com/apache/nuttx/issues/11861

theoparis, 2024-03-09:
Update: I managed to compile using kconfig with clang after editing Makefile.defs to add the following:
```
CC = clang
CXX = clang++
ARCHCPUFLAGS = --target=riscv64-unknown-none -march=rv64gc_zifencei_zicsr -mabi=lp64d
```
I also added `CONFIG_LIBM=y` and unset the gnu toolchain options in .config.
[image]

It'd still be nice to have cmake support but that'll have to wait for the above issue to be solved.

acassis, 2024-03-11:
@theoparis thank you for your effort trying to add support for CMakefile to compile this board.

As @anchao explained currently CMakefile has support only for building NuttX in FLAT mode (for MCUs without MPU), so PROTECTED mode (for MCUs with MPU) and KERNEL mode (processors with MMU) are not supported it.