apache/nuttx #11875: [Discussion]How to elegantly implement ALLSYMS feat in CMake build

xuxin930, 2024-03-08:
I planning to add ALLSYMS feature in CMake build system, 
there is a issue I encountered and I hope to discuss it with you all.

Since we need to complete the final linking **FIRST** and then use the linked ELF to generate the symbol table.

https://github.com/apache/nuttx/blob/53aef8e93d7613567980febd2ccac0ace1055e7d/arch/sim/src/Makefile#L354-L365
https://github.com/apache/nuttx/blob/53aef8e93d7613567980febd2ccac0ace1055e7d/arch/sim/src/Makefile#L385-L395

---

this means:
1. `allsyms.c` is must generated after nuttx elf linking
2. we have to execute link multiple times, at least **three** times

this will cause problems when using CMake, 
because we cannot add `dependency` to a `target`, and this `dependency` also depends on this `target` at the same time.
```
              need allsyms to gen final elf with syms
nuttx       --------------------------------------->     allsyms.c
         <---------------------------------------
               need nuttx elf to export symtable 

``` 

this creates circular dependency issues.

---

The solution I'm currently thinking of is this:
1. when ALLSYMS feature is enabled , add an empty allsyms.c file dependency for nuttx.
2. add post execution process , run mkallsym and link **twice**

It looks a bit like this
```cmake
if(CONFIG_ALLSYMS)
  file(TOUCH ${CMAKE_BINARY_DIR}/allsyms.c) 
  add_library(allsyms OBJECT)
  nuttx_add_library_internal(allsyms)
  target_compile_definitions(allsyms PRIVATE $<TARGET_PROPERTY:nuttx,NUTTX_KERNEL_DEFINITIONS>)
  target_compile_options(allsyms PRIVATE $<TARGET_PROPERTY:nuttx,NUTTX_KERNEL_COMPILE_OPTIONS>)
  target_sources(allsyms PUBLIC ${CMAKE_BINARY_DIR}/allsyms.c)
  add_dependencies(nuttx allsyms) 

  add_custom_command(
    TARGET nuttx
    POST_BUILD
    COMMAND ${CMAKE_COMMAND} -E remove allsyms.c
    COMMAND ${NUTTX_DIR}/tools/mkallsyms.py ${CMAKE_BINARY_DIR}/nuttx allsyms.c
    COMMAND ${CMAKE_COMMAND} --build ${CMAKE_BINARY_DIR} -t nuttx
    COMMAND ${CMAKE_COMMAND} -E remove allsyms.c
    COMMAND ${NUTTX_DIR}/tools/mkallsyms.py ${CMAKE_BINARY_DIR}/nuttx allsyms.c
    COMMAND ${CMAKE_COMMAND} --build ${CMAKE_BINARY_DIR} -t nuttx
    DEPENDS nuttx 
    WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
    COMMAND_EXPAND_LISTS) 
endif()
``` 

---
What I doubt is whether it is **reasonable** to run `cmake --build `during the cmake call ？
although it will **ONLY** re-execute the action of generating allsyms.o
Is there a better way to solve this problem that I don't know of ?

anchao, 2024-03-11:
How about adding multiple execution targets?

```
add_executable(nuttx)
add_executable(nuttx_allsyms1)
add_executable(nuttx_allsyms2)
```
and update `allsyms.o` by nesting `add_custom_command() POST_BUILD`.

xuxin930, 2024-03-11:
> How about adding multiple execution targets?

Okay let me give it a try

xuxin930, 2024-04-02:
> How about adding multiple execution targets?
> 
> ```
> add_executable(nuttx)
> add_executable(nuttx_allsyms1)
> add_executable(nuttx_allsyms2)
> ```
> 
> and update `allsyms.o` by nesting `add_custom_command() POST_BUILD`.

This plan is perfect, and much better than what I designed before, 👍
thank you very much. I submitted in this PR https://github.com/apache/nuttx/pull/12035

closed 2024-04-03 by acassis