apache/nuttx #11877: STM32: UART write issue

maxikrie, 2024-03-09:
Hello, 

I am trying to write to a UART port on the nucleo-h743zi2 development board (also the stm32f4discovery development board), unfortunately with no success. However on the ESP32S3-devkit it works.

The code is the following, where "/dev/ttyS1" is the UART device (I tried USART2 on TX pin PD5 and UART5 on TX pin PC12).
```
  int fd = open("/dev/ttyS1", O_RDWR | O_NOCTTY | O_NONBLOCK);
  if (fd < 0)
    {
      printf("open failed: %d\n", errno);
      return 1;
    }

  int ret = write(fd, "Hello, World!!\n", 16);
  if (ret < 0)
    {
      printf("write failed: %d\n", errno);
      return 1;
    }
  printf("write returned: %d\n", ret);

  close(fd);
```
The code returns that 16 bytes were written successfully, but I don't receive anything on my PC (connected with a USB-to-serial cable). If I run the same code on a ESP32 (UART1 on TX pin 17) I see the expected output in the console.

I have also tried the other way around, sending data from the console to the boards and receiving data with the serialrx example. This works for either nucleo-h743zi2 and ESP32S3-devkit.

When I check the wires with a logic analyzer, I see that that the TX line is normally 0 on the STM32 while it is normally 1 on the ESP32. When I execute the code, the line goes briefly goes to 1 (but doesn't contain any information). On the ESP32 I see the expected information ("Hello, World!!\n") on the line.

Thanks for any hints!

acassis, 2024-03-09:
@maxikrie I used zmodem running on STM32F4Discovery and it worked, so the TX/RX is working. BTW, are you using DMA? Could be related to same issue they are discussing in the mailing list.

maxikrie, 2024-03-10:
@acassis Thanks for the hint! It lead me to the solution.

The issue in my example was the `O_NONBLOCK` flag. I guess immediate  `close(fd);` causes the bytes not to be written in this case? I was mislead by the fact that `write` would return the correct amount of bytes written. Also, this example was working on the ESP32, which confused me.

acassis, 2024-03-11:
@maxikrie I think these differences between ESP32 and STM32 could happen because the former has internal HW FIFO and the later doesn't.

More info about O_NONBLOCK please read the man pages:
$ man 2 open
$ man 2 fcntl

Also I found this page with useful information about it: https://wiki.sei.cmu.edu/confluence/display/c/FIO32-C.+Do+not+perform+operations+on+devices+that+are+only+appropriate+for+files

maxikrie, 2024-03-11:
@acassis Thanks a lot!

closed 2024-03-11 by maxikrie