apache/nuttx #11907: [Discussion] Rust integration in Nuttx

rushabhvg, 2024-03-13:
As a part of GSOC 2024, I wish to integrate Rust into Nuttx. For that, a new coding standard for Rust programming language will be required and there's no mention of it anywhere. Before writing any rust code, a coding standard would be required. Please give your inputs. I was able to find C coding standards at: [C Coding Standard](https://nuttx.apache.org/docs/latest/contributing/coding_style.html)

rushabhvg, 2024-03-13:
@xiaoxiang781216 @anchao @raiden00pl Please give your suggestions.

xiaoxiang781216, 2024-03-13:
NuttX doesn't support Rust yet, that's why no Rust coding standard exist in official document.
As far as I know, Rust defines an official coding standard, so I would suggest that NuttX follow this style too. @patacongo @acassis what do you thinking?

ppisa, 2024-03-13:
I think that it would worth to discuss Rust on NuttX with  @lupyuen https://lupyuen.github.io/ . I have seen lot of useful information on his site.

rushabhvg, 2024-03-13:
Hello @lupyuen then maybe I can start with Rust Apps for Nuttx? How should we communicate?

lupyuen, 2024-03-13:
@rushabhvg Let's discuss over email: luppy@appkaki.com. I'm in Singapore (GMT+8) so my response may be slow. Thanks :-)

acassis, 2024-03-13:
> NuttX doesn't support Rust yet, that's why no Rust coding standard exist in official document. As far as I know, Rust defines an official coding standard, so I would suggest that NuttX follow this style too. @patacongo @acassis what do you thinking?

@xiaoxiang781216 you mean for Rust applications or Rust code used on kernel right? I think it makes sense because Rust developer could feel uncomfortable reading/write Rust code using NuttX Coding Style

TimJTi, 2024-03-13:
Is this a suggestion that NuttX RTOS itself will move from C to Rust? Or a mix of the two? Sorry if it's a dumb question but I know nothing about rust other than what's on my car...

patacongo, 2024-03-13:
I would think that a mix of the two could be a maintenance problem.

acassis, 2024-03-13:
> Is this a suggestion that NuttX RTOS itself will move from C to Rust? Or a mix of the two? Sorry if it's a dumb question but I know nothing about rust other than what's on my car...

I thought exactly that when I read Xiang comments at first time, but no, he was just saying: let Rust code to use Rust coding style, that makes sense. But as Greg commented we need to figure out how to improve the maintenance to have two coding styles. Maybe just having CI checker for .c and .rs is enough, not sure

TimJTi, 2024-03-13:
> I thought exactly that when I read Xiang comments at first time, but no, he was just saying: let Rust code to use Rust coding style, that makes sense. But as Greg commented we need to figure out how [...]

Where would Rust code be used - approved to be used - though?

patacongo, 2024-03-13:
> Where would Rust code be used - approved to be used - though?

It is somewhat less concerning if restricted to architecture- or board- specific logic as Alan suggested in an dev email.

Any use of use of Rust in the OS must have the support of the NuttX community.  Not just the small group of people with special interests in this issue.

cederom, 2024-03-13:
Hello @rushabhvg and thank you for your interest in NuttX RTOS :-) My suggestions for the workplan in 5 simple steps is presented below :-)

1. Limit Rust use for APPLICATIONS ONLY and NOT THE KERNEL.
2. Work on application layer just like other programming languages do in NuttX (i.e. existing Basic code or new ZigLang apps created by @lupyuen).
3. Work out optional setup that will allow porting and running Rust applications on NuttX. Use most generic and well known Rust tool set for integration like [1].
4. Port some demo applications including peripherals use cases (i.e. UART, SPI, I2C, graphical display, audio, input, network).
5. Create documentation on how to setup local environment for Rust build, how to setup NuttX to work with Rust, how to build a firmware with NuttX and example Rust applications, how to port new Rust applications to NuttX (including KConfig, Makefile, etc).

[1] https://github.com/rust-embedded/awesome-embedded-rust

rushabhvg, 2024-03-14:
Okay @cederom I will follow this worklplan. ANd, work with @lupyuen .

lupyuen, 2024-03-14:
Hi All: @rushabhvg couldn't register on the NuttX Mailing List to read our thoughts about Rust. Any idea how we can troubleshoot this? Thanks :-)

lupyuen, 2024-03-14:
Hi @rushabhvg thanks for chatting over email (and listening to my rant on Rust pros and cons :-) I suggest we begin with this:
1. Let's build and run a simple Rust App on NuttX: https://gist.github.com/lupyuen/7be4bedc6a109b2c3d1201aee6030428
2. If you see this error, let's discuss over email: _"Can't link soft-float modules with double-float modules"_
3. Assuming our Rust App runs OK on NuttX: Let's port this LED Blinky App from C to Rust: https://github.com/lupyuen2/wip-pinephone-nuttx-apps/blob/nim/examples/hello/hello_main.c#L40-L85
4. ULEDIOC_SETALL is explained here: https://lupyuen.github.io/articles/nim#blink-an-led
5. Then we'll do the same for larger Rust Apps, hopefully we'll work out the Rust Coding Guidelines along the way. Thanks :-)

acassis, 2024-03-14:
> I have tried creating my account through [this url](https://lists.apache.org/list.html?dev@nuttx.apache.org) when I click on "Subscribe to list" button, nothing happens. Kindly guide.

You can send a email to dev-subscribe@nuttx.apache.org and then reply the received email

acassis, 2024-03-14:
@lupyuen I think you pointed to hello world in C, the hello world in rust is at hello_rust. But as Dan Gohman ( @sunfishcode ) commented in this Rustix presentation, the integration has some issues, the way it was done we will need a wrapper (extern "C" {} ) for all functions of NuttX. I don't know if he still available to help us, let me ping him and see.

lupyuen, 2024-03-15:
For the record: This PR explains why we're not using Rust Crates, `cargo build` and `Cargo.toml` in NuttX: https://github.com/apache/nuttx/pull/5566

Which presents interesting challenges for Rust Apps in NuttX :-) @rushabhvg Let's document the workarounds that we used (patching of ELF Header, RUSTFLAGS=-O), I think it will be highly useful for other devs

lupyuen, 2024-03-15:
Some folks are wondering: Why not use the Rust Standard Library, instead of the barebones `no_std`? (Rust Core Library) Isn't the Rust Standard Library a lot simpler for doing Console I/O and other POSIX Operations?

1. Traditionally when we build firmware for Embedded Devices, we use no_std: https://docs.rust-embedded.org/book/intro/no-std.html

1. Each Embedded OS will have its own C Function for Console I/O. On NuttX: It just happens that we call the C Standard Library, because NuttX supports POSIX. On FreeRTOS: It would be a totally different C Function.

1. NuttX is a little different from other RTOS because it's compatible with POSIX. In theory, Rust Standard Library could run on NuttX, thanks to POSIX. But it needs work.

1. Rustix Project is working on supporting Rust Standard Library on NuttX. But it seems to have stalled: 

   https://www.reddit.com/r/rust/comments/zfwaqf/integrating_rustix_on_nuttx/

   https://github.com/bytecodealliance/rustix/tree/nuttx

   https://www.youtube.com/watch?v=JTJW6kOqf9I

1. In any case, running Rust Standard Library on Embedded Devices will have impact on Memory Size and Real-Time Performance. Consider Rust on ESP32: It supports Rust Standard Library as well as no_std: https://docs.esp-rs.org/book/overview/using-the-standard-library.html

   ESP32 Rust Standard Library is not recommended if we need Small Memory Footprint, Direct Hardware Control and Real-Time Performance: https://docs.esp-rs.org/book/overview/using-the-core-library.html

1. So for NuttX: We should probably provide Minimal Rust Wrappers around the POSIX Functions that we actually need, instead of supporting the entire Rust Standard Library. If we look at [LED Blinky](https://github.com/lupyuen2/wip-pinephone-nuttx-apps/blob/nim/examples/hello/hello_main.c#L40-L85): We'll need open(), close(), ioctl(), sleep(), ...

   The Rust Wrappers should allow Strings and Pointers to be passed safely to/from NuttX. Maybe the Rustix Project has something we can borrow: https://github.com/bytecodealliance/rustix/tree/nuttx

lupyuen, 2024-03-16:
FYI @rushabhvg Linux Kernel has Coding Guidelines and Templates that might be helpful:
- Coding Guidelines for Rust in Linux Kernel: https://docs.kernel.org/rust/coding-guidelines.html
- Rust Template for Out-Of-Tree Linux Kernel Module: https://github.com/Rust-for-Linux/rust-out-of-tree-module
- Rust Crate for Linux Kernel Module: https://rust-for-linux.github.io/docs/v6.8-rc3/kernel/
  (Since we don't allow Rust Crates in NuttX, we might need a similar preloaded library for NuttX Interfaces)
- The links above are from the "Rust For Linux" website: https://rust-for-linux.com/

lupyuen, 2024-03-17:
FYI: Zephyr OS is also working on supporting Rust Apps and Rust Drivers: https://github.com/zephyrproject-rtos/zephyr/issues/65837

patacongo, 2024-03-17:
@lupyuen I have a question that maybe you can answer.  Everything that I have read says that Rust runs "on top of an OS".  I can't find the exhaustive list of OS interfaces required by Rust.  Such a thing must exist somewhere to support porting to other OSs.

I checked a couple of specific interfaces.  malloc(), for example, seems to be required by Rust.

In KERNEL and PROTECTED build modes, malloc() is not available in the kernel.  kmm_malloc() must be used instead to allocate from the kernel heap.  So I suppose that in these modes you would need wrappers around all of the application OS functions required by Rust to use the internal OS functions:  To fix functionality in some cases.  What about cancellation points and returned errno values?  There is no errno within the OS.

lupyuen, 2024-03-17:
Hi @patacongo :-) There are 2 "flavours" of Rust, depending on the Rust Libraries that we use:
- [Rust Standard Library](https://doc.rust-lang.org/std/): This is used by most Rust Apps on desktops and servers. Supports Heap Memory and the Rust Equivalent of POSIX Calls. 
- [Rust Core Library](https://doc.rust-lang.org/core/index.html) (`no_std`): Barebones Rust Library that runs on Bare Metal, used by Rust Embedded Apps. Calls minimal libc functions, doesn't support Heap Memory and POSIX. 

The malloc() that you mentioned: It's called by the __Rust Standard Library__. [(Like this)](https://github.com/rust-lang/rust/blob/c8813ddd6d2602ae5473752031fd16ba70a6e4a7/library/std/src/sys/pal/unix/alloc.rs#L14)

For Kernel Dev [(like Linux)](https://rust-for-linux.com/third-party-crates#introduction:~:text=Some%20of%20those%20open%2Dsource%20libraries%20are%20potentially%20usable%20in%20the%20kernel%20because%20they%20only%20depend%20on%20core%20and%20alloc%20(rather%20than%20std)%2C%20or%20because%20they%20only%20provide%20macro%20facilities.): We'll use the __Rust Core Library__. Which doesn't support Heap Memory and doesn't need malloc().

But most Kernel Drivers will need Kernel Heap. That's why Linux Kernel also supports the [`alloc` Rust Library / Crate](https://doc.rust-lang.org/alloc/#). To implement Rust `alloc`, Linux Kernel calls krealloc() to allocate Kernel Heap. [(Like this)](https://github.com/torvalds/linux/blob/741e9d668aa50c91e4f681511ce0e408d55dd7ce/rust/kernel/allocator.rs#L46)

For NuttX Kernel: We'll implement Rust `alloc` by calling kmm_malloc().

Since we're calling Rust Core Library in the Kernel, we won't touch any POSIX Application Interfaces. So if we need to support the Kernel Equivalent of Cancellation Points and Errno, we'll have to build the Rust Library ourselves. [(Here's the Rust Library for Linux Kernel)](https://rust-for-linux.github.io/docs/v6.8-rc3/kernel/)

patacongo, 2024-03-17:
> Since we're calling Rust Core Library in the Kernel, we won't touch any POSIX Application Interfaces. So if we need to support the Kernel Equivalent of Cancellation Points and Errno, we'll have to build the Rust Library ourselves

No, I meant the opposite.  The kernel should never initiate a cancellation  point and should never modify the errno.  My concern was that any application interfaces called from a kernel-embedded, run-time library might do just that.

Related:  NuttX does not support shared libraries in these build modes.  This means that there might be multiple copies of whatever necessary run-time support is needed in each user and kernel address space.  The C libraries really have this same problem at present.

acassis, 2024-03-20:
That is an interesting interview with Linus Torvalds about Rust: https://www.youtube.com/watch?v=YyRVOGxRKLg

Although he doesn't program in Rust, he seems to agree it is an important addition to Linux.

nakajimayoshi, 2025-03-17:
Late to the party but using make isn't the right way to go about integrating rust into any RTOS. 

Half the appeal of rust isn't just memory safety, but the tools that come with cargo such as a package manager, clippy, etc. what are the roadblocks to making a fully fledged build system with cargo instead of make?

nakajimayoshi, 2025-03-18:
@acassis Nice to see you 😄 

Usually the way to get around this issue where previous build steps exist is to integrate a build.rs command in the cargo project, that way you can still use make to compile the underlying OS but have the entire build system managed with cargo as the top layer, handling compilation of user-space applications.

https://doc.rust-lang.org/cargo/reference/build-scripts.html

ppisa, 2025-03-18:
@nakajimayoshi I think that NuttX should be considered as a operating system and standard library base. In this case build by make/Cmake etc. is right way. It should be configured and provide runtime environment. It allows even load of ELF binaries at runtime. They can be build out of NuttX build when NuttX export provides standard C library, syscalls base headerfiles and low level RUST adaptation. So you can use all bells and whistlers of the RUST best practices build system exactly same way as on the GNU/Linux. If you do not want to build runtime loadable binary then you include NuttX kernel library (.a files) in the final link phase and you again work exactly same way as on the GNU/Linux when you link static GLibc.

There are some thinks to resolve:

-  for NuttX export on RISC-V and may it be others there is unresolved issue https://github.com/apache/nuttx/issues/6082
- I consider dependence on RUST in Linux kernel and other projects at significant threat to long term open source sustainability until there is full featured RUST compiler based on fully GNU or other copyleft licensed compiler (LLVM has private commercial branches, i.e. Apple which can prevail if companies in control decide stop the community version funding and community version is used (as I know) only for cherry-picking but kept on M1 and other far behind closed one). So R.M.S. 1984 experience leading to GNU project start matters there. There are too many PR proclamations from the RUST proponent, but when I have looked for company which would fund my student which step in to fill this RUST fundamental gap (https://dspace.cvut.cz/bitstream/handle/10467/113390/F3-DP-2024-Dupak-Jakub-thesis.pdf) then there has been found nobody willing to fund the GCC RUST fronted development continuation (the borrow checker integration in this case)... So I suggest to be very careful about dependency on RUST and its build ecosystem for now.

ppisa, 2025-03-18:
As for https://github.com/apache/nuttx/issues/6082 problem, I would like to provide generic solution, but it would require change in all RISC-V BSPs and probably even others  and my resources are limited.

I have a little nasty workaround for our PMSM motor control project on ESP32C3+iCE40

https://gitlab.fel.cvut.cz/otrees/risc-v-esp32/ice-v-pmsm
```
git clone https://gitlab.fel.cvut.cz/otrees/risc-v-esp32/ice-v-pmsm.git
cd ice-v-pmsm
git submodule update --init
export PATH=/opt/xpack/riscv-none-elf-gcc/bin:/opt/espressif/esptool:$PATH
make sw
```
It clones NuttX, configures it, changes its Makefiles to allow export of the right preprocessed ld-scripts a
then uses the OMK https://gitlab.com/pikron/projects/imxrt-devel/-/wikis/omk-manual build system to build control application which includes PXMC motion control library https://pxmc.org/ (PXMC can be built to system-less, GNU/Linux, NuttX, RTEMS thanks to OMK build system). OMK in uLAN builds for all previously stated plus Windows userpspace and Windows, Linux and NuttX drivers including runtime  loadable one even on NuttX.  The OMK started from our company makesystem used in uLAN project https://ulan.sourceforge.net/ and has been used at our university for our CAN https://ortcan.sourceforge.net/ and DDS https://orte.sourceforge.net/ projects.

So there is no obstacle to provide same level of portability for RUST projects, but it should not be intrusive push to all others to bend according RUST demand.

I am willing to help by my experience to RUST to do integration with NuttX right way and my colleagues at https://www.elektroline.cz/ who are much bigger RUST zealots would be happy to use results and contribute back.

lupyuen, 2025-03-18:
Hi @nakajimayoshi: `cargo` is now supported for Rust Apps on NuttX, thanks to the implementation of Rust Standard Library by @no1wudi. I wrote about it here: https://lupyuen.org/articles/rust7.html

As for Rust in NuttX Kernel: We have no plans to support that.

nakajimayoshi, 2025-03-18:
> I think that NuttX should be considered as a operating system and standard library base.

That's great! Don't know how it relates to this conversation though 😄 .

> I consider dependence on RUST in Linux kernel and other projects at significant threat to long term open source sustainability until there is full featured RUST compiler based on fully GNU or other copyleft licensed compiler

cc @lupyuen 
I think there is some misinterpretation of the original comment. Whether or not to include rust in the NuttX kernel is an entirely separate discussion, and not what I care about.

The goal is to have user Rust applications use only cargo for project and .spk builds. I.e. if and ONLY if you're making a user application with rust, then you manage the project with cargo, rather than make. (This can still call make under the hood, but the point is that the developer facing build system is all invoked via cargo). 

I don't really want to get into the whole "is Rust even safe for long term OSS development" because I'm talking exclusively about applications, not kernel code.

ppisa, 2025-03-18:
> > I think that NuttX should be considered as a operating system and standard library base.
> 
> That's great! Don't know how it relates to this conversation though 😄 .

That means that if you export NuttX build for target then it is yet another C library for the RUST build, like GLibc, and you can use all RUST tools and conventions because you no longer dependent on NuttX build system. You only need to transform CFLAGS and Co. from nuttx-export/scripts/Make.defs to the RUST target defintion and specify that nuttx-export/libs/* and optionally some nuttx-export/startup/esp32c3_head.o should be included into target executable linking.

That CFLAGS transformation to achieve ABI compatible RUST code build can be pain (even worse of embedded), because that is not 1:1 between GCC and LLVM (which again emphasizes my previously stated problem with RUST GCC frontend missing). See the FOSDEM talk Understanding ABI Compatibility
between Compilers targeting RISC-V - ABI Extractor by Luis Silva

https://fosdem.org/2025/events/attachments/fosdem-2025-4838-abi-extractor-understanding-abi-compatibility-between-compilers-targeting-risc-v/slides/237904/abi-extra_oLilQNM.pdf

xiaoxiang781216, 2025-03-18:
@no1wudi already port and upstream the NuttX Rust development with Cargo. @nakajimayoshi you just need follow  or https://nuttx.apache.org/docs/latest/guides/rust.htmlhttps://lupyuen.org/articles/rust7.html.

closed 2025-03-23 by xiaoxiang781216