apache/nuttx #11928: How to fix:  undefined reference to `arm_netinitialize' when configuring pppd on nucleo-f103rb:nsh

gcmurray1990, 2024-03-16:
I am trying to enable the pppd example on nuttx version 12.2.2.1 for the nucleo-f103rb:nsh configuration. 
I want to do this so that I can dial a phone call(using SIM7600-G-H, and I figured that pppd would be the best way to do this. That being said, I am using pppd to connect to my SIM.
However, after enabling network support, tcp support, and wireless support, I am getting the same error message:
arm-none-eabi-ld:/nuttxspace/nuttx/staging/libarch.a(arm_initialize.o): in function `up_initialize':
nuttxspace/nuttx/arch/arm/src/common/arm_initialize.c:139: undefined reference to `arm_netinitialize'

In the file listed: it says 
#if defined(CONFIG_NET) && !defined(CONFIG_NETDEV_LATEINIT)
void arm_netinitialize(void);
#else
#  define arm_netinitialize()
#endif

But network support is already enabled. 
I also tried enabling all the different types of sockets, ARP, NET_TUN, and pretty much tried enabling every network setting to no avail.

How can I fix this undefined reference to arm_netinitialize so that I can get pppd working and try to dial a number?

Any guidance is much appreciated and I am happy to provide more info if necessary.

acassis, 2024-03-17:
Hi @gcmurray1990 you need to enable CONFIG_NETDEV_LATEINIT to avoid the symbol be requested. Since PPP means the network will be enabled after the system boot, then calling arm_netinitialize() doesn't make sense.

Note: for PPP support you don't need to enable wireless support, although modem is a kind a "wire - less connection", it is not wireless.

gcmurray1990, 2024-03-19:
Thank you! That fixed it.

closed 2024-03-19 by gcmurray1990