apache/nuttx #11930: Question about RISCV cpu_start

Giri2801, 2024-03-17:
Since RISCV spec allows WFI to be implemented as a NOP, shouldn't asm("WFI"); be in a while loop, as is commonly done?

https://github.com/apache/nuttx/blob/master/arch/risc-v/src/common/riscv_cpustart.c#L78

```
void riscv_cpu_boot(int cpu)
{
  ....
  /* Enable machine software interrupt for IPI to boot */

  up_enable_irq(RISCV_IRQ_SOFT);

  /* Wait interrupt */

  asm("WFI");

#ifdef CONFIG_BUILD_KERNEL
  /* Initialize the per CPU areas */
 ...
}
```

acassis, 2024-03-17:
I don't understand your point. Maybe @lupyuen @xiaoxiang781216 or @tmedicci could have a better understanding, because they are more used to RISC-V than I.

lupyuen, 2024-03-18:
Hi @xiaoxiang781216 and @tmedicci : It might be good to update the WFI code in NuttX. Right now we assume that WFI will loop forever, but it's possible that the behaviour will change for future RISC-V SoCs. We could follow the pattern used by Linux Boot Code, with WFI inside a Loop: [arch/riscv/kernel/head.S](https://github.com/torvalds/linux/blob/f6cef5f8c37f58a3bc95b3754c3ae98e086631ca/arch/riscv/kernel/head.S#L185-L188) 

```text
/* This will loop forever */
.Lloop:
	wfi
	j .Lloop
```

Why WFI inside a Loop? I suppose WFI might be good for reducing Power Consumption, for the SoCs that actually implement WFI as Loop Forever. For other SoCs, it will simply loop forever while consuming extra power. Thanks!

Giri2801, 2024-03-18:
Hi @lupyuen , I'm working in [Shakti Microprocessor](https://shakti.org.in/) team, and our hardware team had recently changed the implementation of WFI to wait for 1024 cycles, and then go through. I had observed that Nuttx was not booting up after that, so just put in this query. Thanks for the quick response!

lupyuen, 2024-03-18:
@Giri2801 Your project sounds cool! Hope I understand you correctly: 
1. At Startup: NuttX calls [qemu_rv_start](https://github.com/apache/nuttx/blob/master/arch/risc-v/src/qemu-rv/qemu_rv_start.c#L98-L159)
2. Which calls [riscv_cpu_boot](https://github.com/apache/nuttx/blob/master/arch/risc-v/src/qemu-rv/qemu_rv_start.c#L147-L153) start each CPU
3. [riscv_cpu_boot](https://github.com/apache/nuttx/blob/master/arch/risc-v/src/common/riscv_cpustart.c#L65-L79) calls WFI to wait for an interrupt (IPI) before it proceeds
4. Your implementation of WFI waits 1024 cycles. (Because the RISC-V Spec says WFI is a Hint, not an actual Implementation)
5. So riscv_cpu_boot is stuck waiting for the interrupt. Is this correct?

Are we suggesting that we should use a [Spin-Wait](https://github.com/torvalds/linux/blob/f6cef5f8c37f58a3bc95b3754c3ae98e086631ca/arch/riscv/kernel/head.S#L323-L352) for the CPU to boot? Thanks!

acassis, 2024-03-18:
Amazing @Giri2801 !!!! Please submit support to Shakti microprocessor to mainline, this way more people in India could test it!

lupyuen, 2024-03-19:
> If we proceed before actually getting interrupt, this_task() returns invalid pointer, and I'm getting an access fault. Hence we need a loop here, because we need to wait till IPI comes to signla all initializations by core 0 is done

Hi @xiaoxiang781216 and @tmedicci : Would you know how we can fix this? Thanks.

Giri2801, 2024-04-03:
Hi,
I had found [this](https://github.com/riscv-software-src/riscv-pk/blob/master/machine/mentry.S#L306-L321) peace of code in riscv-pk. I think this would help us here. The loop should be something like :
```
while(1) {
    asm("wfi");
    # check if interrupt pending 
    # is set for IPI, that is,
    # software interrupt pending
    if (pending) {
         break;
    }
}
```

lupyuen, 2024-04-03:
@Giri2801 Sorry I'm not so great at RISC-V Assembly, wonder if you could implement this in your fork of NuttX Kernel, and verify if it works? Thanks

inochisa, 2024-04-25:
> Hi, I had found [this](https://github.com/riscv-software-src/riscv-pk/blob/master/machine/mentry.S#L306-L321) peace of code in riscv-pk. I think this would help us here. The loop should be something [...]

I think `wfi` loop is necessary, As AIA allow `wfi` to respond interrupt from another privileged mode (sec 5.5), this will break if we use a S-mode `wfi` but have a M-mode interrupt.