apache/nuttx #11932: There are many references to CONFIG_SPI_DMAPRIO in the kernel but it is not defined in none Kconfig or source file

acassis, 2024-03-17:
I think a modification at some eons damaged few SPI drivers, there are references to CONFIG_SPI_DMAPRIO but without any definition of it.

I didn't investigate further, but probably some boards could be non-functional now.

```
$ git grep CONFIG_SPI_DMAPRIO
arch/arm/src/at32/at32_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/at32/at32_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/at32/at32_spi.c:#      error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32/stm32_i2s.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32/stm32_i2s.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32/stm32_i2s.c:#      error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32/stm32_i2s.c:#      error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32/stm32_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32/stm32_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32/stm32_spi.c:#      error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32/stm32_spi.c:#      error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32f0l0g0/stm32_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32f0l0g0/stm32_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32f0l0g0/stm32_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32f7/stm32_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32f7/stm32_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32f7/stm32_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32h7/stm32_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32h7/stm32_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32h7/stm32_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32h7/stm32_spi_slave.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32h7/stm32_spi_slave.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32h7/stm32_spi_slave.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32l4/stm32l4_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32l4/stm32l4_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32l4/stm32l4_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32l5/stm32l5_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32l5/stm32l5_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32l5/stm32l5_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32u5/stm32_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32u5/stm32_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32u5/stm32_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32wb/stm32wb_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32wb/stm32wb_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32wb/stm32wb_spi.c:#    error "Illegal value for CONFIG_SPI_DMAPRIO"
arch/arm/src/stm32wl5/stm32wl5_spi.c:#  if defined(CONFIG_SPI_DMAPRIO)
arch/arm/src/stm32wl5/stm32wl5_spi.c:#    define SPI_DMA_PRIO  CONFIG_SPI_DMAPRIO
arch/arm/src/stm32wl5/stm32wl5_spi.c:#      error "Illegal value
```

Or maybe people started copying it blindly from stm32_spi.c. This is a point of attention. @raiden00pl did you notice it?

raiden00pl, 2024-03-18:
I think this is more common problem. You can find many ifdefs with the `DMAPRIO` phrase that are missing in Kconfig and it's been like that since I started using NuttX. I have no idea what the story is behind these :)

raiden00pl, 2024-03-18:
Another problem  are definitions like this:

```
#ifde CONFIG_XXX
#  define CONFIG_XXX something
#endif
```

where CONFIG_XXX is not defined in Kconfig at all. There are many places like this in the code. Such definitions should be handled by the default statement in Kconfig.

acassis, 2024-03-18:
> Another problem are definitions like this:
> 
> ```
> #ifde CONFIG_XXX
> #  define CONFIG_XXX something
> #endif
> ```
> 
> where CONFIG_XXX is not defined in Kconfig at all. There are many places like this in the code. Such definitions should be handled by the default statement in Kconfig.

Agree!

patacongo, 2024-03-18:
There is only a single DMA priority for SPI.  If I wanted multiple SPI DMAs with different priorities, a single SPI_DMAPRIO value would not support that.

closed 2026-10-19 by acassis