apache/nuttx #11935: MIMXRT1170-EVK board (not EVKB) can't enter nShell on 12.4

JamesZhou2023, 2024-03-18:
The codebase I used is Nuttx 12.4.
HW is MIMXRT1170-EVK board (not EVKB).
The configure command is  "./tools/configure.sh -l imxrt1170-evk:nsh".

But I can't find any content on my serial terminal.

PetervdPerk-NXP, 2024-03-18:
Could you check the dip switches on your EVK for the boot method?

Furthermore could you enable `CONFIG_DEBUG_FEATURES` this should yield more output.

JamesZhou2023, 2024-03-18:
@PetervdPerk-NXP 
Thanks for your quickly reply.

I already do some test.
1. If Do not select "System Type -> Use D-Cache ", then it can enter NSH.
2. If select "Use D-Cache" and "Memory Manager Warning Output", then it can enter NSH.

Reproduce this issue:
1. select "Use D-Cache"
2. Do not select "Memory Manager Warning Output"
then it can not enter NSH.

PetervdPerk-NXP, 2024-03-18:
Thanks for figuring out a working sequence.
Could you share the log output of a booting IMXRT1170-EVK?

JamesZhou2023, 2024-03-18:
Bnx_start: Entry
nx_start: nxsem_initialize
mm_initialize: Heap: name=Umem, start=0x20241948 size=779960
mm_addregion: [Umem] Region 1: base=0x20241aac size=779600
nx_start: g_npidhash == 4
mm_malloc: Allocated 0x20241ac0, size 24
nx_start: g_npidhash == 4 ----1
mm_malloc: Allocated 0x20241ad8, size 120
mm_malloc: Allocated 0x20241b50, size 24
m

JamesZhou2023, 2024-03-18:
I add some log in nx_start.c

  /* Initialize the logic that determine unique process IDs. */
  sinfo("g_npidhash == 4\n");

  g_npidhash = 4;
  while (g_npidhash <= CONFIG_SMP_NCPUS)
    {
      g_npidhash <<= 1;
    }

  g_pidhash = kmm_zalloc(sizeof(*g_pidhash) * g_npidhash);
  DEBUGASSERT(g_pidhash);

  /* IDLE Group Initialization **********************************************/

  for (i = 0; i < CONFIG_SMP_NCPUS; i++)
    {
      int hashndx;

      /* Assign the process ID(s) of ZERO to the idle task(s) */

      hashndx            = PIDHASH(i);
      g_pidhash[hashndx] = &g_idletcb[i].cmn;

      /* Allocate the IDLE group */

      sinfo("g_npidhash == 4 ----1 \n");
      DEBUGVERIFY(group_allocate(&g_idletcb[i], g_idletcb[i].cmn.flags));
      sinfo("g_npidhash == 4 ----2 \n");
      g_idletcb[i].cmn.group->tg_info->argv = &g_idleargv[i][0];
      sinfo("g_npidhash == 4 ----3 \n");
#ifdef CONFIG_SMP
      /* Create a stack for all CPU IDLE threads (except CPU0 which already
       * has a stack).
       */

      if (i > 0)
        {
          DEBUGVERIFY(up_cpu_idlestack(i, &g_idletcb[i].cmn,
                                       CONFIG_IDLETHREAD_STACKSIZE));
        }
#endif

JamesZhou2023, 2024-03-19:
@PetervdPerk-NXP 
More info:
If I modify "Optimization Level" to "Suppress Optimization" or "custom Optimization", it always works fine. 

I am unsure whether this issue is a problem with MIMXRT1170 or a problem with the entire 12.4 system.

PetervdPerk-NXP, 2024-03-19:
Could you check if your problems occur on this commit ?
https://github.com/apache/nuttx/commit/9906163bebdfa121b6266917f7413b961206510d

PetervdPerk-NXP, 2024-03-19:
The https://github.com/apache/nuttx/commit/9906163bebdfa121b6266917f7413b961206510d commit is the first commit supporting 1170EVK.

JamesZhou2023, 2024-03-19:
I checked out the code based on 2023-12-8, and it has the same behaviors as Ver 12.4.

PetervdPerk-NXP, 2024-03-19:
Which toolchain are you using? I'm running the following
```
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
```

JamesZhou2023, 2024-03-20:
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

JamesZhou2023, 2024-03-20:
I update my toolchain to the latest version, but the problem still exist.

arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

JamesZhou2023, 2024-03-20:
@PetervdPerk-NXP 

Can you reproduce this issue in your environment with the original V12.4 code?

In my environment, if I use the original V12.4 release code and use the "imxrt1170-evk:nsh" configuration, this issue can be reproduced 100%.