apache/nuttx #11956: `raspberrypi-pico:nshsram` configuration gets "ERROR: A RAM binary should have an entry point at the beginning: 20000001 (not 200000c1)"

znmeb, 2024-03-20:
I'm getting this error with the `raspberrypi-pico:nshsram` configuration. See attached script and logfiles for the details. It fails the same way with`raspberrypi-pico-w:nshsram`. None of the other Pico or Pico W configurations do this.

[test-error.zip](https://github.com/apache/nuttx/files/14672705/test-error.zip)

znmeb, 2024-03-20:
In case it matters, the host system is Windows Subsystem for Linux Ubuntu 22.04 LTS.

acassis, 2024-03-25:
Same issue confirmed on Linux:
```
Generating: nuttx.uf2
tools/rp2040/elf2uf2 nuttx nuttx.uf2;
ERROR: A RAM binary should have an entry point at the beginning: 20000001 (not 200000c1)

make: *** [tools/Unix.mk:550: nuttx] Error 253
```

@YuuichiNakamura any idea?

znmeb, 2024-03-25:
It fails the same way on a Raspberry Pi 32-bit Debian 11 and 64-bit Ubuntu 22.04.

acassis, 2024-03-29:
@masayuki2009 could you please ping Yuichi Nakamura? He is the author of nshsram board profile.

masayuki2009, 2024-03-30:
@znmeb 
Could you provide the following information?
- NuttX code base (release version or branch and commit id)
- Toolchain version

znmeb, 2024-03-30:
> @znmeb Could you provide the following information?
> 
>     * NuttX code base (release version or branch and commit id)
> 
>     * Toolchain version

Host is Ubuntu 22.04 LTS "jammy" on `amd64` 

NuttX is version 12.4.0, acquired via `git clone --branch nuttx-12.4.0 https://github.com/apache/nuttx.git nuttx`.

gcc-multilib; g++-multilib:
```
g++-11-multilib/jammy-updates,jammy-security,now 11.4.0-1ubuntu1~22.04 amd64 [installed,automatic]
g++-multilib/jammy,now 4:11.2.0-1ubuntu1 amd64 [installed]
gcc-11-multilib/jammy-updates,jammy-security,now 11.4.0-1ubuntu1~22.04 amd64 [installed,automatic]
gcc-multilib/jammy,now 4:11.2.0-1ubuntu1 amd64 [installed]
```
cross-compilers - these are installed by the Raspberry Pi Pico SDK installer:
```
binutils-arm-none-eabi/jammy,now 2.38-3ubuntu1+15build1 amd64 [installed,automatic]
gcc-arm-none-eabi/jammy,now 15:10.3-2021.07-4 amd64 [installed]
libnewlib-arm-none-eabi/jammy,now 3.3.0-1.3 all [installed,automatic]
libstdc++-arm-none-eabi-dev/jammy,now 15:10.3-2021.07-4+17 all [installed,automatic]
libstdc++-arm-none-eabi-newlib/jammy,now 15:10.3-2021.07-4+17 all [installed,automatic]
```

znmeb, 2024-03-30:
I can try this on Debian 12 "bookworm" if you wish - the tools that ship with it are slightly newer than the ones on Ubuntu 22.04 LTS. I can also try it on the beta of Ubuntu 24.04 LTS; I'm doing everything in containers.

masayuki2009, 2024-03-31:
@znmeb 
Please try https://github.com/apache/nuttx/pull/12025

znmeb, 2024-03-31:
> @znmeb Please try #12025

OK - how do I clone the patched source repository in a script? I can run the whole test set if I have that.

closed 2024-03-31 by xiaoxiang781216

masayuki2009, 2024-03-31:
@znmeb 
You'll be able to clone my repository directly.
However, since the PR was merged you can use the latest master.

znmeb, 2024-04-01:
> @znmeb You'll be able to clone my repository directly. However, since the PR was merged you can use the latest master.

Tests are done ... there's still one error. 
[errors.log.zip](https://github.com/apache/nuttx/files/14818787/errors.log.zip)

By the way, if you want to look at the scripts, they're here at the moment:

https://github.com/AlgoCompSynth/CLAMS/tree/forth-itc/RPI-Pico-W-Dev-Setup

masayuki2009, 2024-04-01:
> Tests are done ... there's still one error. [errors.log.zip](https://github.com/apache/nuttx/files/14818787/errors.log.zip)
>
> By the way, if you want to look at the scripts, they're here at the [...]

OK, I need to change raspberrypi-pico-w:nshsram  as well.

```
...
Testing configuration raspberrypi-pico-w:nshsram                                                                                                                                               
...creating /home/znmeb/nuttxspace/raspberrypi-pico-w..nshsram                                                                                                                                 
...
```

masayuki2009, 2024-04-01:
@znmeb 
Please try https://github.com/apache/nuttx/pull/12028

znmeb, 2024-04-01:
OK - in progress. It turns out the clone operation is

```
git clone --branch fix-rp2040-defconfigs https://github.com/masayuki2009/incubator-nuttx.git nuttx
```

I can expand the test suite to look at other rp2040 boards. However, I don't have any of them, so I was only testing Pico and Pico W. I do have some ESP32-C3 and ESP32-S3 hardware I can test on.

znmeb, 2024-04-01:
OK ... both of them work now.