apache/nuttx #12013: NSH: startup script crash

maxikrie, 2024-03-27:
Hello,

I am trying to run a startup script from a SD Card. I have followed the instructions for enabling the startup and everything works in principle however sometimes it crashes.

I created the nsh_romfsimg.sh from an empty rc.sysinit.template and rcS.template contains
```
mount -t vfat /dev/mmcsd0 /mnt/sdcard
sh /mnt/sdcard/startup/startup.txt
```
The startup script contains 
`echo hello`

Sometimes this crashes with the following output
```
_assert: Assertion failed panic: at file: :0 task: sh 0x802f4f1
up_dump_register: R0: 38002f18 R1: 38003334 R2: 00000005  R3: 00000001
up_dump_register: R4: 00000000 R5: 00000000 R6: 00000000  FP: 00000000
up_dump_register: R8: 00000000 SB: 00000000 SL: 00000000 R11: 00000000
up_dump_register: IP: f4240000 SP: 380039e0 LR: 08008485  PC: 08009b66
up_dump_register: xPSR: 21000000 PRIMASK: 00000001 CONTROL: 00000000
up_dump_register: EXC_RETURN: ffffffe9
dump_stack: User Stack:
dump_stack:   base: 0x38003330
dump_stack:   size: 00001992
dump_stack:     sp: 0x380039e0
stack_dump: 0x380039e0: 38003334 38002f18 24001198 01000000 00640000 00000001 38003a14 38003330
stack_dump: 0x38003a00: 00002f18 00000118 38002f18 08006275 24003768 080048a7 380032f8 38003330
stack_dump: 0x38003a20: 00000000 380032f8 00000000 080238af 00000000 38003330 0000c230 08023a79
stack_dump: 0x38003a40: 00000000 00000000 00000000 08023867 00000000 00000000 38002f18 00000000
stack_dump: 0x38003a60: 00000000 080225bd 38003300 ffffffff 00000000 0800d907 ffffffff 38003b00
stack_dump: 0x38003a80: 00000000 080340c9 00000000 00000001 38003300 00000002 ffffffff 38003b00
stack_dump: 0x38003aa0: 00000000 080340db 38003300 00000002 00000000 0802f501 38003300 00000002
stack_dump: 0x38003ac0: 00000000 08021fcf 00000000 38003300 00000002 0802f4f1 00000000 0800a87f
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE      USED   FILLED       CPU   COMMAND
dump_task:       0     0   0 FIFO     Kthread N-- Running            0000000000000000 0x24002e84      2040      1104    54.1%     82.1%   Idle_Task
dump_task:       1     1 224 RR       Kthread --- Waiting Semaphore  0000000000000000 0x380005b8      1992       740    37.1%      0.0%   hpwork 0x240029a0 0x240029b4
dump_task:       2     2 100 RR       Task    --- Waiting Semaphore  0000000000000000 0x38001170      4064      2024    49.8%     12.3%   nsh_main
dump_task:       3     3 100 RR       Task    N-- Waiting Mutex:939537296 0000000000000000 0x38003330      1992      1992   100.0%!     5.4%   sh
```
It seems that there is a stack overflow for the 'sh' command, but I cannot find where the stack is allocated for this.

When I execute the command from the shell after startup (for the case it went through), I haven't yet observed a crash.

Thanks!
Max

acassis, 2024-03-29:
@maxikrie you need to debug the crash looking the function at PC 08009b66

More info:
https://nuttx.apache.org/docs/12.3.0/guides/cortexmhardfaults.html

anjiahao1, 2024-04-25:
use https://nuttx.apache.org/docs/12.3.0/guides/minidumpserver.html can help you to debug.

anjiahao1, 2024-04-25:
@maxikrie `SYSTEM_NSH_STACKSIZE` try change this config