apache/nuttx #12019: riscv/cmake: protected mode issue

yf13, 2024-03-28:
Here are issues encountered with PROTECTED mode risc-v device:

```
$ rm -rf * && cmake ~/Projects/Nuttx/nuttx -DBOARD_CONFIG=canmv230/pnsh -G Ninja 
-- Initializing NuttX
--   Board:  canmv230
--   Config: pnsh
--   Appdir: /home/yf/Projects/Nuttx/nuttx/../apps
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/riscv64-unknown-elf-gcc
riscv64-unknown-elf-gcc: error:  -Wstrict-prototypes -fno-common -Wall -Wshadow -Wundef -Wno-attributes -Wno-unknown-pragmas: No such file or directory
-- Configuring done
CMake Error at cmake/nuttx_add_library.cmake:97 (add_library):
  Cannot find source file:

    /home/yf/Projects/Nuttx/nuttx/libs/libc/spawn/lib_task_spawn.c
Call Stack (most recent call first):
  cmake/nuttx_add_library.cmake:129 (nuttx_add_system_library)
  libs/libc/CMakeLists.txt:20 (nuttx_add_kernel_library)

CMake Error at CMakeLists.txt:422 (add_executable):
  No SOURCES given to target: nuttx_user

CMake Error at cmake/nuttx_add_library.cmake:135 (add_library):
  No SOURCES given to target: kc
Call Stack (most recent call first):
  libs/libc/CMakeLists.txt:20 (nuttx_add_kernel_library)

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

xiaoxiang781216, 2024-03-28:
@xuxin930 could you take a look?

xuxin930, 2024-03-29:
hi @yf13 
try this patch https://github.com/apache/nuttx/pull/12022

to use protected mode, you need to add the userspace source file of the kernel directory.

acassis, 2024-03-29:
@yf13 if it worked, please submit a documentation similar to this one: https://nuttx.apache.org/docs/latest/platforms/arm/stm32f4/index.html#protected-mode-build