apache/nuttx #12050: build error with kernel mode

gneworld, 2024-04-04:
Have a build error after enabling CONFIG_BUILD_KERNEL and CONFIG_ARCH_VMA_MAPPING in sabre-6quad:knsh, as shown below

`nuttx$ make -j12
Create version.h
LN: platform/board to /home/official/apps/platform/dummy
Register: hello
Register: init
Register: sh
Register: getprime
CC:  clock/clock_settime.c In file included from /home/official/nuttx/include/nuttx/addrenv.h:39,
                 from addrenv/addrenv.c:30:
/home/official/nuttx/include/arch/arch.h:141:18: error: 'ARCH_SHM_MAXPAGES' undeclared here (not in a function)
  141 |   uintptr_t *shm[ARCH_SHM_NSECTS];
      |                  ^~~~~~~~~~~~~~~
Makefile:61: recipe for target 'addrenv.o' failed
make[1]: *** [addrenv.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /home/official/nuttx/include/nuttx/arch.h:82,
                 from clock/clock_initialize.c:37:`
  
  By moving the related code in arch.h into addrenv.h, this issue can be fixed; however, is there a better solution for this issue?

acassis, 2024-04-04:
@masayuki2009 since you work a lot with sabre-6quad board, could you please take a look?

gneworld, 2024-04-07:
@masayuki2009  could you please take a look at this compiling errors in kernel mode？

gneworld, 2024-04-07:
just change   ARCH_PG2SECT(ARCH_SHM_MAXPAGES)
to  ARCH_PG2SECT((CONFIG_ARCH_SHM_NPAGES * CONFIG_ARCH_SHM_MAXREGIONS))
could fix this issue

closed 2024-04-07 by gneworld

acassis, 2024-04-07:
@gneworld should it be modified into mainline?