apache/nuttx #12086: when i start nuttx (sim:nsh), it throw a signal which is SIGSEGV

shikoumuzi, 2024-04-07:
i want to use cpp stl in nuttx program, so i set 
```
140 CONFIG_HAVE_CXX=y
141 CONFIG_HAVE_CXXINITIALIZE=y
142 CONFIG_UCLIBCXX=y
143 CONFIG_LIBSUPCXX=y
144 CONFIG_CXX_STANDARD="gnu++17"
145 CONFIG_CXX_EXCEPTION=y
146 CONFIG_CXX_RTTI=y
```
in meanuconfig

when i start nuttx in gdb, it get me this message

[image]

how should i do to solve it, thank you.

acassis, 2024-04-08:
@masayuki2009 @xiaoxiang781216 any idea?

xiaoxiang781216, 2024-04-08:
please try the stock sim:libcxxtest first.

shikoumuzi, 2024-04-08:
> please try the stock sim:libcxxtest first.

i did that,  but when i use 'make distclean' it still going to have this problem. 
However, a while ago, I compared the configuration of cxxtest and nsh and found that need to turn on an optimization level of at least -O1 in menuconfig, and then it  work properly.
thank you for your reply

shikoumuzi, 2024-04-08:
> please try the stock sim:libcxxtest first.

i did that, but when i use 'make distclean' it still going to have this problem.
However, a while ago, I compared the configuration of cxxtest and nsh and found that need to turn on an optimization level  at least -O1 in menuconfig, and then it work properly.
thank you for your reply