apache/nuttx #12111: esp32: Compile error regarding assert statement in esp-hal-3rdparty code

dumitrache-adrian92, 2024-04-09:
Tested on two devkits, one with an esp32-WROOM and one with an esp32-WROVER.

Setting up NuttX like this:
```bash
./tools/configure.sh -l esp32-{devkitc, sparrow-kit}:nsh
make -j16 EXTRAFLAGS="-DESP32_IGNORE_CHIP_REVISION_CHECK"
```

will lead to the following error:
```bash
Espressif HAL for 3rd Party Platforms: 966f1bbf7d80d3fac9a43f414e864985cbcfd680
CC:  chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_table.c In file included from chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_fields.c:16:
<path>/nuttxspace/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/soc/include/soc/chip_revision.h:36:66: error: expected ',' before ')' token
 _Static_assert(CONFIG_ESP_REV_MIN_FULL <= CONFIG_ESP_REV_MAX_FULL);
                                                                  ^
CC:  chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_utility.c make[1]: *** [Makefile:144: chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_fields.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [tools/LibTargets.mk:164: arch/xtensa/src/libarch.a] Error 2
make: *** Waiting for unfinished jobs....
CC:  chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_fields.c In file included from chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_fields.c:16:
<path>/nuttxspace/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/soc/include/soc/chip_revision.h:36:66: error: expected ',' before ')' token
 _Static_assert(CONFIG_ESP_REV_MIN_FULL <= CONFIG_ESP_REV_MAX_FULL);
                                                                  ^
make[1]: *** [Makefile:144: chip/esp-hal-3rdparty/components/efuse/esp32/esp_efuse_fields.o] Error 1
make: *** [tools/LibTargets.mk:164: arch/xtensa/src/libarch.a] Error 2
```

Basically that assert doesn't have the error argument.

I made a simple fix on a personal repo, and it seems to work (I simply added the necessary argument):

```bash
make -j16 EXTRAFLAGS="-DESP32_IGNORE_CHIP_REVISION_CHECK" ESP_HAL_3RDPARTY_URL="https://github.com/dumitrache-adrian92/esp-hal-3rdparty.git" ESP_HAL_3RDPARTY_VERSION="9e8144ccf86281c8897c9507e4eeda8e1e904367"
```

I'm not sure how to contribute to the esp-hal-3rdparty repo though (since it doesn't seem to be pull request based). I can take care of this if anyone explains the process to me since I couldn't find any info.

acassis, 2024-04-09:
@dumitrache-adrian92 I think esp-hal-3rdparty are maintained directly by Espressif

@tmedicci please take a look

tmedicci, 2024-04-09:
Hi @dumitrache-adrian92 , 

That's strange... We continuously test the boards with `ESP32_IGNORE_CHIP_REVISION_CHECK` set on our internal CI. I double-checked locally and found no trouble (I used the exact commands you posted).

Could you please double-check the compiler being used? The easiest way to do that is by running the `nxdiag` application (https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32/boards/esp32-devkitc/index.html#nxdiag)

dumitrache-adrian92, 2024-04-10:
Sure:
```
nsh> nxdiag --all
Nxdiag Report:

NuttX RTOS info:
        Hostname: 
        Release: 12.4.0
        Build: 7f1e80f0de-dirty Apr 10 2024 08:48:44
        Arch: xtensa
        Defconfig: esp32-devkitc:nsh-dirty

Host system OS:
        Ubuntu 23.10 Linux cave 6.5.0-27-generic #28-Ubuntu SMP PREEMPT_DYNAMIC Thu Mar  7 18:21:00 UTC 2024 x86_64 x86_64
```

pkarashchenko, 2024-04-10:
I just tried to compile this on MAC with the latest master and it built successfully.

tmedicci, 2024-04-10:
Can you please run this defconfig directly:
```
./tools/configure.sh -l esp32-devkitc:nxdiag
```

or, at least, enable the following configs in your setup:
```
CONFIG_SYSTEM_NXDIAG_COMP_FLAGS=y
CONFIG_SYSTEM_NXDIAG_CONF=y
CONFIG_SYSTEM_NXDIAG_ESPRESSIF=y
CONFIG_SYSTEM_NXDIAG_HOST_PATH=y
```

Particularly, I'm interest in such `nxdiag` output (which is missing in your last message):

```
Espressif specific information:

Bootloader version:
        esp32: v5.1-dev-3972-g1559b6309f
        esp32s2: Bootloader image not found
        esp32s3: Bootloader image not found
        esp32c2: Bootloader image not found
        esp32c3: Bootloader image not found
        esp32c6: v5.1-dev-3972-g1559b6309f
        esp32h2: Bootloader image not found

Toolchain version:
        clang: clang version 16.0.6
        gcc: gcc (GCC) 13.2.1 20230801
        xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
        xtensa-esp32s2-elf-gcc: xtensa-esp32s2-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
        xtensa-esp32s3-elf-gcc: xtensa-esp32s3-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
        riscv32-esp-elf-gcc: riscv32-esp-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
        riscv64-unknown-elf-gcc: riscv64-unknown-elf-gcc (SiFive GCC-Metal 10.2.0-2020.12.8) 10.2.0

Esptool version: 4.7.0

HAL version: sync/release_v5.1.c-nuttx-20230814-50-g966f1bbf7d
```

dumitrache-adrian92, 2024-04-10:
@tmedicci 
```
Nxdiag Report:

Espressif specific information:

Bootloader version:
        esp32: v5.1-dev-3972-g1559b6309f
        esp32s2: Bootloader image not found
        esp32s3: Bootloader image not found
        esp32c2: Bootloader image not found
        esp32c3: Bootloader image not found
        esp32c6: Bootloader image not found
        esp32h2: Bootloader image not found

Toolchain version:
        clang: Not found
        gcc: gcc (Ubuntu 13.2.0-4ubuntu3) 13.2.0
        xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
        xtensa-esp32s2-elf-gcc: Not found
        xtensa-esp32s3-elf-gcc: Not found
        riscv32-esp-elf-gcc: Not found
        riscv64-unknown-elf-gcc: Not found

Esptool version: 4.7.0

HAL version: sync/release_v5.1.c-nuttx-20230814-55-g9e8144ccf8
```
Note the HAL version is different because I had to use my own repo to compile.

dumitrache-adrian92, 2024-04-20:
Hello, that was indeed the problem. Thank you for the help! Not sure how I ended up getting such an old version of the compiler.

closed 2024-04-20 by dumitrache-adrian92