apache/nuttx #12122: Can't build custom board

fdcavalcanti, 2024-04-10:
Hi everyone,

I'm trying to create a custom board that is based on a Nucleo-144. I want to have this board versioned so I simply copied the directory `boads/arm/stm32f7/nucleo-144` to my repository.
On the configs directory, I removed all but one of the configs. On defconfig, I added those lines below CONFIG_ARCH:
```
CONFIG_ARCH_BOARD_CUSTOM=y
CONFIG_ARCH_BOARD_CUSTOM_DIR_RELPATH=y
CONFIG_ARCH_BOARD_CUSTOM_DIR="../../test-nucleo-144"
CONFIG_ARCH_BOARD_CUSTOM_NAME="test-nucleo-144"
```
And commented:
```
# CONFIG_ARCH_BOARD="nucleo-144"
# CONFIG_ARCH_BOARD_NUCLEO_144=y
```
Other than that, I removed everything from Kconfig.

I can open menuconfig just fine after executing the configure script with:
`./tools/configure.sh -a ../nuttx-apps -l ../../test-nucleo-144/configs/nsh`

Running make I get this error:

```
$ make -j
Create version.h
LN: platform/board to /home/filipe/rad-stm-fw-poc/nuttxspace/nuttx-apps/platform/dummy
Register: nsh
Register: sh
CC:  dirent/lib_opendir.c chip/stm32_allocateheap.c:168:6: warning: #warning "DTCM excluded from the heap" [-Wcpp]
  168 | #    warning "DTCM excluded from the heap"
      |      ^~~~~~~
CPP:  /home/filipe/rad-stm-fw-poc/nuttxspace/nuttx/../../hsro-nucleo-144/scripts/f746-flash.ld-> /home/filipe/rad-stm-fw-poc/nuttxspace/nuttx/../../hsro-nucleo-144/scripts/f746-flash.ld.tmp make[2]: *** No rule to make target 'libboard.a'.  Stop.
make[1]: *** [Makefile:181: board/libboard.a] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [tools/Unix.mk:546: nuttx] Error 2
```

This issue is very similar to #2206 but I can't seem to make it work.
Any thoughts? I would love to fix this and then improve the custom board documentation.

Thanks.

acassis, 2024-04-10:
@fdcavalcanti I just saw that our documentation for custom board is very shallow: https://nuttx.apache.org/docs/latest/guides/customboards.html# we need something like Custom Apps that @TimJTi wrote.

The error means that some Makefile rules or path to create libboard.a is missing.

BTW, if this board is a commercial board (i.e. from ST) then it is better to add it as official board into NuttX mainline.

fdcavalcanti, 2024-04-10:
Hi @acassis. Its not a commercial board. I'm using a Nucleo-144 integrating in a different equipment as a proof of concept. Its just easier for me to version only the board files and use the custom board settings.

TimJTi, 2024-04-10:
@fdcavalcanti - have you tried the instructions to move it completely out-of-tree? It works for me.

@acassis  - I "ported" the custom boards stuff from the old Wiki (Confluence?) to the current documentation tree and added a bit to it too if I recall. I will revisit it tomorrow and see if it still holds true or needs enhancement - but it is only for out-of-tree boards. I think there's different documentation somewhere or other for in-tree new boards - I'll look, and see if it needs any work that I can see.

fdcavalcanti, 2024-04-10:
Hello @TimJTi thanks for the input.
Yes I tried. I also applied the modifications in the issue description above.
This is really an issue now because I think porting to a similar board, out-of-tree, should been simple. I need to do this PoC and am stuck in what seems to be a basic issue.

I'll keep trying to figure it out and hopefully I can contribute on fixing and documenting.

Thanks

fdcavalcanti, 2024-04-10:
Right, so I got it to compile by reading some issues from `make export` in #10388 
Just had to add `CONFIG_ARCH_BOARD_COMMON=y`.
I understand that there is a common directory inside `boards/stm32f7` but since no defconfig had it, I didn't even think about it before.
I'll keep working on it, try and track down other issues and I'll update the docs.

Thanks everyone.

closed 2024-10-16 by fdcavalcanti

shtirlic, 2025-04-12:
Sorry for resurrecting this thread, `CONFIG_ARCH_BOARD_COMMON=y` works fine with my out of tree board, the problem is that after doing `make savedconfig` it vanishes from `defconfig` and I should always manually add it to resulting defconfig in my out of tree board config. Is it  intended behavior or bug?

fdcavalcanti, 2025-04-14:
I'm not sure what would be the expected output with out of tree builds, could be a bug. Could you show the steps you take to build until it fails?
This way we can use a dummy out of tree defconfig and test.

shtirlic, 2025-04-15:
I am using https://github.com/shtirlic/picocalc-nx as out of tree boards, as usual I do `make distclean` then `./tools/configure.sh ../picocalc-nx/configs/nsh` . In this repo defcofing contains  `CONFIG_ARCH_BOARD_COMMON=y` and everything works fine and builds, if I change something via `make menuconfig` and do `make savedefconfig` after the resulting `defconfig` does not contain `CONFIG_ARCH_BOARD_COMMON=y` but `.config` contains it, so  `make savedefconfig` don's export/save the `CONFIG_ARCH_BOARD_COMMON=y` value

fdcavalcanti, 2025-04-15:
Go to `nuttx/tools/Unix.mk`and modify the `savedefconfig` instruction (around line 748).

Add this line:
`$(Q) grep "CONFIG_ARCH_BOARD_COMMON=" .config >> defconfig.tmp; true`

Let me know if it works, but should be something along those lines.

shtirlic, 2025-04-20:
@fdcavalcanti Thank you, it works, should I submit PR for this?

acassis, 2025-04-20:
@shtirlic yes, I think it is a bug that nobody noticed yet. Even in native boards, if the use selected the CONFIG_ARCH_BOARD_COMMON it is not included in the defconfig. So some drivers that only exist in common board will not work. Please submit the PR

shtirlic, 2025-04-20:
@acassis submited the https://github.com/apache/nuttx/pull/16241

cederom, 2025-08-04:
I just got into this problem with a quick project. Out-of-tree board and apps seems default in most cases. We should improve documentation and examples here, maybe even some example templates :-)