apache/nuttx #12127: When i use cmake to build sim:libcxxtest, it display  undefined reference to `NXpthread_setspecific'

shikoumuzi, 2024-04-11:
when i use cmake -B build -DBOARD_CONFIG=sim:libcxxtest
it can not compile
[image]
but when i use make to build sim libcxxtest, it is running properly
what can i do to solve this problem， thanks

xiaoxiang781216, 2024-04-12:
@xuxin930 could you check why cmake is different from make?

xuxin930, 2024-04-12:
hi @raiden00pl @xiaoxiang781216 
my guess is that the redefined symbols for sim in cmake do not have an `if block`.
let me fix it

https://github.com/apache/nuttx/blob/62a47994090860b8daea0ab61121bcbcd4448ff1/arch/sim/src/nuttx-names.in#L99-L112
https://github.com/apache/nuttx/blob/62a47994090860b8daea0ab61121bcbcd4448ff1/cmake/nuttx_redefine_symbols.cmake#L78-L94

raiden00pl, 2024-04-12:
@xuxin930 good find. 

Another problem I have with this configuration (and with libcxx in general) is a lot of warnings during compilation.
It looks like compiler flags from here doesn't work: https://github.com/apache/nuttx/blob/62a47994090860b8daea0ab61121bcbcd4448ff1/libs/libxx/libcxx.cmake#L99-L124 
I haven't had a chance to look for the cause of this problem yet. I'm not sure if this problem only occurs for me on arch linux

xuxin930, 2024-04-12:
> I haven't had a chance to look for the cause of this problem yet. I'm not sure if this problem only occurs for me on arch linux

Okay, let me check the toolchain file settings and compile otpions settings again.

xuxin930, 2024-04-15:
> I haven't had a chance to look for the cause of this problem yet. I'm not sure if this problem only occurs for me on arch linux

hi @raiden00pl 
there are two errors:
1. the toolchain file is not synchronized and is lagging behind. I will update it first.
2. CMake `set_source_files_properties` does not perform append, but replaces. 😔

regarding the second one, I have implemented `nuttx_set_source_files_properties` instead of `set_source_files_properties` and will submit a batch of bugfixes tomorrow.

raiden00pl, 2024-04-15:
@xuxin930 great ! thanks for solving the issues

closed 2024-06-29 by acassis