apache/nuttx #12133: Inconsistency in function headers style

raiden00pl, 2024-04-13:
Our coding style guide shows function header like this:
```

/****************************************************************************
 * Name: <Global function name>
 *
 * Description:
 *   Description of the operation of the function.
 *
 * Input Parameters:
 *   A list of input parameters, one-per-line, appears here along with a
 *   description of each input parameter.
 *
 * Returned Value:
 *   Description of the value returned by this function (if any),
 *   including an enumeration of all possible error values.
 *
 * Assumptions/Limitations:
 *   Anything else that one might need to know to use this function.
 *
 ****************************************************************************/
 
```
The end of the header is preceded by a new line:
```
 *
 ****************************************************************************/
```

but there are many places in the code where this new line is not added, e.g.:

```
/****************************************************************************
 * Name: sensor_custom_unregister
 *
 * Description:
 *   This function unregister character node and release all resource about
 *   upper half driver. This API corresponds to the sensor_custom_register.
 *
 * Input Parameters:
 *   dev   - A pointer to an instance of lower half sensor driver. This
 *           instance is bound to the sensor driver and must persists as long
 *           as the driver persists.
 *   path  - The user specifies path of device, ex: /dev/uorb/xxx
 ****************************************************************************/
```

And from the doc https://nuttx.apache.org/docs/latest/contributing/coding_style.html#function-headers:
> Function header sections. Within the function header, the following data sections must be provided:
> ...
> Each of these data sections is separated by a single line like *.

Should we correct these places to match the coding standard and pay more attention when reviewing code ?

I know it's a trivial problem, but for some reason it really irritates me when I don't see this additional empty line :)

xiaoxiang781216, 2024-04-13:
Yes, I always point out the minor point during review. Maybe it's better to add the check into nxstyle.

acassis, 2024-04-14:
Agree! Better to include it in the checkpatch!

yf13, 2024-04-15:
Is it possible to put the rule in `tools/uncrustify.cfg`? thus we can use uncrusity tool to fix the formating issue.