apache/nuttx #12180: tools/nuttx-gdbinit: No such file or directory

liukangcc, 2024-04-19:
I flowed this [qemugdb](https://nuttx.apache.org/docs/latest/guides/qemugdb.html) document to debug the nuttx binary.

When I inputd the below command, there was a error message in the gdb window.
`gdb-multiarch -ix tools/nuttx-gdbinit nuttx`

I searched the tools directory, and I couldn't find the nuttx-gdbinit file the this directory. I can find this file in the older nuttx version, so whether there is a new document about the qemugdb?

Thanks for your help!

xiaoxiang781216, 2024-04-19:
The script change to https://github.com/apache/nuttx/blob/master/tools/gdb

acassis, 2024-04-19:
> The script change to https://github.com/apache/nuttx/blob/master/tools/gdb

@anjiahao1 could you please fix https://nuttx.apache.org/docs/latest/guides/qemugdb.html ?

Since nuttx-gdbinit doesn't exist anymore, the documentation currently is useless!

anjiahao1, 2024-04-20:
> > The script change to https://github.com/apache/nuttx/blob/master/tools/gdb
> 
> @anjiahao1 could you please fix https://nuttx.apache.org/docs/latest/guides/qemugdb.html ?
> 
> Since nuttx-gdbinit doesn't exist anymore, the documentation currently is useless!

[pr 12187](https://github.com/apache/nuttx/pull/12187) fixed it.

closed 2024-04-20 by xiaoxiang781216