apache/nuttx #12194: RNDIS is not working with ping

acassis, 2024-04-20:
I'm following this tutorial https://www.youtube.com/watch?v=8noH8v7xNgs

But ping if failing:

```
nsh> ping 10.0.0.1
ERROR: socket() failed: 93
```

acassis, 2024-04-20:
Enabling CONFIG_NET_ICMP_SOCKET fixed the error message. It is strange because in the video tutorial only requires CONFIG_NET_ICMP maybe something changed in the NET stack. @wengzhe do you know why it is required now?

wengzhe, 2024-04-20:
It is weird that the tutorial doesn't need `CONFIG_NET_ICMP_SOCKET`, the ping command uses `socket(AF_INET,SOCK_DGRAM,IPPROTO_ICMP)` at the very beginning (https://github.com/apache/nuttx-apps/blob/c1ec7db8a9420217365dd7c2ff27b15bb09272e5/system/ping/ping.c#L461), and the `CONFIG_NET_ICMP_SOCKET` is always needed for `IPPROTO_ICMP` socket since it was added (https://github.com/apache/nuttx/commit/cccc86da)

acassis, 2024-04-20:
Thank you @wengzhe! Something I noticed in the video: I only pinged from computer to the board, maybe it explains why I didn't see the issue before. But I still need to do more investigation, because ping is not working in any direction now.