apache/nuttx #12209: rv-virt TLSF_MANGER usage issue

yf13, 2024-04-23:
It seems that TLSF heap manager doesn't work well with `rv-virt` on Ubuntu 22.04.

## Reproduce:

- Used configurations: `rv-virt/knsh64`, `rv-virt/knsh32`, or `rv-virt/nsh64`
- Before building, do `kconfig-tweak -e MM_TLSF_MANAGER -d MM_DEFAULT_MANAGER` to enable TLSF manager.
- Build and run nuttx with qemu 6,2 
- expect Nuttx booting up and show `nsh> `prompt.

Please teach if the usage is incorrect.

acassis, 2024-04-29:
@masayuki2009 @lupyuen did you see this issue?

xiaoxiang781216, 2024-04-30:
this is due to compiler generate infinite recursion for ffs on RISCV, @Gary-Hobson will provide a patch fix it.

Gary-Hobson, 2024-04-30:
@yf13 The fixed patch has been submitted  https://github.com/apache/nuttx/pull/12263

You can add the -s -S parameter after the qemu startup command to connect to gdb for debugging

```
qemu-system-riscv64 -semihosting -M virt,aclint=on -cpu rv64 -smp 8 -bios none -kernel nuttx -nographic -S -s
gdb-multiarch nuttx  -ex "target remote 127.0.0.1:1234"
```

closed 2024-04-30 by xiaoxiang781216