apache/nuttx #12218: Error formatting sdcard

JorgeGzm, 2024-04-23:
Hello,

I`m trying  format the sdcard and have the error bellow:

```
nsh> uname -a
NuttX 12.5.1 28044f7d5a-dirty Apr 23 2024 20:36:15 arm linum-stm32h753bi
nsh> mkfatfs /dev/mmcsd0
bchdev_register: blkdev="/dev/mmcsd0" chardev="/dev/tmpc000001" readonly=F
find_blockdriver: pathname="/dev/mmcsd0"
mmcsd_open: Entry
mmcsd_geometry: Entry
mmcsd_geometry: available: true mediachanged: true writeenabled: true
mmcsd_geometry: nsectors: 1947648 sectorsize: 512
mmcsd_geometry: Entry
mmcsd_geometry: available: true mediachanged: false writeenabled: true
mmcsd_geometry: nsectors: 1947648 sectorsize: 512
mkfatfs_clustersearch: Configuring with 32 sectors/cluster...
mkfatfs_tryfat12: nfatsects=179 nclusters=60851 (max=4080)
mkfatfs_tryfat12: WARNING:  Too many clusters for FAT12: 60851 > 4078
mkfatfs_clustersearch: WARNING:  Cannot format FAT12 at 32 sectors/cluster
mkfatfs_tryfat16: nfatsects=238 nclusters=60848 (min=4081 max=60928)
mkfatfs_selectfat: Selected FAT16
mkfatfs_configfatfs: Sector size:          512 bytes
mkfatfs_configfatfs: Number of sectors:    1947648 sectors
mkfatfs_configfatfs: FAT size:             16 bits
mkfatfs_configfatfs: Number FATs:          2
mkfatfs_configfatfs: Sectors per cluster:  32 sectors
mkfatfs_configfatfs: FS size:              238 sectors
mkfatfs_configfatfs:                       60848 clusters
mkfatfs_configfatfs: Root directory slots: 512
mkfatfs_configfatfs: Volume ID:            00000000
mkfatfs_configfatfs: Volume Label:         "           "
mmcsd_write: startsector: 0 nsectors: 1 sectorsize: 512
mmcsd_writesingle: startblock=0
stm32_dmapreflight: invalid IDMA address buffer:0x380026e8 end:0x380028e7
bchlib_write: ERROR: Write failed: -14
mkfatfs_devwrite: ERROR:  write failed: size=512 pos=0 error=-14
mmcsd_close: Entry
nsh: mkfatfs: mkfatfs failed: 14
nsh> 
```

My board is using a STM32H753.

acassis, 2024-04-24:
I saw @davids5 replied you in Discord channel. I'll post his comment here for future reference:
"See 'invalid IDMA address buffer:0x380026e8 end:0x380028e7' The buffer used by mkfatfs is not allined - try change the malloc  to memalign(ARMV7M_DCACHE_LINESIZE,sizeneeded)"

JorgeGzm, 2024-04-25:
I verified that the error occurs because the stm32_dmapreflight function checks whether the buffer used by IDMA is allocated in RAM4

```c
#if defined(CONFIG_STM32H7_SDMMC_IDMA) && defined(CONFIG_ARCH_HAVE_SDIO_PREFLIGHT)
static int stm32_dmapreflight(struct sdio_dev_s *dev,
                              const uint8_t *buffer, size_t buflen)
...
      /* For SDMMC1, IDMA cannot access SRAM123 or SRAM4. */

      if (((uintptr_t)buffer >= SRAM123_START &&
          (uintptr_t)buffer + buflen <= SRAM123_END) ||
          ((uintptr_t)buffer >= SRAM4_START &&
          (uintptr_t)buffer + buflen <= SRAM4_END))
        {
          mcerr("invalid IDMA address "
                "buffer:0x%08" PRIxPTR " end:0x%08" PRIxPTR "\n",
                (uintptr_t)buffer, (uintptr_t)(buffer + buflen - 1));
          return -EFAULT;
...
```

I check the function fat_buffer_alloc returned a memory region in RAM4
[image]

I changed the arm_addregion function to use only DTCM memory, but it doesn't seem like the right thing to do.

Even with this change I have the error below.

```bash
nsh> mkfatfs /dev/mmcsd0
mkfatfs_tryfat12: WARNING:  Too many clusters for FAT12: 60307 > 4078
mkfatfs_clustersearch: WARNING:  Cannot format FAT12 at 32 sectors/cluster
stm32_sdmmc_interrupt: ERROR: TX FIFO underrun, remaining: 512
mmcsd_eventwait: ERROR: Awakened with 14
mmcsd_writesingle: ERROR: CMD24 transfer failed: -5
bchlib_write: ERROR: Write failed: -5
mkfatfs_devwrite: ERROR:  write failed: size=512 pos=0 error=-5
nsh: mkfatfs: mkfatfs failed: 5
```