apache/nuttx #12223: with ESP32 connecting to 6 BLE devices

WJ-JAY, 2024-04-24:
We are connecting 6 BLE devices using ESP32. After running for a period of time, typically a few hours, the following assert occurs.

Hardware Platform：esp32-wrover-kit
Branch：Master
Bluetooth stack：nimble
SMP：disenable
log：

[2024/4/23 20:55:37] _assert: Current Version: NuttX  12.5.1 aa0877247a-dirty Apr 23 2024 17:11:35 xtensa
[2024/4/23 20:55:37] _assert: Assertion failed (g_current_regs[(0)]) == ((void*)0): at file: common/xtensa_irqdispatch.c:58 task: luat process: luat 0x4010d488
[2024/4/23 20:55:37] up_dump_register:    PC: 00000000    PS: 00000000
[2024/4/23 20:55:37] up_dump_register:    A0: 00000000    A1: 00000000    A2: 00000000    A3: 00000000
[2024/4/23 20:55:37] up_dump_register:    A4: 00000000    A5: 00000000    A6: 00000000    A7: 00000000
[2024/4/23 20:55:37] up_dump_register:    A8: 00000000    A9: 00000000   A10: 00000000   A11: 00000000
[2024/4/23 20:55:37] up_dump_register:   A12: 00000000   A13: 00000000   A14: 00000000   A15: 00000000
[2024/4/23 20:55:37] up_dump_register:   SAR: 00000000 CAUSE: 00000000 VADDR: 00000000
[2024/4/23 20:55:37] up_dump_register:  LBEG: 00000000  LEND: 00000000  LCNT: 00000000
[2024/4/23 20:55:37] dump_stacks: ERROR: Stack pointer is not within the stack
[2024/4/23 20:55:37] dump_stack: IRQ Stack:
[2024/4/23 20:55:37] dump_stack:   base: 0x3ffcf780
[2024/4/23 20:55:37] dump_stack:   size: 00004096
[2024/4/23 20:55:37] stack_dump: 0x3ffcf780: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf7a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf7c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf7e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf800: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf820: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf840: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf860: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf880: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf8a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf8c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf8e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf900: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf920: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf940: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf960: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf980: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf9a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf9c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcf9e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:37] stack_dump: 0x3ffcfa00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfa20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfa40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfa60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfa80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfaa0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfac0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfae0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfb00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfb20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfb40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfb60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfb80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfba0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfbc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfbe0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfc00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfc20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfc40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfc60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfc80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfca0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfcc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfce0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfd00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfd20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfd40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfd60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[2024/4/23 20:55:38] stack_dump: 0x3ffcfd80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... 187 lines cut ...]
[2024/4/23 20:55:39] I (213) boot: ## Label            Usage          Type ST Offset   Length
[2024/4/23 20:55:39] I (221) boot:  0 nvs              WiFi data        01 02 00009000 00004000
[2024/4/23 20:55:39] I (228) boot:  1 otadata          OTA data         01 00 0000d000 00002000
[2024/4/23 20:55:39] I (236) boot:  2 phy_init         RF data          01 01 0000f000 00001000
[2024/4/23 20:55:39] I (244) boot:  3 ota_0            OTA app          00 10 00010000 00200000
[2024/4/23 20:55:39] I (251) boot:  4 ota_1            OTA app          00 11 00210000 00200000
[2024/4/23 20:55:39] I (259) boot: End of partition table
[2024/4/23 20:55:39] I (263) esp_image: segment 0: paddr=00210020 vaddr=3f400020 size=4c674h (312948) map
[2024/4/23 20:55:39] I (379) esp_image: segment 1: paddr=0025c69c vaddr=3ffcfb20 size=017ach (  6060) load
[2024/4/23 20:55:39] I (381) esp_image: segment 2: paddr=0025de50 vaddr=40080000 size=021c8h (  8648) load
[2024/4/23 20:55:39] I (388) esp_image: segment 3: paddr=00260020 vaddr=400d0020 size=828fch (534780) map
[2024/4/23 20:55:39] E (576) esp_image: invalid segment length 0x20202020
[2024/4/23 20:55:39] E (576) boot: OTA app partition slot 1 is not bootable
[2024/4/23 20:55:39] I (576) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=52b08h (338696) map
[... cut ...]
[2024/4/23 20:55:40] I (922) esp_image: segment 4: paddr=000fca74 vaddr=4008afd8 size=04c80h ( 19584) load
[2024/4/23 20:55:40] I (938) boot: Loaded app from partition at offset 0x10000
[2024/4/23 20:55:40] I (938) boot: Disabling RNG early entropy source...
[2024/4/23 20:55:40] A__esp32_start: ESP32 chip revision is v3.0
[2024/4/23 20:55:40] BI (328) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
[2024/4/23 20:55:40] 
[2024/4/23 20:55:40] NuttShell (NSH) NuttX-12.5.1
[2024/4/24 10:05:07] nsh>

acassis, 2024-04-24:
@tmedicci Did you know what could be this issue?

@WJ-JAY does this issue happen if you connect only 1 BLE device? Or only 2 devices?

WJ-JAY, 2024-04-25:
@acassis   I can try later, but connecting 6 BLEs is our top priority. We are eager to solve this issue, and we would greatly appreciate it if we succeed.  In the future, we would like to enable both WIFI and connecting to 6 BLE devices simultaneously. Is this also possible?

tmedicci, 2024-04-25:
Hi @WJ-JAY ,

Please try enabling stack coloration to check if it isn't a stack overflow (`CONFIG_STACK_COLORATION`). If you identify some task whose stack is near its capacity, please increase it.

Also,  there is a list of saved interrupt states that contains `NR_IRQSTATE_FLAGS` entries. It is initialized [here](https://github.com/apache/incubator-nuttx/blob/09464ff9bc03e86d9138e0987ae4371bbc51c0e4/arch/xtensa/src/esp32/esp32_ble_adapter.c#L2908). Try to increase [CONFIG_ESP32_BLE_INTERRUPT_SAVE_STATUS](https://github.com/apache/incubator-nuttx/blob/09464ff9bc03e86d9138e0987ae4371bbc51c0e4/arch/xtensa/src/esp32/esp32_ble_adapter.c#L129) to let more interrupt states to be saved.

WJ-JAY, 2024-04-27:
Hi @tmedicci ,
I made adjustments to the stack sizes of each thread as you advised, but the previous issue still occurred. It seems that the problem is not caused by stack overflow. I also set CONFIG_ESP32_BLE_INTERRUPT_SAVE_STATUS to 5. Below is the log when the issue occurred. Please help analyze, thank you.

nsh> Call form key client. Key: 0. value: 1
Local Server Excute Time : 0ms Service Id :2
_assert: Current Version: NuttX  12.5.1 8b9bed1544-dirty Apr 26 2024 18:35:37 xtensa
_assert: Assertion failed (g_current_regs[(0)]) == ((void*)0): at file: common/xtensa_irqdispatch.c:58 task: luat process: luat 0x4010d698
up_dump_register:    PC: 00000000    PS: 00000000
up_dump_register:    A0: 00000000    A1: 00000000    A2: 00000000    A3: 00000000
up_dump_register:    A4: 00000000    A5: 00000000    A6: 00000000    A7: 00000000
up_dump_register:    A8: 00000000    A9: 00000000   A10: 00000000   A11: 00000000
up_dump_register:   A12: 00000000   A13: 00000000   A14: 00000000   A15: 00000000
up_dump_register:   SAR: 00000000 CAUSE: 00000000 VADDR: 00000000
up_dump_register:  LBEG: 00000000  LEND: 00000000  LCNT: 00000000
dump_stacks: ERROR: Stack pointer is not within the stack
dump_stack: IRQ Stack:
dump_stack:   base: 0x3ffceeb0
dump_stack:   size: 00008192
stack_dump: 0x3ffd0b60: 00000000 00000000 00000000 00000009 00000008 00000000 00000008 00000009
stack_dump: 0x3ffd0b80: deadbeef deadbeef deadbeef deadbeef 00000030 00000060 3ffd0cc0 3ffd0ce0
stack_dump: 0x3ffd0ba0: 00000008 3f408d26 deadbeef 00000008 80104210 3ffd0be0 fffffffc 3ffd0c20
stack_dump: 0x3ffd0bc0: 64363938 00000042 3ffd0cf0 3ffd0d10 801018d2 3ffd0c00 3ffd0c20 3f408cf7
stack_dump: 0x3ffd0be0: 00000038 3f408d27 00000000 3f408d27 800d75ac 3ffd0c20 00000060 3f408cf7
stack_dump: 0x3ffd0c00: 3ffd0ce0 3ffd0cc0 00000008 3f408d59 800d75d8 3ffd0c70 00000000 3f408cf7
stack_dump: 0x3ffd0c20: 00000060 4010413c 401040ac 40104068 00000000 3ffc1bf8 000000c4 00000000
stack_dump: 0x3ffd0c40: 0000000a 3ffd0d00 00000008 3f408e60 0000000a 3ffd0d00 00000008 3f408e60
stack_dump: 0x3ffd0c60: 8008e475 3ffd0cb0 00000000 3f408cf7 3ffd0ce0 3ffd0cc0 00000008 00000000
stack_dump: 0x3ffd0c80: 3ffd0ce0 3ffd0cc0 00000008 40104068 3ffd0ce0 3ffd0cc0 00000008 3ffd0d20
stack_dump: 0x3ffd0ca0: 8008e4e1 3ffd0d00 3ffd0ca0 3ffd0eb0 3ffd0ce0 3ffd0cc0 00000008 00000006
stack_dump: 0x3ffd0cc0: 3ffd0d20 3ffd0d00 3f408f76 3ffd0ca0 8008e4e1 3ffd0d00 00000008 00000006
stack_dump: 0x3ffd0ce0: 3f408f76 3ffd0cc0 3ffd0d20 3ffd0d00 8008e7b9 3ffd0d40 3f408e61 00000000
stack_dump: 0x3ffd0d00: 3ffd0d20 3ffd0d00 8008e7b9 3ffd0d40 3f408e61 00000000 00000008 00000000
stack_dump: 0x3ffd0d20: 00000000 00000000 00000008 00000000 8008e341 3ffd0d60 3ffe7530 3ffffa58
stack_dump: 0x3ffd0d40: 3ffceeb0 00002000 00000350 3f408f81 80082df5 3ffd0e50 3f40466f 0000003a
stack_dump: 0x3ffd0d60: 3ffffaf8 3ffffaf8 4010d698 3ffcd69c 7474754e 3ffb0058 0000000c 00000000
stack_dump: 0x3ffd0d80: 80089db9 3ffd00d0 3ffb80ec 3ffe38c0 00000023 3ffb12d6 00000000 00000002
stack_dump: 0x3ffd0da0: 00000007 2e323100 00312e35 3ffe2cd8 000000ff 3ffcdcc0 62380001 64656239
stack_dump: 0x3ffd0dc0: 34343531 7269642d 41207974 32207270 30322036 31203432 35333a38 0037333a
stack_dump: 0x3ffd0de0: 8008cbb8 3ffd0e10 40089d18 657478b8 0061736e 3ffb9274 40083d28 3f88e560
stack_dump: 0x3ffd0e00: 8008e8fd 3ffffa58 3ffc3250 3f40464b 3f40466f 0000003a 3f9047ec 3f88e560
stack_dump: 0x3ffd0e20: 00001fd0 0000003a 00060023 00000001 3ffceeb0 00000000 00000001 3ffe9500
stack_dump: 0x3ffd0e40: 80080fe2 3ffd0e70 00000004 3ffd0c48 3f40464b 3ffd0df0 3ffb8d40 00000200
stack_dump: 0x3ffd0e60: 40080b2e 3ffd0e90 20000000 3ffd0c48 3ffc22bc 3ffcd63c 00000000 00076bb8
stack_dump: 0x3ffd0e80: 4008cca4 3ffd0eb0 00000001 3ffd0d20 0000001d 20000000 0000001f 00000004
stack_dump: 0x3ffd0ea0: 8008a9fc 3ffd0d20 deadbeef deadbeef 00000000 00000000 00000000 00000000
dump_stack: User Stack:
dump_stack:   base: 0x3ffe7530
dump_stack:   size: 00008144
dump_stack:     sp: 0x3ffe9050
stack_dump: 0x3ffe9030: 00000006 00000000 00000000 00000006 80134f36 3ffe9060 3ffe91f0 3f9047ec
stack_dump: 0x3ffe9050: 8013500a 3ffe90e0 3f9047ec 3f88e560 3ffe91f0 00000003 4012c2a8 3ffe90a0
stack_dump: 0x3ffe9070: 3ffe90a0 4012a774 00000001 00000006 8012a796 3ffe90d0 3f9047ec 3f8e2d54
stack_dump: 0x3ffe9090: 00000001 8013500a 3ffe90e0 3f9047ec 00000001 8012bca4 00000000 00000000
stack_dump: 0x3ffe90b0: 3f9047ec 00000002 40134b7c 3ffe90e0 3ffe90e0 fffffc2a 00000000 3f904864
stack_dump: 0x3ffe90d0: 80137d01 3ffe9120 3f9047ec 3f88e560 3f88e560 00000000 3f9047ec 00000000
stack_dump: 0x3ffe90f0: 00000000 3f9019a0 3f8eb7a0 3f8eb7b0 00000001 00000000 00000000 3f44c430
stack_dump: 0x3ffe9110: 8012c380 3ffe9140 3f9047ec 3f8f0cd8 00000001 3f8dd230 00000000 00000001
stack_dump: 0x3ffe9130: 8012c3ac 3ffe9190 3f9047ec 3f903c80 42140000 42100000 3ffe91e0 3ffffa58
stack_dump: 0x3ffe9150: 3f901ad8 00000000 3f8ffa50 00000000 00000007 00000010 ffffffff 40080af4
stack_dump: 0x3ffe9170: 3f8eb788 3f88e560 00000001 3f8eb790 801274a4 3ffe91b0 3f9047ec 3f903c80
stack_dump: 0x3ffe9190: ffffffff 3ffd0e50 00000004 3ffe9108 8012bcb2 3ffe91d0 3f9047ec 3ffe92a0
stack_dump: 0x3ffe91b0: ffffffff 00060820 3ffe91e0 3ffffa58 8012c558 3ffe91f0 3ffe93a0 40127494
stack_dump: 0x3ffe91d0: 40127494 3ffe9260 3ffe9214 3ffe9260 80128158 3ffe9270 00000000 40127494
stack_dump: 0x3ffe91f0: 3ffe93a0 8012c558 3ffe91f0 3ffe93a0 3f9047ec 3ffe92a0 00000000 00000002
[... cut ...]
backtrace|12: 0x000e343a 0x000e345c 0x000e3470 0x0014846a 0x001020e0 0x00150cb8
backtrace|13: 0x00150a9c 0x0015086a 0x001508a6 0x001508be 0x00123544 0x00122827 0x001216ee 0x001226a0
backtrace|13: 0x001020e0 0x00150cb8
backtrace|1?? Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6952
load:0x40078000,len:15488
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3752
entry 0x40080648

rftafas, 2024-05-06:
Simple question: is this some form of mesh? Or is this a set up with 1 host and 5 slave devices?

rftafas, 2024-05-06:
> > Simple question: is this some form of mesh? Or is this a set up with 1 host and 5 slave devices?
> 
> Yes, it's 1 host and 5 slaves

And who is crashing?

WJ-JAY, 2024-05-07:
> > > Simple question: is this some form of mesh? Or is this a set up with 1 host and 5 slave devices?
> > 
> > 
> > Yes, it's 1 host and 5 slaves
> 
> And who is crashing?

We now use esp32 to run nuttx as the host and other modules as the slave. The crashing is esp32

rftafas, 2024-05-07:
> > > > Simple question: is this some form of mesh? Or is this a set up with 1 host and 5 slave devices?
> > > 
> > > 
> > > Yes, it's 1 host and 5 slaves
> > 
> > 
> > And who is crashing?
> 
> We now use esp32 to run nuttx as the host and other modules as the slave. The crashing is esp32

All six? Or is it the host?
Do you have Wi-Fi enabled as well?

WJ-JAY, 2024-05-08:
> All six? Or is it the host? Do you have Wi-Fi enabled as well?

Only the ESP32 is running NuttX and acting as the host, while the other 6 modules are of different models and not running NuttX. The exception is the ESP32 host, which is currently not enabled for  WiFi.