apache/nuttx #12260: esp32 build with many warning

xiaoxiang781216, 2024-04-29:
@acassis could you fix the following esp warning:
https://github.com/apache/nuttx/actions/runs/8881086734/job/24382775822?pr=12256
```
====================================================================================
Configuration/Tool: esp32s3-lcd-ev/lvgl
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Building NuttX...
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_fields.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_api.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_fields.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_api.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_fields.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_api.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_fields.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_api.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_fields.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/esp32s3/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_utility.o' given more than once in the same rule
Makefile:143: target 'chip/esp-hal-3rdparty/components/efuse/src/esp_efuse_api.o' given more than once in the same rule
```

acassis, 2024-04-29:
@xiaoxiang781216 I don't know nothing about esp-hal-3rdparty, it was added when I was leaving Espressif.
Maybe @xiaoxiang781216 knows how to fix it.

xiaoxiang781216, 2024-04-29:
who is the maintainer of Espressif?

eren-terzioglu, 2024-04-30:
Thanks for notifying, working on it.

eren-terzioglu, 2024-04-30:
Created https://github.com/apache/nuttx/pull/12267

acassis, 2024-04-30:
Closed by #12267

closed 2024-04-30 by acassis