apache/nuttx #12275: `rv-virt` kernel builds are brokken

yf13, 2024-05-02:
On Ubuntu 22.04 with `qemu-system-riscv` targets and `gcc-riscv64-unknown-elf` toolchain (GCC 10.2) and latest upstream with ending commit cc9c3ed80b.

- The build from `rv-virt/nsh` and `rv-virt/nsh64` can boot.
- The build from both `rvivirt/knsh32` and `rv-virt/knsh64` can't boot. The `ABC` string is invibsile from console.

lupyuen, 2024-05-04:
@inochisa Any idea why rv-virt:knsh64 crashes on startup? I tested on macOS with QEMU 7.0.0

https://gist.github.com/lupyuen/ce07dd0c21118cebe8e83a8d61439112

```text
→ qemu-system-riscv64 -nographic -semihosting -M virt,aclint=on -cpu rv64 -kernel nuttx

OpenSBI v1.0
...
ABC[    0.000000] riscv_exception: EXCEPTION: Illegal instruction. MCAUSE: 0000000000000002, EPC: 000000008020853a, MTVAL: 0000000014d79073
[    0.000000] riscv_exception: PANIC!!! Exception = 0000000000000002
[    0.000000] _assert: Current Version: NuttX  12.5.1-RC0 0e67a79 May  4 2024 08:54:24 risc-v
[    0.000000] _assert: Assertion failed panic: at file: common/riscv_exception.c:119 task: Idle_Task process: Kernel 0x80200dec
[    0.000000] up_dump_register: EPC: 000000008020853a
[    0.000000] up_dump_register: A0: 00000000804072c0 A1: 0000000000000000 A2: 0000000000000000 A3: 0000000000000001
[    0.000000] up_dump_register: A4: 0000000000000000 A5: ffffffffffffffff A6: 0000000000000000 A7: fffffffffffffff8
[    0.000000] up_dump_register: T0: 0000000080200050 T1: 0000000000000007 T2: 0000000000001000 T3: 00000000804072f8
[    0.000000] up_dump_register: T4: 00000000804072f0 T5: 0000000000000027 T6: 0000000000000001
[    0.000000] up_dump_register: S0: 00000000804072c0 S1: 0000000000000015 S2: 0000000000989680 S3: 0000000002004000
[    0.000000] up_dump_register: S4: 000000000200bff8 S5: 0000000000000000 S6: 8000000a00006800 S7: 000000000000007f
[    0.000000] up_dump_register: S8: 0000000080017038 S9: 0000000080038ea0 S10: 0000000000000000 S11: 0000000000000000
[    0.000000] up_dump_register: SP: 0000000080406b50 FP: 00000000804072c0 TP: 000000008003e000 RA: 000000008020851e
[    0.000000] dump_stack: User Stack:
[    0.000000] dump_stack:   base: 0x80406010
[    0.000000] dump_stack:   size: 00003056
[    0.000000] dump_stack:     sp: 0x80406b50
[    0.000000] stack_dump: 0x80406b30: 80401b30 00000000 00000038 00000000 80400af8 00000000 8020851e 00000000
[    0.000000] stack_dump: 0x80406b50: 80400c70 00000000 80400ae8 00000000 80401b30 00000000 80401b2c 00000000
[    0.000000] stack_dump: 0x80406b70: 80400af8 00000000 8020835e 00000000 80400af8 00000000 802019b6 00000000
[    0.000000] stack_dump: 0x80406b90: 80400af8 00000000 80200fd8 00000000 80600000 00000000 00400000 00000000
[    0.000000] stack_dump: 0x80406bb0: 00000000 00000000 87000000 00000000 80200000 00000000 80406000 00000000
[    0.000000] stack_dump: 0x80406bd0: 00000000 00000000 80200706 00000000 00000000 00000000 00000001 00000000
[    0.000000] stack_dump: 0x80406bf0: 8003df30 00000000 8020004a 00000000 00000000 00000000 00000000 00000000
[    0.000000] dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE      USED   FILLED    COMMAND
[    0.000000] dump_tasks:   ----   --- --- -------- ------- --- ------- ---------- ---------------- 0x80400a80      2048      1016    49.6%    irq
[    0.000000] dump_task:       0     0   0 FIFO     Kthread - Running            0000000000000000 0x80406010      3056       704    23.0%    Idle_Task
QEMU: Terminated

→ qemu-system-riscv64 --version
QEMU emulator version 7.0.0
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers
```

inochisa, 2024-05-04:
@yf13  @lupyuen  Can you try opensbi v1.1. I think this may be caused by SSTC.

lupyuen, 2024-05-04:
The RISC-V Exception:

```text
EXCEPTION: Illegal instruction. MCAUSE: 0000000000000002, EPC: 000000008020853a, MTVAL: 0000000014d79073
```

Points to this code:

```text
riscv_write_stime():
/private/tmp/nuttx/nuttx/arch/risc-v/src/common/riscv_mtimer.c:135
  WRITE_CSR(CSR_STIMECMP, value);
    80208538:	57fd                	li	a5,-1
    8020853a:	14d79073          	csrw	0x14d,a5
```

So CSR_STIMECMP is probably missing. How do we get Opensbi v1.1? Is it in the NuttX Doc?

lupyuen, 2024-05-04:
@inochisa STIMECMP needs QEMU 7.2 I think?

https://github.com/qemu/qemu/commit/e46e2628e9fcce39e7ae28ac8c24bcc643ac48eb

inochisa, 2024-05-04:
> The RISC-V Exception:
> 
> ```
> EXCEPTION: Illegal instruction. MCAUSE: 0000000000000002, EPC: 000000008020853a, MTVAL: 0000000014d79073
> ```
> 
> Points to this code:
> 
> ```
> riscv_write_stime():
> /private/tmp/nuttx/nuttx/arch/risc-v/src/common/riscv_mtimer.c:135
>   WRITE_CSR(CSR_STIMECMP, value);
>     80208538:	57fd                	li	a5,-1
>     8020853a:	14d79073          	csrw	0x14d,a5
> ```
> 
> So CSR_STIMECMP is probably missing.  How do we get Opensbi v1.1? Is it in the NuttX Doc?

For now you can build from source, the newer qemu should ship with possible opensbi.

I will write a document for it. Or just disable the SSTC as  temporary solution.

inochisa, 2024-05-04:
> @inochisa STIMECMP needs QEMU 7.2 I think?
> 
> [qemu/qemu@e46e262](https://github.com/qemu/qemu/commit/e46e2628e9fcce39e7ae28ac8c24bcc643ac48eb)

I think you are true. My system is arch and always has newest qemu. So I forgot this issue. Thanks.

lupyuen, 2024-05-04:
OK lemme upgrade QEMU and retest. If it works, let's update the NuttX Docs for QEMU Version thanks!

@yf13 Please try QEMU 7.2 or newer thanks!

lupyuen, 2024-05-04:
Tested OK on QEMU 7.2.9: https://gist.github.com/lupyuen/b2520054b5ecc12a2f33909aaa46121c

@inochisa Could you update the NuttX Doc to say that `QEMU 7.2.9 or later is required, to support RISC-V "Sstc" Extension`. Thank you so much!

inochisa, 2024-05-04:
> Tested OK on QEMU 7.2.9: https://gist.github.com/lupyuen/b2520054b5ecc12a2f33909aaa46121c
> 
> @inochisa Could you update the NuttX Doc to say that `QEMU 7.2.9 or later is required, to support RISC-V "Sstc" Extension`. Thank you so much!

Good, In addition, I will add some comments for users want to use its own OpenSBI.

yf13, 2024-05-04:
@lupyuen and @inochisa, I sent patch 12279  to disable SSTC as I am using stock QEMU (ver 6.2 with SBI v0.9) on Ubuntu 22.04 most of the time, I believe this is easier for people  to start with.

However, there is remaining issue even if SSTC is off, the "usleep 5" command doesn't return back to nsh, this indicts that the rv-virt timer related setup logic might be incomplete when SSTC is off?

inochisa, 2024-05-04:
> @lupyuen and @inochisa, I sent patch #12279 to disable SSTC as I am using stock QEMU (ver 6.2 with SBI v0.9) on Ubuntu 22.04 most of the time, I believe this is easier for people to start with.
>

I do not think it is a good idea. QEMU for risc-v is always active developed. If we stick on old version, we will lost most of the new features. In addition, 7.2.9 is not too new for most distribution.
 
> However, there is remaining issue even if SSTC is off, the "usleep 5" command doesn't return back to nsh, this indicts that the rv-virt timer related setup logic might be incomplete when SSTC is off?

This does worth exploring. If SSTC is off, the timer should go the old path.  I do not change any old code when adding SSTC, so it is more like another problem

lupyuen, 2024-05-04:
@inochisa Is it possible that this is causing the Timer Issue with SSTC Disabled? https://github.com/apache/nuttx/pull/12178#discussion_r1575842471

Could you help to do a Regression Test based on your patch: If SSTC is Disabled, will `usleep` work? Thanks

inochisa, 2024-05-04:
> @inochisa Is it possible that this is causing the Timer Issue with SSTC Disabled? [#12178 (comment)](https://github.com/apache/nuttx/pull/12178#discussion_r1575842471)
> 
> Could you help to do a Regression Test based on your patch: If SSTC is Disabled, will `usleep` work? Thanks

This is weird, I have tested on my machine. I can confirm QEMU 9.0.0 works without SSTC.

inochisa, 2024-05-04:
> @inochisa Do you mean that if ARCH_RV_EXT_SSTC=N, then usleep works OK on rv-virt:knsh64?
> 

Yes.

> If so, I will test ARCH_RV_EXT_SSTC=N with usleep on my old version of QEMU. Thanks

This worth a try. Thanks.

lupyuen, 2024-05-04:
@inochisa Sorry `ARCH_RV_EXT_SSTC=N` failed to build with rv-virt:knsh64. Did I miss something?

https://gist.github.com/lupyuen/749f8235603781c29a351c066cd2fcf9

```bash
$ git clone https://github.com/inochisa/nuttx --branch qemu-s-mode
$ git clone https://github.com/apache/nuttx-apps apps
$ cd nuttx && ./tools/configure.sh rv-virt:knsh64
$ make menuconfig
## Disable ARCH_RV_EXT_SSTC
$ grep SSTC .config

$ make

./chip/qemu_rv_start.c:54:4: error: #error "Target requires kernel in S-mode, enable CONFIG_ARCH_USE_S_MODE"
   54 | #  error "Target requires kernel in S-mode, enable CONFIG_ARCH_USE_S_MODE"
      |    ^~~~~
ERROR: riscv64-unknown-elf-gcc failed: 1
       command: riscv64-unknown-elf-gcc -MT ./qemu_rv_start.o  -M '-fno-common' '-Wall' '-Wstrict-prototypes' '-Wshadow' '-Wundef' '-Wno-attributes' '-Wno-unknown-pragmas' '-Wno-psabi' '-Os' '-fno-strict-aliasing' '-fomit-frame-pointer' '-ffunction-sections' '-fdata-sections' '-nostdlib' '-g' '-mcmodel=medany' '-march=rv64imafdc' '-mabi=lp64d' '-isystem' '/private/tmp/qemu-s-mode/nuttx/include' '-D__NuttX__' '-D__KERNEL__' '-pipe' '-I' '/private/tmp/qemu-s-mode/nuttx/arch/risc-v/src/chip' '-I' '/private/tmp/qemu-s-mode/nuttx/arch/risc-v/src/common' '-I' '/private/tmp/qemu-s-mode/nuttx/sched' ./chip/qemu_rv_start.c
make[2]: *** [/private/tmp/qemu-s-mode/nuttx/tools/Config.mk:230: qemu_rv_start.ddc] Error 1
make[1]: *** [Makefile:233: .depend] Error 2
make: *** [tools/Unix.mk:620: pass2dep] Error 2
```

inochisa, 2024-05-04:
> @inochisa Sorry `ARCH_RV_EXT_SSTC=N` failed to build with rv-virt:knsh64. Did I miss something?
> 
> https://gist.github.com/lupyuen/749f8235603781c29a351c066cd2fcf9
> 
> ```shell
> $ git clone https://github.com/inochisa/nuttx --branch qemu-s-mode
> $ git clone https://github.com/apache/nuttx-apps apps
> $ cd nuttx && ./tools/configure.sh rv-virt:knsh64
> $ make menuconfig
> ## Disable ARCH_RV_EXT_SSTC
> $ grep SSTC .config
> 
> $ make
> 
> ./chip/qemu_rv_start.c:54:4: error: #error "Target requires kernel in S-mode, enable CONFIG_ARCH_USE_S_MODE"
>    54 | #  error "Target requires kernel in S-mode, enable CONFIG_ARCH_USE_S_MODE"
>       |    ^~~~~
> ERROR: riscv64-unknown-elf-gcc failed: 1
>        command: riscv64-unknown-elf-gcc -MT ./qemu_rv_start.o  -M '-fno-common' '-Wall' '-Wstrict-prototypes' '-Wshadow' '-Wundef' '-Wno-attributes' '-Wno-unknown-pragmas' '-Wno-psabi' '-Os' '-fno-strict-aliasing' '-fomit-frame-pointer' '-ffunction-sections' '-fdata-sections' '-nostdlib' '-g' '-mcmodel=medany' '-march=rv64imafdc' '-mabi=lp64d' '-isystem' '/private/tmp/qemu-s-mode/nuttx/include' '-D__NuttX__' '-D__KERNEL__' '-pipe' '-I' '/private/tmp/qemu-s-mode/nuttx/arch/risc-v/src/chip' '-I' '/private/tmp/qemu-s-mode/nuttx/arch/risc-v/src/common' '-I' '/private/tmp/qemu-s-mode/nuttx/sched' ./chip/qemu_rv_start.c
> make[2]: *** [/private/tmp/qemu-s-mode/nuttx/tools/Config.mk:230: qemu_rv_start.ddc] Error 1
> make[1]: *** [Makefile:233: .depend] Error 2
> make: *** [tools/Unix.mk:620: pass2dep] Error 2
> ```

I think you disable CONFIG_ARCH_USE_S_MODE by mistake....

grep should give
```shell
$ cat .config | grep SSTC
# CONFIG_ARCH_RV_EXT_SSTC is not set
```

lupyuen, 2024-05-04:
@inochisa Sorry I goofed.

@yf13 `usleep` works OK with `ARCH_RV_EXT_SSTC=N`, `rv-virt:knsh64`, QEMU 7.0.0 and OpenSBI v1.0.

Could you update to OpenSBI v1.0? I have a hunch that OpenSBI v0.9 might be buggy, we probably shouldn't use it? Thanks!

https://gist.github.com/lupyuen/da195386a4873c572aa26961e4c81d79

```text
→ qemu-system-riscv64 --version
QEMU emulator version 7.0.0
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers

→ qemu-system-riscv64 -nographic -semihosting -M virt,aclint=on -cpu rv64 -kernel nuttx
OpenSBI v1.0
NuttShell (NSH)
nsh> usleep 5
nsh> ostest
...
ostest_main: Exiting with status -1
nsh>
```

yf13, 2024-05-04:
@lupyuen  and @inochisa  thanks for all the information!

Here I tried `opensbi 1.3-1ubuntu0.22.04.2` got via `apt instalal opensbi` with QEMU 6.2 on Ubuntu 22.04, usleep works now with patch #12279.  I don't have other opensbi versions at hand to try but it looks like QEMU 6.2 is fine now.

So it is good that we finally can support the Ubuntu stock QEMU v6.2 this way: `qemu-system-riscv64 -M virt,aclint=on -semihosting -kernel nuttx -bios /usr/lib/riscv64-linux-gnu/opensbi/generic/fw_dynamic.bin`

lupyuen, 2024-05-04:
Thanks @yf13 ! Moving ahead, we should mandate OpenSBI v1.0 or later. Version 0.9 sounds kinda sus :-)

yf13, 2024-05-04:
@lupyuen and @inochisa, I checked with QEMU 6.2 and OpenSBI `v1.0-3ubuntu1` here, it works well. So saying v1.0 in docs should be fine.

xiaoxiang781216, 2024-05-04:
@yf13 could you update opensbi to the workable verion:
https://github.com/apache/nuttx/blob/master/arch/risc-v/src/opensbi/Make.defs

closed 2024-05-04 by yf13

yf13, 2024-07-31:
@xiaoxiang781216, sorry for late response as I just noticed it. I never used the src/opensbi but a glimpse on git log shows it is v1.3.1 since commit 1cb879773a, which is much recent than the v1.0 version I am using with QEMU v6.2