apache/nuttx #12309: lte_send_atcmd_sync says inappropriate I/O for device

gcmurray1990, 2024-05-09:
I am using NuttX 12.2.1 on an stm32 nucleo f103rb with a SIM7600G-H to try to use an AT command (via the LTE library). 

This is how I am using the AT command:
int customhello_main(int argc, char *argv[])
{
  char respbuff[15];
  int x = **lte_send_atcmd_sync("AT\r", 3,
                        respbuff, 15,
                        15);**
  printf("%s", respbuff);
  printf("%d",x);

  return 0;
}

In summary I am just sending an AT command hoping to get “OK”, but instead am getting the error code -25 (inappropriate i/o for device). I tried finding more info on this error but it is very general.

**How do I set up IO appropriately for AT commands and avoid these issues?**

More details:

- For configuration: I have GPIO debug features enabled. Under stm32 peripheral support, I have USART2 enabled for the SIM . Under I/O I have FILE stream enabled. Under networking, I have late driver init, udp, and tcp enabled. Under LTE I have deprecated API enabled. 

- I have tried using USART1 instead of USART2, but I am getting the same error.

- Also, when running make, I am seeing this:

CC:  chip/stm32_dumpgpio.c In file included from chip/stm32_dumpgpio.c:34:
chip/stm32_dumpgpio.c: In function 'stm32_dumpgpio':
chip/stm32_dumpgpio.c:110:9: warning: format '%x' expects argument of type 'unsigned int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
  110 |   _info("GPIO%c pinset: %08x base: %08x -- %s\n",
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  111 |        g_portchar[port], pinset, base, msg);
      |                          ~~~~~~
      |                          |
      |                          uint32_t {aka long unsigned int}
chip/stm32_dumpgpio.c:110:28: note: format string is defined here
  110 |   _info("GPIO%c pinset: %08x base: %08x -- %s\n",
      |                         ~~~^
      |                            |
      |                            unsigned int
      |                         %08lx

acassis, 2024-05-10:
Hi @gcmurray1990 thanks for bring this issue to our attention.

About the warning you can fix using: %"PRIx32"

@SPRESENSE @masayuki2009 any idea why this error is happening?

raiden00pl, 2024-05-23:
With some changes, LTE lib works also with nrf91 (you just need to get rid of alt1250 specific code).

acassis, 2024-05-23:
@raiden00pl did you get it working?

raiden00pl, 2024-05-23:
yes, but this lib is designed to work with socket interface, not a simple serial device. To make this work you have to implement usersock device with LTE_CMDID ioctls like here https://github.com/apache/nuttx/blob/07f0e0c166b8bf3c65561531c8ea7f87a1bb9b53/arch/arm/src/nrf91/nrf91_modem_sock.c#L479 (in this case lte_send_atcmd_sync() that uses LTE_CMDID_SENDATCMD is not supported)

acassis, 2024-05-23:
Nice! Thank you @raiden00pl !!! Maybe it could be a reference for @gcmurray1990 !!!

acassis, 2024-05-23:
Maybe we need some board profile example to nRF91, when I git grep CONFIG_LTE_LAPI it only returns `boards/arm/cxd56xx/spresense/configs/lte/defconfig`

gcmurray1990, 2024-05-23:
@raiden00pl , @acassis , and @SPRESENSE Thank you all for being so helpful and looking into this. I appreciate everyone's time very much.

acassis, 2024-05-23:
You are welcome, if you face some issue integrating usrsocket to support SIM7600G-H please let us know