apache/nuttx #12331: How to use NAND Flash device simulator

laoniaokkk, 2024-05-13:
I add NAND simulation under sim/nsh configuration, when 'nand' is executed, the ‘/dev/nand‘ device will be generated, but when using "mkfatfs /dev/nand" or "echo hello > /dev/nand", some errors will be generated and the program will be killed. I don't know how to use this device, actually I want to try to port the YAFFS2 filesystem based on this emulated NAND, and it has been successfully ported in bare metal, I need some help. 
In addition, there are some problems with this NAND driver: 
1. Some debugging and logging must be enabled；
2. Compilation error under HOST_ARM64; 
3. SPI NAND cannot be supported

And this is my .config
[config.txt](https://github.com/apache/nuttx/files/15290764/config.txt)

Error logs:
NuttShell (NSH) NuttX-12.5.1-RC0
MOTD: username=admin password=Administrator
nsh> nand
nand_ram: [LOWER | initialize]
Driver running!
nand_wrapper: [UPPER 1 | ioctl] Command: 1537, Arg : 139985771286160
nand_wrapper: [UPPER 1 | ioctl] Done
Page size: 512
nsh> echo hello > /dev/nand
nand_wrapper: [UPPER 2 | ioctl] Command: 1537, Arg : 139985771083416
nand_wrapper: [UPPER 2 | ioctl] Done
nand_wrapper: [UPPER 3 | isbad] Blocks: 0
nand_ram: [LOWER 1 | rawread] Page 0
nand_ram: [LOWER 1 | rawread] Done
nand_ram: [LOWER 2 | rawread] Page 1
nand_ram: [LOWER 2 | rawread] Done
*** stack smashing detected ***: <unknown> terminated
已放弃 (核心已转储)

acassis, 2024-05-13:
Hi @laoniaokkk did you look at https://nuttx.apache.org/docs/latest/applications/testing/nand_sim/index.html ?

The "stack smashing detected" means you are passing some memory limit.

Maybe @resyfer could give you more details. Sarav, could you please take a look?

resyfer, 2024-05-16:
Hi @acassis, the execution weirdly stops in-between execution it seems. I am not too knowledgeable about the memory limits in the simulator.

For me the stack trace for mkfatfs on `/dev/nand` sort of went like this:
[mkfatfs.c:mkfatfs():L326](https://github.com/apache/nuttx-apps/blob/bc68ce993b1ec385cb7d8f11b5e8fca96d53f3be/fsutils/mkfatfs/mkfatfs.c#L326)
[fs_open.c:open():L453](https://github.com/apache/nuttx/blob/master/fs/vfs/fs_open.c#L453)
[fs_open.c:nx_vopen():L300](https://github.com/apache/nuttx/blob/master/fs/vfs/fs_open.c#L300)
[fs_open.c:file_vopen():L199](https://github.com/apache/nuttx/blob/master/fs/vfs/fs_open.c#L199)
[fs_blockproxy.c:block_proxy():L167](https://github.com/apache/nuttx/blob/master/fs/driver/fs_blockproxy.c#L167)
[bchdev_register.c:bchdev_register():L61](https://github.com/apache/nuttx/blob/master/drivers/bch/bchdev_register.c#L61)
[bchlib_setup.c:bchlib_setup():L75](https://github.com/apache/nuttx/blob/master/drivers/bch/bchlib_setup.c#L75)
[fs_openblockdriver.c:open_blockdriver():L86](https://github.com/apache/nuttx/blob/master/fs/driver/fs_openblockdriver.c#L86)
[fs_mtdproxy.c:mtd_proxy():L164](https://github.com/apache/nuttx/blob/master/fs/driver/fs_mtdproxy.c#L164)

And in ftl.c, it crashes in between [this area](https://github.com/apache/nuttx/blob/master/drivers/mtd/ftl.c#L839-L864). My config:

- Disabled:
  - `CONFIG_FTL_READAHEAD`
  - `CONFIG_FTL_WRITEBUFFER`

Thus `FTL_HAVE_RWBUFFER` is also disabled. So the entire area has nothing to crash on, and I think, like @acassis  said, it might be due to a memory limit. I am not sure what can prevent this, and I need help on this.

Both `mkfatfs` and `echo` commands provided by @laoniaokkk stop at this point.

`nand` (the device simulator) relies heavily on memory. Currently, the entire memory is allocated in an entire contiguous chunk, which _might_ be a cause. Of course this can be changed to be more lazy in nature.

The sim device is still experimental so a lot of features are missing or might not work in certain circumstances. It has been tested mostly by using littlefs. Thank you for pointing this out!