apache/nuttx #12335: esp32: make bootloader is not working for building from the source code

acassis, 2024-05-13:
While trying to build the bootloader from source code I'm getting this error:

```
-- Building ESP-IDF components for target esp32
CMake Error at /nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/tools/cmake/build.cmake:542 (message):
  /usr/bin/python: Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')
```

These are the steps
-----------------------------

Configure the board:

```
$ ./tools/configure.sh esp32-devkitc:nsh
```

Enable CONFIG_ESP32_BOOTLOADER_BUILD_FROM_SOURCE :

```
$ make menuconfig

System Type  --->
    Bootloader and Image Configuration  --->
    [*] Build binaries from source
```

Then try to build the bootloader:

```
$ make bootloader
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/chip/bootloader/esp-nuttx-bootloader'...
remote: Enumerating objects: 210, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 210 (delta 63), reused 76 (delta 46), pack-reused 78
Receiving objects: 100% (210/210), 32.23 KiB | 6.45 MiB/s, done.
Resolving deltas: 100% (111/111), done.
Submodule 'esp-idf' (https://github.com/espressif/esp-idf.git) registered for path 'esp-idf'
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf'...
Submodule path 'esp-idf': checked out '1559b6309fe207add518828246a05f80582ee59d'
Creating Bootloader configuration
Building Bootloader binaries
CMake Warning:
  Ignoring empty string ("") provided on the command line.

-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Initialising new submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
-- Initialising new submodule components/bt/controller/lib_esp32...
Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib_esp32'
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/components/bt/controller/lib_esp32'...
Submodule path 'components/bt/controller/lib_esp32': checked out 'b867411caf88b485b2fa8b6d272692b6535a267a'
-- Initialising new submodule components/bt/controller/lib_esp32c2/esp32c2-bt-lib...
Submodule 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib' (https://github.com/espressif/esp32c2-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib'
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/components/bt/controller/lib_esp32c2/esp32c2-bt-lib'...
Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out '10a3ea96e083dd0a7a18089407361fd52f0cbc47'
-- Initialising new submodule components/bt/controller/lib_esp32c3_family...
Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c3_family'
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/components/bt/controller/lib_esp32c3_family'...
Submodule path 'components/bt/controller/lib_esp32c3_family': checked out 'bba9af9259e0999ef246426d31a793fe0a3ff4db'
-- Initialising new submodule components/bt/controller/lib_esp32c6/esp32c6-bt-lib...
Submodule 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib' (https://github.com/espressif/esp32c6-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
Cloning into '/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib'...
Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out '658a412455d652be3b07a602a988b25d97b00e03'
-- Initialising new submodule components/bt/controller/lib_esp32h2/esp32h2-bt-lib...
[... 77 lines cut ...]
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/tools/cmake/build.cmake:542 (message):
  /usr/bin/python: Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  /nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:58 (project)

-- Configuring incomplete, errors occurred!
See also "/nuttxspace/nuttx/arch/xtensa/src/esp32/bootloader/esp-nuttx-bootloader/.build-esp32/CMakeFiles/CMakeOutput.log".
make[1]: *** [chip/Bootloader.mk:178: bootloader] Error 1
make: *** [tools/Unix.mk:600: bootloader] Error 2
```

acassis, 2024-05-13:
@tmedicci any idea?

tmedicci, 2024-05-17:
Hi @acassis !

I tested it locally (and our internal CI also tests it daily). I could not find any problems. I suspect this is somehow related to the outdated `kconfig-frontends`, but I was not even able to build it locally (it requires very old libraries). Can you check with `kconfiglib`?

acassis, 2024-05-17:
Thank you @tmedicci actually this issue was reported by @casaroli and I just tried to reproduced the issue. I think he is used kconfiglib, but if this is some issue caused by Kconfig, it should be modified to work with kconfig-frontends and kconfiglib.

tmedicci, 2024-05-24:
@acassis and @casaroli , the bootloader built from the sources uses a wrapper around an IDF-provided repository (the `esp-nuttx-bootloader`). Kconfiglib is a dependency of ESP-IDF, so it must be installed (not by NuttX, but for `esp-nuttx-bootloader`). This is why we are removing support for the IDF's bootloader.

NuttX adopted kconfiglib later and it uses it by default. This required users that still want to use kconfig-frontends to uninstall kconfiglib (or, equivalently, uninstall for their python environment): this is why this problem has risen.

I'm closing this issue because IDF's bootloader is about to be deprecated. I recommend keeping kconfiglib installed to enable building the bootloader from the sources. There are a list of possible workarounds to be used (if kconfig-frontends is required) until we completely deprecate the use of the IDF's bootloader:
- Install kconfiglib in a python environment and activate/deactivate it before/after running `make bootloader`;
- Modify NuttX's build system to enable selecting a preferred tool (instead of using kconfiglib by default, when available).

tmedicci, 2024-05-24:
You're welcome, Alan!

> Actually kconfiglib is not the default option for NuttX, it is optional: 
https://nuttx.apache.org/docs/latest/quickstart/install.html

I meant that, if available, kconfiglib will be used instead of kconfig-frontends:
https://github.com/apache/nuttx/blob/cfa544357eb3081e49f54e5fe3eef8392f26d9ab/tools/Unix.mk#L639