apache/nuttx #12407: riscv m-timer is schduled multiple times per interrupt.

g2gps, 2024-05-27:
While profiling some of the litex layers, I found that the riscv mtimer is scheduled through openSBI multiple times per alarm interrupt. This isn't ideal, as the openSBI ecall is relatively expensive on platforms which don't  have the  supervisor time (sstc) extension. I don't think there is any reason for this behavior to remain as it is. The only requirement from the riscv machine timer is that it is written once to clear the pending interrupt. 

I've done proof on concept locally where the next update time is cached as part of the instance, and updated once at the end of the timer interrupt, which works, but it needs to be tested more thoroughly.

What is a good test configuration to verify that updated behavior is correct? I was thinking qemu-rv:knsh32 with ostest enabled, along with testing on our platform, as it's the only other suitable platform we have available.

Open to suggestions.

acassis, 2024-05-27:
@g2gps I think testing on qemu-rv:knsh32 is a good valid initial test, but it is important to test in other boards supported by NuttX. @lupyuen what do you think?

lupyuen, 2024-05-27:
@g2gps Yep it's good to test on knsh32 and knsh64. Most of the 64-bit RISC-V ports were derived from knsh64 with OpenSBI, without SSTC.

g2gps, 2024-05-27:
Ok, thanks for the suggestions. I'll put together a PR early next week, when I have a chance to test this out properly.

closed 2024-06-02 by acassis