apache/nuttx #12415: Add EXT2/3/4 FS support to NuttX

acassis, 2024-05-28:
I found a nice project that could be integrated on NuttX to support ext2, ext3 and ext4:

https://github.com/gkostka/lwext4

To get it BSD only we need to remove the extended attribute support as explained in the readme.

anoraktrend, 2024-07-30:
They might want to reimplement it, instead.

acassis, 2024-07-30:
Hi @anoraktrend nice you see you here! I remember browsing your git repository sometime ago.

anoraktrend, 2024-07-30:
> Hi @anoraktrend nice you see you here! I remember browsing your git repository sometime ago.

Howdy! I was looking at this as a potential alternative kernel for Iglunix, (frankensrc, etc), because it's a neat project. ext4 is something that is something of a requirement, so this issue is a good one for that reason.

acassis, 2024-07-30:
Amazing @anoraktrend welcome aboard! We are glad you enjoyed it!

acassis, 2024-07-30:
I didn't know about Iglunix, but used many small Linux distros in the past like Slitaz and many others.
Does Iglunix use Toybox? Marco Casaroli is just porting it to NuttX (shell and some commands are already working), I think it could be something useful for you!

anoraktrend, 2024-07-31:
> I didn't know about Iglunix, but used many small Linux distros in the past like Slitaz and many others. Does Iglunix use Toybox? Marco Casaroli is just porting it to NuttX (shell and some commands are already working), I think it could be something useful for you!

Yep, toybox, busybox, mksh, and the LLVM suite. We could replace some of busybox & toybox with @chimera-linux's chimerautils (a neat thing, it ports a lot of the FreeBSD userland to Linux).

acassis, 2024-07-31:
Interesting! Some chimerautils tools seems too big for a small microcontroller, but some tools could be ported/reused on NuttX. Also I noticed that some tools are open Linux specific /proc and /sys files that only exist on Linux. These tools need to be modified as well.

xiaoxiang781216, 2024-10-07:
we are porting https://github.com/gkostka/lwext4 to nuttx