apache/nuttx #12421: BMP280 first read BUG

acassis, 2024-05-28:
Everytime bmp280 run for the first time after reset the board it is reporting incorrect value.

Near to the sea level it should return near to 1013 hPa, but:
```
nsh> bmp280                                                                     
Absolute pressure [hPa] = 609.820007                                            
Temperature [C] = 25.740000                                                     
```

Running it at second time, everything is fine:
```
nsh> bmp280                                                                     
Absolute pressure [hPa] = 1008.929993                                           
Temperature [C] = 20.930000                                                     
nsh>
```

@jerpelea did you see this issue before (git blame show you are author) ?

I'm testing on RaspberryPi Pico: https://github.com/apache/nuttx/pull/12420

raiden00pl, 2024-05-29:
You can check if the time between changing the sensor mode and reading the data is correct. Try to add mdelay(10) here: https://github.com/apache/nuttx/blob/3cc65d5d35e779d64acc8a2a0f47fbe65b305c61/drivers/sensors/bmp280_uorb.c#L635

fdcavalcanti, 2024-05-29:
That's odd.. I don't remember seeing that.
I'll have a look in the datasheet.
Maybe a quick fix would be running a read right after driver instance is created. But sure, we should understand the root cause.

acassis, 2024-05-29:
@raiden00pl it makes no difference.

Modification:
```
$ git diff
diff --git a/drivers/sensors/bmp280_uorb.c b/drivers/sensors/bmp280_uorb.c
index 7acdb664d9..62e195b825 100644
--- a/drivers/sensors/bmp280_uorb.c
+++ b/drivers/sensors/bmp280_uorb.c
@@ -632,6 +632,8 @@ static int bmp280_fetch(FAR struct sensor_lowerhalf_s *lower,
       up_mdelay(6);
     }
 
+  up_mdelay(10);
+
   /* Read pressure & data */
 
   ret = bmp280_getregs(priv, BMP280_PRESS_MSB, buf, 6);
```

Before modification:

```
NuttShell (NSH) NuttX-12.5.1
 
nsh> dmesg
board_bmp280_initialize: Initializing BMP280!
bmp280_checkid: devid: 0x58
bmp280_initialize: T1 = 27551
bmp280_initialize: T2 = 25860
bmp280_initialize: T3 = 50
bmp280_initialize: P1 = 38045
bmp280_initialize: P2 = -10494
bmp280_initialize: P3 = 3024
bmp280_initialize: P4 = 9513
bmp280_initialize: P5 = -303
bmp280_initialize: P6 = -7
bmp280_initialize: P7 = 8588
bmp280_initialize: P8 = -6299
bmp280_initialize: P9 = -242
sensor_custom_register: Registering /dev/uorb/sensor_baro0
bmp280_register: BMP280 driver loaded successfully!
nsh> bmp280
Absolute pressure [hPa] = 609.820007
Temperature [C] = 25.740000
nsh> bmp280
Absolute pressure [hPa] = 1012.669983
Temperature [C] = 24.139999
nsh>
```

After modification:

```
NuttShell (NSH) NuttX-12.5.1
nsh> dmesg
board_bmp280_initialize: Initializing BMP280!
bmp280_checkid: devid: 0x58
bmp280_initialize: T1 = 27551
bmp280_initialize: T2 = 25860
bmp280_initialize: T3 = 50
bmp280_initialize: P1 = 38045
bmp280_initialize: P2 = -10494
bmp280_initialize: P3 = 3024
bmp280_initialize: P4 = 9513
bmp280_initialize: P5 = -303
bmp280_initialize: P6 = -7
bmp280_initialize: P7 = 8588
bmp280_initialize: P8 = -6299
bmp280_initialize: P9 = -242
sensor_custom_register: Registering /dev/uorb/sensor_baro0
bmp280_register: BMP280 driver loaded successfully!
nsh> bmp280
Absolute pressure [hPa] = 609.820007
Temperature [C] = 25.740000
nsh>
```

raiden00pl, 2024-05-31:
I see there is no waiting for any status flag before reading the data. Make sure that `measuring` bit in `status` register is 0. Max measurement time fir standard resolution mode according to doc is 13.3 ms, so 10 ms delay may not be sufficient. 
 I'm not sure if bmp280_activate() is called before data fetch in this case