apache/nuttx #12456: Loopback Configuration Bug in mpfs_ethernet.c

sastel, 2024-06-05:
@acassis on `nuttx-12.5.0`, switching on `CONFIG_NET_LOOPBACK` to get the `lo` localhost working breaks `eth0` interface so that I can no longer ping the board.

The `mpfs_ethernet.c` file turns on the `NETWORK_CONTROL_LOOPBACK_LOCAL` flag in `mpfs_macconfig()` if `CONFIG_NET_LOOPBACK` is defined. I think that this is a misunderstanding of what `CONFIG_NET_LOOPBACK` is for.

```
config NET_LOOPBACK
    bool "Local loopback"
    select ARCH_HAVE_NETDEV_STATISTICS
    default n
    ---help---
        Add support for the local network loopback device, lo.
```

Proposed fix is below. I'd like to open a PR and merge this fix if the NuttX developers approve.

```diff
diff --git a/arch/risc-v/src/mpfs/mpfs_ethernet.c b/arch/risc-v/src/mpfs/mpfs_ethernet.c
index 382477fa9e..0a5755d354 100644
--- a/arch/risc-v/src/mpfs/mpfs_ethernet.c
+++ b/arch/risc-v/src/mpfs/mpfs_ethernet.c
@@ -3104,7 +3104,7 @@ static int mpfs_macconfig(struct mpfs_ethmac_s *priv)
   net_config |= NETWORK_CONFIG_SGMII_MODE_ENABLE | NETWORK_CONFIG_PCS_SELECT;
 #endif

-#ifdef CONFIG_NET_LOOPBACK
+#ifdef CONFIG_MPFS_MAC_LOOPBACK
   net_control |= NETWORK_CONTROL_LOOPBACK_LOCAL;
 #endif
```

acassis, 2024-06-05:
So, is this PolarFire Loopback something specific of is Ethernet controller? If so your modification seems correct.

@pussuw could you please take a look?

sastel, 2024-06-05:
Yes `NETWORK_CONTROL_LOOPBACK_LOCAL` is a bit in the `NETWORK_CONTROL` MAC register on MPFS.