apache/nuttx #12476: how to port a custom board to nuttx？

snikeguo, 2024-06-08:
I have a board named "XA1634" that utilizes the STM32H723ZG as its CPU chip. I am want to port the XA1634 to NuttX and compile it in protect mode.
1.copying my “XA1634” folder to board/arm/stm32h7
2. run command:
```
./tools/configure.sh -l XA1634:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/david/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/david/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/david/nuttxspace/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/david/nuttxspace/nuttx/boards/arm/stm32h7/XA1634/include
LN: drivers/platform to /home/david/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/david/nuttxspace/nuttx/arch/arm/include/stm32h7
LN: arch/arm/src/chip to /home/david/nuttxspace/nuttx/arch/arm/src/stm32h7
LN: arch/arm/src/board to /home/david/nuttxspace/nuttx/boards/arm/stm32h7/XA1634/src
mkkconfig in /home/david/nuttxspace/apps/audioutils
mkkconfig in /home/david/nuttxspace/apps/benchmarks
mkkconfig in /home/david/nuttxspace/apps/boot
mkkconfig in /home/david/nuttxspace/apps/canutils
mkkconfig in /home/david/nuttxspace/apps/crypto
mkkconfig in /home/david/nuttxspace/apps/database
mkkconfig in /home/david/nuttxspace/apps/examples/mcuboot
mkkconfig in /home/david/nuttxspace/apps/examples
mkkconfig in /home/david/nuttxspace/apps/fsutils
mkkconfig in /home/david/nuttxspace/apps/games
mkkconfig in /home/david/nuttxspace/apps/graphics
mkkconfig in /home/david/nuttxspace/apps/industry
mkkconfig in /home/david/nuttxspace/apps/inertial
mkkconfig in /home/david/nuttxspace/apps/interpreters/luamodules
mkkconfig in /home/david/nuttxspace/apps/interpreters
mkkconfig in /home/david/nuttxspace/apps/logging
mkkconfig in /home/david/nuttxspace/apps/lte
mkkconfig in /home/david/nuttxspace/apps/math
mkkconfig in /home/david/nuttxspace/apps/mlearning
mkkconfig in /home/david/nuttxspace/apps/netutils
mkkconfig in /home/david/nuttxspace/apps/sdr
mkkconfig in /home/david/nuttxspace/apps/system
mkkconfig in /home/david/nuttxspace/apps/testing
mkkconfig in /home/david/nuttxspace/apps/wireless/bluetooth
mkkconfig in /home/david/nuttxspace/apps/wireless/ieee802154
mkkconfig in /home/david/nuttxspace/apps/wireless
mkkconfig in /home/david/nuttxspace/apps
#
# configuration written to .config
#
root@david-VMware-Virtual-Platform:/home/david/nuttxspace/nuttx# make menuconfig
CP: arch/dummy/Kconfig to /home/david/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/david/nuttxspace/nuttx//Kconfig
LN: platform/board to /home/david/nuttxspace/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/david/nuttxspace/nuttx//include
LN: drivers/platform to /home/david/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/david/nuttxspace/nuttx/arch/arm/include/stm32h7
LN: arch/arm/src/chip to /home/david/nuttxspace/nuttx/arch/arm/src/stm32h7
LN: arch/arm/src/board to /home/david/nuttxspace/nuttx//src
No directory at /home/david/nuttxspace/nuttx//src
make[1]: *** [tools/Unix.mk:309：arch/arm/src/board] error 1
make: *** [tools/Unix.mk:693：menuconfig] error 2
root@david-VMware-Virtual-Platform:/home/david/nuttxspace/nuttx# 
```

[XA1634.zip](https://github.com/user-attachments/files/15748615/XA1634.zip)

acassis, 2024-06-08:
Hi @snikeguo, this error message happens when you didn't define you board correctly, see:

CP: boards/dummy/Kconfig to /home/david/nuttxspace/nuttx//Kconfig
...
LN: include/arch/board to /home/david/nuttxspace/nuttx//include
...
LN: arch/arm/src/board to /home/david/nuttxspace/nuttx//src

These "nuttx//" means your board directory wasn't found, also open your board configs/nsh/defconfig and verify if all names are defined correctly.

Another suggestion: run "make savedefconfig" and compare the generated defconfig against your configs/nsh/defconfig, sometimes it helps to spot the issue

closed 2024-06-11 by snikeguo

acassis, 2024-06-11:
@snikeguo I saw you closed it, buy please include information about what was the issue, it could help other people