apache/nuttx #12480: cmake build failed

snikeguo, 2024-06-09:
win10
```
Administrator@HomePc MSYS /e/nuttxspace
# git clone https://github.com/apache/nuttx.git nuttx
Cloning into 'nuttx'...
remote: Enumerating objects: 762151, done.
remote: Counting objects: 100% (532/532), done.
remote: Compressing objects: 100% (286/286), done.
remote: Total 762151 (delta 139), reused 454 (delta 121), pack-reused 761619
Receiving objects: 100% (762151/762151), 262.71 MiB | 1.05 MiB/s, done.
Resolving deltas: 100% (594694/594694), done.
Updating files: 100% (22207/22207), done.

Administrator@HomePc MSYS /e/nuttxspace
# cd nuttx/

Administrator@HomePc MSYS /e/nuttxspace/nuttx
# cmake -B build -DBOARD_CONFIG=stm32f4discovery:nsh -GNinja
-- Initializing NuttX
--   Board:  stm32f4discovery
--   Config: nsh
--   Appdir: /e/nuttxspace/nuttx/../apps
-- The C compiler identification is GNU 13.3.0
-- The CXX compiler identification is GNU 13.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /mingw64/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /mingw64/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /mingw64/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (26.8s)
-- Generating done (15.8s)
-- Build files have been written to: /e/nuttxspace/nuttx/build

Administrator@HomePc MSYS /e/nuttxspace/nuttx
# cmake --build build -t menuconfig
[0/1] /bin/sh CMakeFiles/menuconfig-bced083.sh 569cf84a54ff31d0
Kconfig:278: warning: the 'modules' option is not supported. Let me know if this is a problem for you, as it wouldn't be that hard to implement. Note that modules are supported -- Kconfiglib just assumes the symbol name MODULES, like older versions of the C implementation did when 'option modules' wasn't used.
Loaded configuration '/e/nuttxspace/nuttx/build/.config'
No changes to save (for '/e/nuttxspace/nuttx/build/.config')

Administrator@HomePc MSYS /e/nuttxspace/nuttx
# cmake --build build
[0/1] Re-running CMake...
CMake Error at boards/CMakeLists.txt:56 (file):
  file Failed to create link
  '/e/nuttxspace/nuttx/build/arch/arm/src/board/board' because existing path
  cannot be removed: Is a directory

-- Configuring incomplete, errors occurred!
FAILED: build.ninja
/usr/bin/cmake.exe --regenerate-during-build -S/e/nuttxspace/nuttx -B/e/nuttxspace/nuttx/build
ninja: error: rebuilding 'build.ninja': subcommand failed

Administrator@HomePc MSYS /e/nuttxspace/nuttx
# cmake --build build -t menuconfig
[0/1] Re-running CMake...
CMake Error at boards/CMakeLists.txt:56 (file):
  file Failed to create link
  '/e/nuttxspace/nuttx/build/arch/arm/src/board/board' because existing path
  cannot be removed: Is a directory

-- Configuring incomplete, errors occurred!
FAILED: build.ninja
/usr/bin/cmake.exe --regenerate-during-build -S/e/nuttxspace/nuttx -B/e/nuttxspace/nuttx/build
ninja: error: rebuilding 'build.ninja': subcommand failed

Administrator@HomePc MSYS /e/nuttxspace/nuttx
# cmake --build build -t clean
[0/1] Re-running CMake...
CMake Error at boards/CMakeLists.txt:56 (file):
  file Failed to create link
  '/e/nuttxspace/nuttx/build/arch/arm/src/board/board' because existing path
  cannot be removed: Is a directory

-- Configuring incomplete, errors occurred!
FAILED: build.ninja
/usr/bin/cmake.exe --regenerate-during-build -S/e/nuttxspace/nuttx -B/e/nuttxspace/nuttx/build
ninja: error: rebuilding 'build.ninja': subcommand failed

```

**btw,  nuttx can be compiled successfully through the "make menuconfig "and " make", but when I enter the "make menuconfig" command, it runs very very  slowly and it takes about several minutes to display menuconfig.**

simbit18, 2024-06-10:
@snikeguo  currently nuttx cannot be compiled with cmake on msys2

simbit18, 2024-11-22:
hi @snikeguo it should be ok now !
can you try ?

closed 2024-11-23 by xiaoxiang781216