apache/nuttx #12489: Targets which use openamp get build errors when configured without any optimization (-O0)

AndreHeinemans-NXP, 2024-06-12:
Targets which use openamp need to be configured with an optimization level setting other that -O0 to be compiled correctly. 

I tested this with these targets
 - verdin-mx8mp:rpmsg
 - nrf5340-dk:rptun_cpunet
 - nucleo-h745zi:nsh_cm7_rptun
 
When compiled with -O0 then this linker error appears:
```
LD: nuttx
arm-none-eabi-ld: /home/andre/projects/nuttx/nuttx/staging/libopenamp.a(remoteproc_virtio.o): in function `metal_io_read':
remoteproc_virtio.c:(.text.metal_io_read+0xe8): undefined reference to `__atomic_load_8'
arm-none-eabi-ld: /home/andre/projects/nuttx/nuttx/staging/libopenamp.a(remoteproc_virtio.o): in function `metal_io_write':
remoteproc_virtio.c:(.text.metal_io_write+0xe0): undefined reference to `__atomic_store_8'
make[1]: *** [Makefile:211: nuttx] Error 1
make: *** [tools/Unix.mk:546: nuttx] Error 2
```

It's weird that it only fails without any optimization. Maybe it doesn't use HW instructions for atomic operations in that case..? I don't know.

Reproduce:
```
tools/configure.sh verdin-mx8mp:rpmsg
# or any other target using openamp

make menuconfig
# set 'Optimization Level' to 'Suppress Optimization' =CONFIG_DEBUG_NOOPT

# note: do a make clean if target was already build before since openamp will not be rebuild automatically after running menuconfig

make
```

acassis, 2024-06-13:
@masayuki2009 @xiaoxiang781216 any idea?

xiaoxiang781216, 2024-06-13:
@CV-Bowen could you take a look?

closed 2024-06-25 by xiaoxiang781216