apache/nuttx #12500: SDcardread issue Tiva123gxl

shijogeorge24, 2024-06-14:
hey all ,

I was working with a Tiva C to use an SD card. Even though the log says the bring-up is successful, I can't mount the device. How can I fix this issue? I have tested with 4GB, 8GB, and 16GB cards. None of them seem to be working.

tm4c_bringup: starting bring up
tm4c_i2ctool: hello register i2c
tm4c_i2c_register: bus info=0
tm4c_i2c_register: bus info=1
tm4c_i2c_register: bus info=2
tm4c_i2c_register: bus info=3
tm4c_bringup: Initializing SPI port 0
tiva_ssibus_initialize: port: 0
ssi_setmodeinternal: mode: 0
ssi_setbitsinternal: nbits: 8
ssi_setbitsinternal: CR0: 00000007
ssi_setfrequencyinternal: frequency: 400000
ssi_setfrequencyinternal: CR0: 00006307 CPSR: 00000002
ssi_enable: CR1: 00000002
tm4c_bringup: Successfully initialized SPI port 0
tm4c_bringup: Binding SPI port 0 to MMC/SD slot 0
ssi_disable: CR1: 00000000
ssi_setmodeinternal: mode: 0
ssi_enable: CR1: 00000002
ssi_disable: CR1: 00000000
ssi_setbitsinternal: nbits: 8
ssi_enable: CR1: 00000002
ssi_disable: CR1: 00000000
ssi_setfrequencyinternal: frequency: 400000
ssi_enable: CR1: 00000002
tiva_ssistatus: Returning SPI_STATUS_PRESENT
ssi_disable: CR1: 00000000
ssi_setfrequencyinternal: frequency: 400000
ssi_enable: CR1: 00000002
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-00
mmcsd_mediainitialize: Send CMD0
tiva_ssiselect: devid: 65536 CS: assert
ssi_transfer: txbuffer: 0x200043b4 rxbuffer: 0x200043bc nwords: 1
ssi_txuint8: TX: 0x200043b4->ff
ssi_rxuint8: RX: 0x200043bc<-00
ssi_transfer: txbuffer: 0x200043b4 rxbuffer: 0x200043bc nwords: 1
ssi_txuint8: TX: 0x200043b4->ff
[... 712 lines cut ...]
mmcsd_sendcmd: CMD0[00000000] R1=00
tiva_ssiselect: devid: 65536 CS: de-assert
mmcsd_mediainitialize: ERROR: Send CMD0 failed: R1=00
tiva_ssiselect: devid: 65536 CS: de-assert
mmcsd_open: Failed to initialize card
ssi_transfer: txbuffer: 0x200042d4 rxbuffer: 0x200042dc nwords: 1
ssi_txuint8: TX: 0x200042d4->ff
ssi_rxuint8: RX: 0x200042dc<-00
nx_mount: ERROR: Bind method failed: -19
nsh: mount: mount failed: 19
nsh> ls /dev
/dev:
 can0
 console
 gpio536870914
 gpio536870932
 i2c0
 i2c1
 i2c2
 i2c3
 mmcsd0
 null
 ttyS0
 ttyS1
nsh>

shijogeorge24, 2024-06-15:
added a pull-up resistor to the CS line. Now, registering works fine, but the mount is still not working.  
 
 
 tm4c_bringup: starting bring up
tm4c_i2ctool: hello register i2c
tm4c_i2c_register: bus info=0
tm4c_i2c_register: bus info=1
tm4c_i2c_register: bus info=2
tm4c_i2c_register: bus info=3
tm4c_bringup: Initializing SPI port 0
tiva_ssibus_initialize: port: 0
ssi_setmodeinternal: mode: 0
ssi_setbitsinternal: nbits: 8
ssi_setbitsinternal: CR0: 00000007
ssi_setfrequencyinternal: frequency: 400000
ssi_setfrequencyinternal: CR0: 00006307 CPSR: 00000002
ssi_enable: CR1: 00000002
tm4c_bringup: Successfully initialized SPI port 0
tm4c_bringup: Binding SPI port 0 to MMC/SD slot 0
ssi_disable: CR1: 00000000
ssi_setmodeinternal: mode: 0
ssi_enable: CR1: 00000002
ssi_disable: CR1: 00000000
ssi_setbitsinternal: nbits: 8
ssi_enable: CR1: 00000002
ssi_disable: CR1: 00000000
ssi_setfrequencyinternal: frequency: 400000
ssi_enable: CR1: 00000002
tiva_ssistatus: Returning SPI_STATUS_PRESENT
ssi_disable: CR1: 00000000
ssi_setfrequencyinternal: frequency: 400000
ssi_enable: CR1: 00000002
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-fe
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-09
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
ssi_transfer: txbuffer: 0x200043fc rxbuffer: 0x20004404 nwords: 1
ssi_txuint8: TX: 0x200043fc->ff
ssi_rxuint8: RX: 0x20004404<-ff
mmcsd_mediainitialize: Send CMD0
tiva_ssiselect: devid: 65536 CS: assert
ssi_transfer: txbuffer: 0x200043b4 rxbuffer: 0x200043bc nwords: 1
ssi_txuint8: TX: 0x200043b4->ff
ssi_rxuint8: RX: 0x200043bc<-ff
ssi_transfer: txbuffer: 0x200043cc rxbuffer: 0x200043d4 nwords: 1
ssi_txuint8: TX: 0x200043cc->40
ssi_rxuint8: RX: 0x200043d4<-ff
[... 457 lines cut ...]
ssi_transfer: txbuffer: 0x2000420c rxbuffer: 0x20004214 nwords: 1
ssi_txuint8: TX: 0x2000420c->ff
ssi_rxuint8: RX: 0x20004214<-ff
ssi_transfer: txbuffer: 0x2000420c rxbuffer: 0x20004214 nwords: 1
ssi_txuint8: TX: 0x2000420c->ff
ssi_rxuint8: RX: 0x20004214<-ff
ssi_transfer: txbuffer: 0x2000420c rxbuffer: 0x20004214 nwords: 1
ssi_txuint8: TX: 0x2000420c->ff
ssi_rxuint8: RX: 0x20004214<-ff
ssi_transfer: txbuffer: 0x2000420c rxbuffer: 0x20004214 nwords: 1
ssi_txuint8: TX: 0x2000420c->ff
ssi_rxuint8: RX: 0x20004214<-ff
ssi_transfer: txbuffer: 0x2000420c rxbuffer: 0x20004214 nwords: 1
ssi_txuint8: TX: 0x2000420c->ff
ssi_rxuint8: RX: 0x20004214<-ff
mmcsd_sendcmd: ERROR: Failed: i=9 response=ff
mmcsd_getcardinfo: ERROR: CMD9/10 failed: R1=ffffffff
tiva_ssiselect: devid: 65536 CS: de-assert
ssi_transfer: txbuffer: 0x2000425c rxbuffer: 0x20004264 nwords: 1
ssi_txuint8: TX: 0x2000425c->ff
ssi_rxuint8: RX: 0x20004264<-ff
mmcsd_geometry: ERROR: mmcsd_getcsd returned -5
nx_mount: ERROR: Bind method failed: -19
nsh: mount: mount failed: 19
nsh>

dishilov, 2024-06-15:
Could you check that the card is formatted Fat32 (xFat or other formats are not supported).

shijogeorge24, 2024-06-15:
Its fat32 microsd card

dishilov, 2024-06-15:
From the mailing list archive, can you try the following advice from Gregory: [mailing list response](https://lists.apache.org/thread/g83dnqs59r1vk46gnkfokrhxyzs4zdt0)

shijogeorge24, 2024-06-15:
So i changed the maximum SPI clock from 20MHZ to 4MHZ still the same
issue exist

dishilov, 2024-06-15:
The handshake is fine at 400khz, try changing to that.

Regards,
Dmitri Shilov.

-------- Original Message --------

shijogeorge24, 2024-06-15:
i checked that its still not working same error

acassis, 2024-06-16:
Shijo, if you can use same sdcard with a STM32 board with SDCard support, this way you confirm your setup is correct

shijogeorge24, 2024-06-16:
The only STM32 board I have with me is the STM32L4R5ZI-P. I don't see
support for that board on this page:
https://nuttx.apache.org/docs/latest/platforms/index.html. Is there any
support for this board?

acassis, 2024-06-17:
I think for STM32L4 family only the board nucleo-l496zg has support. It is better to use a generic STM32F4 board, like STM32F4Discovery

shijogeorge24, 2024-06-18:
i have tested with stm32f411re board 

nsh> mount -t vfat /dev/mmcsd0 /mnt
spi_setmode: mode=0
spi_setbits: nbits=8
spi_send: Sent: 00ff Return: 00ff Status: 02
nsh: mount: mount failed: 19

attaching my config file is there something missing in the config file ?

[configsd.txt](https://github.com/user-attachments/files/15883785/configsd.txt)

shijogeorge24, 2024-06-18:
https://github.com/shijogeorge24/stm32f411re/tree/main/nuttxspace/nuttx

here is the code i am using

acassis, 2024-06-19:
@shijogeorge24 did you try to format the card as FAT32 using NuttX mkfatfs?

https://acassis.wordpress.com/2023/11/25/how-to-format-sdcard-on-nuttx-terminal/

shijogeorge24, 2024-06-20:
The issue has been resolved. The problem was that the CS line was not configured at all 
https://github.com/apache/nuttx/blob/38664cee7853a7fa4334d64a063de5e68ecb0ecc/arch/arm/src/tiva/common/tiva_ssi.c#L1513 . I assumed that whenever we call the function tiva_ssibus_initialize(), it would take care of initializing all the necessary pins, but that was not the case for the CS pin, which was left for the user to configure.
At the moment, I have hardcoded the CS pin in the tiva_ssibus_initialize function. Is there a better way to handle this?

acassis, 2024-06-20:
Hi @shijogeorge24 the chip select should be configured at tm4c_ssidev_initialize() as it is done for MCP2515.

If you finally got everything working, please close this issue.

closed 2024-06-20 by shijogeorge24