apache/nuttx #12502: armv7-a elf file load / kernel build

xyluo555, 2024-06-15:
hello, I find that in the **kernel build mode** for **armv7-a** arch, the apps **(elf file)** contains **R_ARM_THM_JUMP24** relocation sections, take the **nsh** file for example, and by default CONFIG_ARM_THUMB is false. so while loading elf file, the libs>libc>machine>arm>armv7-a>arch_elf.c>**up_relocate** function will report an error “ERROR: Unsupported relocation: ...”.

parts of the relocation section info of nsh are like this:
Relocation section '.rel.text' at offset 0xd3a7c contains 2895 entries:
 Offset     Info    Type            Sym.Value  Sym. Name
00000014  0005ca1e **R_ARM_THM_JUMP24**  0000017d   __aeabi_ldiv0
00000020  00056d0a R_ARM_THM_CALL    00000071   __udivmoddi4
000003d8  0006231e R_ARM_THM_JUMP24  0000017d   __aeabi_idiv0
00000438  00001a2b R_ARM_MOVW_ABS_NC 000003f8   sig_trampoline
0000043c  00001a2c R_ARM_MOVT_ABS    000003f8   sig_trampoline

and the compiler option for nsh are like this:
CC:  nsh_consolemain.c 
**arm-none-eabi-gcc -c -Wstrict-prototypes -Wno-attributes -Wno-unknown-pragmas -Wno-psabi -fno-common -pipe  -fomit-frame-pointer -fno-common -Wall -Wshadow -Wundef -ffunction-sections -fdata-sections -g -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon-vfpv3 -mlittle-endian**    -isystem /home/cloud/nuttx-12.4.0/apps/import/include -isystem /home/cloud/nuttx-12.4.0/apps/import/include -D__NuttX__ -DNDEBUG  -I "/home/cloud/nuttx-12.4.0/apps/include"   nsh_consolemain.c -o  nsh_consolemain.c.home.cloud.nuttx-12.4.0.apps.nshlib.o

so the options like **-mthumb -mthumb-interwork** are not set. 

but why there are some THUMB type relocaiton sections? who can help explan this, thanks very much.
[defconfig.txt](https://github.com/user-attachments/files/15844983/defconfig.txt)

xyluo555, 2024-06-15:
well, it seems that the reason is the linker option. 

below is the linker options:
arm-none-eabi-ld -r -e main -T /home/cloud/nuttx-12.4.0/nuttx/binfmt/libelf/gnu-elf.ld -r -e _start -Bstatic -T/home/cloud/nuttx-12.4.0/apps/import/scripts/gnu-elf.ld -L/home/cloud/nuttx-12.4.0/apps/import/libs -L "/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/**thumb**/v7-a+simd/hard" -L "/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/**thumb**/v7-a+simd/hard" /home/cloud/nuttx-12.4.0/apps/import/startup/crt0.o elf_main.c.home.cloud.nuttx-12.4.0.apps.examples.elf.o --start-group -lmm -lc -lproxies -lxx -lm -lgcc /home/cloud/nuttx-12.4.0/apps/libapps.a /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/**thumb**/v7-a+simd/hard/libgcc.a --end-group -o /home/cloud/nuttx-12.4.0/apps/bin/elf

according to the apps/import/Make.defs Line:87,
the output for the command "arm-none-eabi-gcc -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon-vfpv3 -mlittle-endian --print-file-name=libgcc.a" is:

/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/**thumb**/v7-a+simd/hard/libgcc.a

so. how the output comes? anyone can help explain this? thanks a lot.

acassis, 2024-06-25:
Hi @xyluo555 did you get it working? Try to do "git grep thumb" or similar, it could help you to find some examples.
@xiaoxiang781216 @lupyuen any suggestion?

xyluo555, 2024-06-28:
Yes, the CONFIG_ARM_THUMB should be enabled. 
But I haven't figure out why ld select "/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/thumb/v7-a+simd/hard" as the library path.
actually, as is defined in apps/import/Make.def, while the command "arm-none-eabi-gcc -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon-vfpv3 -mlittle-endian --print-file-name=libgcc.a" is executed, the output is "/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/thumb/v7-a+simd/hard/libgcc.a".

acassis, 2024-06-28:
Yes, sometimes the linker does something very strange, it is not too smart.

BTW, did you get NuttX working on ZYNQ7000 ? If so, please consider submitting support to mainline ;-)

xyluo555, 2024-07-03:
sorry, the porting is on going. there seems to be a bug, as shown in issue #12618 .
currently, the console is ok and kernel & apps images are build separately. but there are more work to be finished.

ldube, 2024-07-04:
> /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/**thumb**/v7-a+simd/hard/libgcc.a
> 
> so. how the output comes? anyone can help explain this? thanks a lot.

@xyluo555 
I think gcc is configured to do this by default. For armv7-a, both ARM and THUMB programs will link to libraries in the thumb directory. See the output of this command, all the options seem to be mapped to thumb:
arm-none-eabi-gcc --print-multi-lib

Also see this [file](https://github.com/gcc-mirror/gcc/blob/master/gcc/config/arm/t-arm-elf) in the gcc tree.
Use "blame" to read the history of the file for more details.