apache/nuttx #12603: Duplicate symbol problem in nuttx

inochisa, 2024-07-01:
When enabling `CONFIG_DEBUG_LINK_WHOLE_ARCHIVE`, building nuttx with qemu-rv core always failed.

```
[0/1] Re-running _CMake..._
-- Configuring done (0.3s)
-- Generating done (0.2s)
-- Build files have been written to: /mnt/files/test/cache/mainline/nuttx/build
[1251/1252] Linking C executable nuttx
FAILED: nuttx
: && /mnt/files/test/toolchain/riscv-none-elf/bin/riscv-none-elf-gcc  -Wl,-melf64lriscv -Wl,--gc-sections -Wl,--whole-archive -nostdlib -Wl,--entry=__start -Wl,--cref -Wl,-Map=nuttx.map CMakeFiles/nuttx.dir/empty.c.obj -o nuttx  -Wl,--script=/mnt/files/test/cache/mainline/nuttx/build/ld.script.tmp  -Wl,--start-group  arch/libarch.a  binfmt/libbinfmt.a  drivers/libdrivers.a  fs/libfs.a  libs/libc/libc.a  mm/libmm.a  sched/libsched.a  boards/libboard.a  apps/libapps.a  apps/builtin/libapps_builtin.a  apps/system/nsh/libapps_nsh.a  apps/system/nsh/libapps_sh.a  apps/testing/getprime/libapps_getprime.a  apps/testing/ostest/libapps_ostest.a  apps/examples/hello/libapps_hello.a  /mnt/files/test/toolchain/riscv-none-elf/bin/../lib/gcc/riscv-none-elf/13.2.0/rv64imafdc_zicsr/lp64d/libgcc.a  -Wl,--end-group && :
/mnt/files/test/toolchain/riscv-none-elf/bin/../lib/gcc/riscv-none-elf/13.2.0/../../../../riscv-none-elf/bin/ld: arch/libarch.a(riscv_allocateheap.c.obj): in function `up_allocate_heap':
/mnt/files/test/cache/mainline/nuttx/arch/risc-v/src/common/riscv_allocateheap.c:70: multiple definition of `up_allocate_heap'; arch/libarch.a(qemu_rv_allocateheap.c.obj):/mnt/files/test/cache/mainline/nuttx/arch/risc-v/src/qemu-rv/qemu_rv_allocateheap.c:115: first defined here
/mnt/files/test/toolchain/riscv-none-elf/bin/../lib/gcc/riscv-none-elf/13.2.0/../../../../riscv-none-elf/bin/ld: warning: nuttx has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I have added `weak_function` to the definition of the function `up_allocate_heap` in riscv_allocateheap.c, but it will fail without enabling `CONFIG_DEBUG_LINK_WHOLE_ARCHIVE`. After googling, I found that ld only uses the first matched symbol in the archive. Is there any suggestion about how to fix this?

acassis, 2024-07-01:
@lupyuen any idea?

lupyuen, 2024-07-01:
@acassis Sorry I'm not sure how we might fix this

inochisa, 2024-07-05:
@lupyuen Is it possible to always enable "--whole-archive" for all library? As `DEBUG_OPT_UNUSED_SECTIONS` is enabled by default, it may have no change for the final binary.

lupyuen, 2024-07-05:
> Is it possible to always enable "--whole-archive" for all library? As `DEBUG_OPT_UNUSED_SECTIONS` is enabled by default, it may have no change for the final binary.

@xiaoxiang781216 and @tmedicci would you know about "--whole-archive" and `DEBUG_OPT_UNUSED_SECTIONS`? Sorry I'm not sure about this.

inochisa, 2024-07-05:
It seems that only using `--whole-archive"` for kernel library is better idea. I found an example in zephyr.

https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/linker/ld/target.cmake#L127-L143

xiaoxiang781216, 2024-07-05:
> > Is it possible to always enable "--whole-archive" for all library? As `DEBUG_OPT_UNUSED_SECTIONS` is enabled by default, it may have no change for the final binary.
> 
> @xiaoxiang781216 and @tmedicci would you know about "--whole-archive" and `DEBUG_OPT_UNUSED_SECTIONS`? Sorry I'm not sure about this.

 --whole-archive will pack all library into one before linking.

inochisa, 2024-07-05:
> --whole-archive will pack all library into one before linking.

The [man](https://www.man7.org/linux/man-pages/man1/ld.1.html) says it changes the symbol searching.

casaroli, 2026-11-08:
Fixed by #12726.

closed 2026-11-08 by casaroli