apache/nuttx #12625: What code editor to use with nuttx?

tommasoclini, 2024-07-04:
I've discovered this os just 2 weeks ago, I still have not understood how to get a code editor or ide working, what are your recommendations? Any tips to understand the bigger image?

acassis, 2024-07-05:
Hi @tommasoclini I think there are many options of code editor or IDE: many people uses VSCode, some people uses Eclipse and some year ago I tried use Code::Blocks, but it was full of bugs.

Currently I prefer to use VIM and compile in the terminal. I know it sounds crazy but I got this suggestion from an old embedded Linux book that I read many years ago: If you are a bricklayer and need to build a house on a rainy day, the IDE will be like an umbrella: you think it will protect you, but at the end of the day you will be completely wet and you won't build a single wall.

IDE are until until the moment that they start slow down and getting issues that you need to fix yourself

tommasoclini, 2024-07-05:
Yeah, I definitely understand, I have used IDEs in the past, but they are so slow and use so many resources.
The only alternatives I've found are VS Code and terminal based editors like Vim or Neovim, given the speed of VS Code if configured correctly and the possibility to use the mouse for easier in person collaboration and debugging, I've gone down that path.
I've found many amazing VS Code extensions.
I've developed some esp-idf and cubeclt based applications, and it has worked perfectly (I haven't had any problems with debugging).
But I have yet to figure out how to use VS Code for developing nuttx based applications, for example, some header files like cstdio or stdio.h are in the nuttx repository, instead VS Code takes me to the ones found with the tool chain I've installed, also header files with the same name, but different locations, VS Code always takes me to the wrong place.
Do you have any tips on how to make VS Code work with nuttx? Is there any kind of documentation or manual?

acassis, 2024-07-05:
@microhobby created an extension for VSCode to get better integration with NuttX:
https://www.youtube.com/watch?v=LgHVQSuywsY

tommasoclini, 2024-07-06:
I'm having trouble debugging an stm32 board(nucleo f446re), do you have the launch.json and the right commands for debugging that board? The breakpoints don't seem to be working, even though I can pause and resume

acassis, 2024-07-06:
I don´t use it, but maybe @JorgeGzm use and could help you. Are you using OpenOCD or J-Link? He added support for both.

JorgeGzm, 2024-07-07:
Hi @tommasoclini , I using the NuttX Helpers(MicroHobby) plugin for vscode and add new tasks

```
// launch.json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "J-Link",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/nuttx/nuttx",
            "request": "launch",
            "type": "cortex-debug",
            "device": "stm32h753bi",
            "interface": "swd",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "liveWatch": {
                "enabled": true,
                "samplesPerSecond": 4
            },
            "preLaunchTask": "Build",
        },
        {
            "name": "ST-Link",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/nuttx/nuttx",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "stutil",
            "serverArgs": [
                "--freq=24000k"
            ],
            "device": "stm32h753bi",
            "interface": "swd",
            "svdFile": "",
            "swoConfig": {}
        },
    ]
}
```
```
// settings.json
{
    "C_Cpp.errorSquiggles": "enabled",
    "C_Cpp.enhancedColorization": "enabled",
    "C_Cpp.dimInactiveRegions": true,
    "C_Cpp.inactiveRegionOpacity": 1,
    "C_Cpp.default.forcedInclude": [
        "${workspaceFolder}/nuttx/include/nuttx/config.h",
    ],
    "C_Cpp.inactiveRegionBackgroundColor": "#000000",//"#793d1c",
    "C_Cpp.inactiveRegionForegroundColor": "#FFFFFF",
    "ctags.languages": [
        "Kconfig",
        "Make"
    ],
    "cortex-debug.variableUseNaturalFormat": false,
    "cortex-debug.stlinkPath": "/usr/bin/st-utils",
    // "cortex-debug.gdbPath": "/usr/bin/arm-none-eabi-gdb",
    "cortex-debug.gdbPath": "gdb-multiarch",
    "nuttx.checkPatch": false,
    "nuttx.checkPatchPath": "${workspaceFolder}/nuttx/tools/checkpatch.sh",
    "nuttx.configureScriptPath": "${workspaceFolder}/nuttx/tools/configure.sh",
    "files.associations": {
    },
    "cmake.sourceDirectory": "/home/jaga/nuttxspace/apps",
    "restructuredtext.syntaxHighlighting.disabled": true,
    "cmake.configureOnOpen": true,
}
```

```
// task.json
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Configure",
            "command": "${workspaceFolder}/nuttx/tools/configure.sh",
            "type": "shell",
            "args": [
                "-l",
                "${command:nuttx.chooseFromConfigureList}"
            ],
            "problemMatcher": [],
            "presentation": {
                "reveal": "always"
            },
            "icon": {
                "id": "list-selection",
                "color": "terminal.ansiBlue"
            },
            "group": "build"
        },
        {
            "label": "Build",
            "type": "shell",
            "command": "make",
            "args": [
                "-j",
                "$(nproc)"
            ],
            "options": {
                "cwd": "${workspaceFolder}/nuttx"
            },
            "problemMatcher": [],
            "icon": {
                "id": "flame",
                "color": "terminal.ansiRed"
            },
            "group": "build"
        },
        {
            "label": "DistClean",
            "type": "shell",
            "command": "make",
            "args": [
                "distclean"
            ],
            "options": {
                "cwd": "${workspaceFolder}/nuttx"
            },
            "problemMatcher": [],
            "icon": {
                "id": "flame",
                "color": "terminal.ansiRed"
            },
            "group": "build"
        },
        {
            "label": "run-ctags",
            "type": "shell",
            "command": "echo",
            "args": [
                "${command:embeddedLinuxDev.regenerateCTags}"
            ],
            "problemMatcher": [],
            "icon": {
                "color": "terminal.ansiGreen",
                "id": "variable"
            },
            "runOptions": {
                "runOn": "folderOpen"
            }
        },
        {
            "label": "J-Link Flash",
            "command": "JLinkExe",
            "type": "shell",
            "args": [
                "-device",
                "STM32H753BI",
                "-if",
                "SWD",
                "-speed",
                "4000",
                "-commanderscript",
                "linum.jlink"
            ],
            "problemMatcher": [],
            "presentation": {
                "reveal": "always"
            },
            "icon": {
                "id": "list-selection",
                "color": "terminal.ansiBlue"
            },
            "group": "build"
        },
        {
            "label": "ST-Link Flash",
            "command": "st-flash",
            "type": "shell",
            "args": [
                "--format",
                "ihex", 
                "write",               
                "${workspaceFolder}/nuttx/nuttx.hex",                
            ],
            "problemMatcher": [],
            "presentation": {
                "reveal": "always"
            },
            "icon": {
                "id": "list-selection",
                "color": "terminal.ansiBlue"
            },
            "group": "build"
        },         
    ]
}

```

tommasoclini, 2024-07-07:
Thank you, I've just discovered this morning that I hadn't enabled debug symbols in my build configuration.
I've managed to debug it using the cppdbg feature of the c++ Microsoft extension, cortex debug extension, and the native debug extension.
Although the native debug didn't provide anything more than the built-in cppdbg, so I'll stick with the built-in and the cortex debug.
The cortex debug adds functionalities like global variables memory view, many other functionality, and it's much more responsive.

closed 2024-07-09 by tommasoclini