apache/nuttx #12629: getpid() not working

TimJTi, 2024-07-04:
My app uses getpid() to determine where to send a signal to - a thread interested in data changes from another thread "registers interest" and getpid() is used to obtain and save the pid for use when a change occurs.

NuttX 12.5 behaves as expected and returns the pid of the calling thread , but from 12.6 onwards it returns 0 every time.

Before I bisect, or look to my own code for an obscure bug, is there a change that's been made meaning we need to perhaps do something else first to get it to work, or there's been a fix/change I should know about?

xiaoxiang781216, 2024-07-05:
do you code run inside kernel or user space?

TimJTi, 2024-07-05:
@xiaoxiang781216 User space... it's configured as the start up app instead of nsh.

TimJTi, 2024-07-05:
@yf13 @xiaoxiang781216 
bisect shows [PR #12320](https://github.com/apache/nuttx/pull/12320) to be the cause. The PR conversation suggests it is breaking lots of things so should perhaps be reverted?
I am using release 12.4 until this is sorted.

yf13, 2024-07-05:
@TimJTi, I am unsure if the approach in [PR #12560](https://github.com/apache/nuttx/pull/12560) works for your case? or is there a config with rv-virt to reproduce your issue?

xiaoxiang781216, 2024-07-05:
> @yf13 @xiaoxiang781216 bisect shows [PR #12320](https://github.com/apache/nuttx/pull/12320) to be the cause. The PR conversation suggests it is breaking lots of things so should perhaps be reverted? I am using release 12.4 until this is sorted.

but the change only impacts the kernel thread, your case should be another issue.

TimJTi, 2024-07-05:
@xiaoxiang781216 but a bisect shows this commit to be the cause?

xiaoxiang781216, 2024-07-05:
could you simplify your case and attach it to pr?

TimJTi, 2024-07-05:
> could you simplify your case and attach it to pr?

@xiaoxiang781216  of course - it's here and I will copy to what I think is the offending PR. 

This is is one thread:
```
typedef struct
{
  pid_t   pid;
  uint8_t signo;
} notify_t;

static notify_t g_notify[CONFIG_MAX_NEWDATA_SIGS];

int newdata_notify(void *callback)
{
  int ret;
  int idx = 0;
  struct sigaction act;

  while ((idx < CONFIG_MAX_NEWDATA_SIGS) &&  g_notify[idx].pid)
    {
      idx++;
    }
  if (idx >= CONFIG_MAX_NEWDATA_SIGS)
    {
      ret = -EINVAL;
      goto errout;
    }

  g_notify[idx].pid      = getpid();     // <-------------------- This returns 0 since the PR commit
  g_notify[idx].signo    = CONFIG_NEWDATA_SIGNO;

  /* then code to set up the required sigaction and error handling */
  return ret;
}
```
This is in another thread:

`  ret = newdata_notify(callback_fn); `

acassis, 2024-07-05:
@xiaoxiang781216 maybe this test code could be added to ostest

yf13, 2024-07-05:
@TimTJi, thanks for the code fragment

Using `rv-virt/nsh` config, I added `getpid()` to the `printf()` line to `hello_main.c`, `getprime_main.c` for simple and threaded use cases and they both work fine when kicked off from NSH.

I also added `getpid()` usage to `nshlib/nsh_essoin.c` after the greeting message and see the non-zero pid dump on console as well.

So far I haven't found ways to reproduce your issue yet.

TimJTi, 2024-07-06:
I will do more investigation next week and:

1. Trace through the calls to getpid() to see if I can see the difference in behaviour
2. Rdd a cll to my function from a different thread (the App has about 6 on the go) in case there's some kind of thread create issue that is only revealed by this PR.
3. Re-run the bisect just in case I am blaming this PR unfairly!!

TimJTi, 2024-07-08:
@yf13 @xiaoxiang781216 
I have re-run the bisect and can confirm it determines this PR as the cause.
If I trace through the call to getpid() I note:
```
NuttX 12.5:    rtcb->pid = 2; rtcb->grouo->tg_pid = 2; The debugger does not associate rtcb with a global variable
NuttX Master:  rtcb->pid = 2; rtcb->group->tg_pid = 0; The debugger notes that rtcb is g_kthread_group
```
Don't know if that helps? There may be other differences but that was the difference that stood out to me. Let me know if you wany me to provide any specific debug data.

Is there some associated or new Kconfig setting I should be making now? Or a change to the way pthreads are created?

xiaoxiang781216, 2024-07-08:
could you print out rtcb->flags?

TimJTi, 2024-07-08:
> could you print out rtcb->flags? and how you create this task?

@xiaoxiang781216 of course. They are the same for both NuttX 12,5 and Master branches - 0x8208

yf13, 2024-07-08:
@TimJTi the `g_kthread_group` is the shared singleton group for all kernel threads, it is only used for kernel threads as shown in source of `group_inilialze()`.

Can you set a breakpoint in `group_initialize()` for your app task and share the backtrace? Hoping this can tell how the app task is created.

TimJTi, 2024-07-08:
I am pretty sure this is the correct backtrace (call stack):
group_initialize@0x20019554 (/home/tim/designs/nuttx/nuttx/sched/group/group_create.c:124)
```
group_initialize@0x20019554 (/home/tim/designs/nuttx/nuttx/sched/group/group_create.c:124)
nxtask_init@0x20016a90 (/home/tim/designs/nuttx/nuttx/sched/task/task_init.c:114)
nxtask_spawn_create@0x20017a70 (/home/tim/designs/nuttx/nuttx/sched/task/task_spawn.c:106)
nxtask_spawn_exec@0x20017be0 (/home/tim/designs/nuttx/nuttx/sched/task/task_spawn.c:222)
task_spawn@0x20017c8c (/home/tim/designs/nuttx/nuttx/sched/task/task_spawn.c:319)
nx_start_application@0x20012ebc (/home/tim/designs/nuttx/nuttx/sched/init/nx_bringup.c:348)
nx_start_task@0x20012f20 (/home/tim/designs/nuttx/nuttx/sched/init/nx_bringup.c:403)
nxtask_start@0x20017900 (/home/tim/designs/nuttx/nuttx/sched/task/task_start.c:109)
??@0x00000000 (Unknown Source:0)
```
In group_initialize it is not seen as a kernel thread, so:
`group = &tcb->group;`
and the group is set as 0.

I just tried changing the config to start nsh rather than my app directly: if I then start my app from nsh it behaves, with getpid() still returning a pid of 0 (i.e. it is not the pid number that's the issue but something else probably)

TimJTi, 2024-07-08:
FYI - my app behaviour is different when I call it from nsh so it could be there's some weird race condition or startup/init issue.
I am investigating.

yf13, 2024-07-08:
@TimJTi, thanks a lot!

Can you confirm if `group_postinitialize()` was reached and `nxtask_init()` returns OK? The `tg_pid` is populated within `group_postinialize()`.

yf13, 2024-07-08:
@TimJTi, thanks! Can you check the `group->tg_pid` value before leaving `nxtask_init()`? The tg_pid is populated in `group_postinialize()`.

TimJTi, 2024-07-09:
@yf13 - I will do, of course, but just sorting some issues with my app to be 100% sure this reported behaviour is not just a side-effect of a bug in my app (I'm thinking your change has revealed a problem that was always there). Please bear with me :-)

TimJTi, 2024-07-09:
@yf13 @xiaoxiang781216 
I apologise for wasting your time here. Although bisect correctly showed that it was the PR referenced here that was to "blame" the fault was in my App. I believe I had a race condition related to a FIFO being used between 2 threads, and the faster (?) startup that resulted from this PR revealed the issue.

Now I have reworked the code, it behaves well - and better overall - whether I run my app from nsh or as the startup app.

Thank you for helping me find a bug that would have come back to bite me one day in the future for sure!

acassis, 2024-07-09:
Hi @TimJTi I suggest you to create some blog post about it. If you faced this issue and wasted time with it, other people could have some issue. Think how you can help to avoid other people feeling the same pain.

TimJTi, 2024-07-09:
> Hi @TimJTi I suggest you to create some blog post about it. If you faced this issue and wasted time with it, other people could have some issue. Think how you can help to avoid other people feeling the same pain.

@acassis I would agree with you, in principle, but I think the bug I had was down to my inexperience as I learn how to write acceptable App code for NuttX!

I can't 100% be sure exactly what the bug was as I found quite a few issues as I switched between full and supressed optimization, and from running the app from nsh or as the startup app.

But my strong feeling is that the default setup while I am developing (supressed optimization), even running as the startup app, slowed down the creation of my threads enough for the code to "get away" with trying to read data from a queue and send signals, as the thread wanting the data was running - but the PR "blamed" here sped up the thread starting enough to reveal the problem. But it might have been something else LOL.

A code review from an experienced POSIX/LINUX developer would have found it I'm sure...but I am learning on the job here and on my own!!!! I should have stuck to bare metal.....sigh!!!!!!!!

yf13, 2024-07-09:
@TimJTi never mind. I am glad that you've solved it.