apache/nuttx #12665: SHT4X sensor driver

linguini1, 2024-07-11:
Hello,

I am relatively new contributor, and I am looking to add support for the SHT4X temperature and humidity sensor. I have a branch that is a work in progress here: https://github.com/linguini1/nuttx/tree/sht41-sensor

I have been able to accomplish most of the driver by looking at the SHT21 and SHT3X drivers made before, which are similar. Most of the i2c logic I have already tested previously on another platform, but I would like to now test on a NuttX build to make sure that all parts of the driver work correctly.

I am struggling to register the driver properly. I have found where the SHT3X was registered on an ESP32 board on the code base, as well as this article: https://www.embeddedrelated.com/showarticle/1668.php

I have tried adding the registration logic to the `rp2040_common_bringup()` but receive a linker error that my `sht4x_register` function cannot be found, despite the inclusion of the correct headers.

This also leads me to another question; is there a way to dynamically load these drivers, or must the drivers always be started in the board initialization logic by pre-configuring which sensor drivers to include?

davidgnx, 2024-07-11:
Hi @linguini1!

First of all: the code looks beautiful, understandable and nicely commented, thanks for this.
Also thank you for your motivation and welcome to the NuttX community!

I know that your fork is work-in-progress, but I tried to figure out what is wrong to help you.
During this "review" I have noticed a few things that you are probably not aware of...
These are just suggestions, but they may improve the usability of this driver. :)

1.) **Temperature range:**
  The sensor is capable of measuring temperatures between -40 C and +125 C with a resolution of 0.01 C and an accuracy of 0.1 C (SHT45).
  Your choice to store the results in an int16_t represented as mC or mF limits the theoretical measurement range to: -32.768 C ... +32.767 C in Celsius mode and to -35.98 C ... 0.43 C in Fahrenheit mode. This means that in Fahrenheit mode this driver will barely be able to measure any temperature above the freezing point of water (which is fairly suboptimal for many users).
  
2.) **sht4x_calc_temp():**
  There are massive integer overflows/underflows for huge ranges of valid input values (input values: raw measurement data points). This is related to 1.), see above.
 
3.) **Humidity range:**
  The sensor is capable of measuring relative humidity between 0 and 100 with a resolution of 0.01 %RH and an accuracy of 1 %RH (SHT45).
  You store the result of the measurement with a resolution of 1% RH, therefore you lose two digits precision. 
  Hint: The accuracy of the sensor is 1%, so losing two digits from the resolution is not necessarily a problem, it depends on the use case.
  
4.) **sht4x_calc_hum():**
  hum is an unsigned integer (uint16_t), therefore it can never be < 0. Those values, that are supposed to be < 0 will be mapped to 100 instead of 0 because of an integer overflow/underflow. So your clamped output will look like [ ..., 100, 100, 100, 0, 0, 1, 1, ..., 99, 99, 100, 100, 100, 100 ...] instead of [ ..., 0, 0, 0, 0, 1, 1, ..., 99, 99, 100, 100, 100, 100 ...] (note the wrongly mapped values below zero).
  
5.) **sht4x_register():**
  This may (or may not) answer your original question, I didn't verify, just looked at your code briefly. You declare the function in include/nuttx/sensors/sht4x.h, but you do not define it anywhere, or at least I haven't found it in the fork you've linked. There is a function named sht**21**_register() though (defined in drivers/sensors/sht4x.c), I guess it's a typo or copy-paste error.

If this was the problem, feel free to close this issue.

Greetings,
David

p.s.: **The second question (dynamically loading drivers) is not answered by this post, so feel free to comment, thank you!**

linguini1, 2024-07-11:
Thank you for the kind words and great suggestions @davidgnx!

1), 2), 3) You're correct, I didn't consider the integer precisions. I was porting this from another system where I had previously used floats, but wanted integers to be easier for less powerful hardware. Lots of good considerations that I will take into account, thank you!

5) You're very correct, that was not intended. Thanks for the catch!

6) It seems I did forget to rename the register function... never develop late at night I guess!

I will keep this issue open just a little longer to see if there is any information about dynamically registering drivers, if that's alright. Thanks for your help!

acassis, 2024-07-11:
Nice work @linguini1 !!!

Suggestion: run the checkpatch to catch the coding style issues:

```
$ ./tools/checkpatch.sh -f drivers/sensors/sht4x.c
```

You can find about our Coding style here: https://nuttx.apache.org/docs/latest/contributing/coding_style.html

If you decide to create a blog post talking about your first experience creating a driver for NuttX, please share it on our Reddit: https://www.reddit.com/r/nuttx/

linguini1, 2024-07-11:
Oh yes, I'm sure my work so far violates all the style rules. Saving my formatting until the end. I will be sure to make a post!

PR #12715 (2024-07-22): Add support for the Sensirion SHT4x temperature and humidity sensor
## Summary

This PR introduces a new driver for the SHT4x family of temperature and humidity sensors that operate over the I2C protocol. It provides a POSIX interface for all of the sensor's features: reading data, serial number, performing a reset and operating the sensor's heater within the safe usage limits.

This PR also closes #12665.

## Impact

An easier interface for future developers to interact with this family of sensors, rather than having to write their own driver.

## Testing

In order to test the driver, I perfo [... cut]
closed 2024-07-22 by acassis