apache/nuttx #12685: Add BLE support for esp32s3

tommasoclini, 2024-07-13:
This is a feature request, is this already planned? When are we going to get support for BLE on esp32 s3?

pkarashchenko, 2024-07-13:
Lats time I was using BLE on ESP32 S3 it was working. What issues do you experience with BLE on ESP32 S3?

pkarashchenko, 2024-07-13:
Have you tried https://github.com/apache/nuttx/tree/master/boards/xtensa/esp32s3/esp32s3-devkit/configs/blewifi ?

tommasoclini, 2024-07-13:
I've tried a few bt commands, but bt advertise start didn't work, ioctl(SISIOCBTADVSTART) returned 5, or something like that.
I've asked because in the documentation it says that it is not supported, correct me if I'm wrong.
Do I need to run other commands before trying bt advertise?
Did I do something wrong?
I was using the blewifi example configuration

pkarashchenko, 2024-07-13:
I think that maybe you are mixing NuttX BLE stack support with ESP32 S3 BLE. I was using ESP32 S3 BLE with nimble stack and the BLE driver was capable to communicating and my application successfully paired with the device (my ESP32 S3 based device was a host).
I do not have much knowledge in NuttX BLE stack, so can't help much.

acassis, 2024-07-13:
@tommasoclini unfortunately the native NuttX stack is very primitive. It is a BLE Stack that Intel released many year ago, it was used on zephyr as well, but there Nordic helped to improve it and added more features.

@pkarashchenko the blewifi board profile doesn't have NIMBLE example, could you please submit your board config example using NIMBLE?

tmedicci, 2024-08-23:
Just to check it once again, did you already use ESP32-S3 with nimBLE, right @pkarashchenko ?

Can you provide us with a simple defconfig example (or at least point which configs/changes were necessary) to close this issue?

maxikrie, 2024-10-03:
@pkarashchenko I would also be interested in a simple defconfig for ESP32 with nimble stack. Thanks!

pkarashchenko, 2024-10-03:
I will bake and publish some sample config early next week

maxikrie, 2024-10-12:
@pkarashchenko I am not sure if I missed it, did you publish it already? I assume you are employing the closed ble controller of espressif for this similar to softdevice of Nordic as in the nrf52840-dk:sdc_nimble example?

pkarashchenko, 2024-10-13:
@maxikrie you didn't. I had too many things flying around, so didn't managed to update on time. Here is my configuration
```
#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_ARCH_LEDS is not set
# CONFIG_NIMBLE_PORTING_EXAMPLE is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
# CONFIG_WIRELESS_BLUETOOTH_HOST is not set
CONFIG_ALLOW_BSD_COMPONENTS=y
CONFIG_ARCH="xtensa"
CONFIG_ARCH_BOARD="esp32s3-devkit"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32S3_DEVKIT=y
CONFIG_ARCH_CHIP="esp32s3"
CONFIG_ARCH_CHIP_ESP32S3=y
CONFIG_ARCH_CHIP_ESP32S3WROOM1=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_STACKDUMP=y
CONFIG_ARCH_XTENSA=y
CONFIG_BLUETOOTH_TXCMD_STACKSIZE=4096
CONFIG_BLUETOOTH_TXCONN_STACKSIZE=4096
CONFIG_BOARDCTL_RESET=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_DEBUG_ASSERTIONS=y
CONFIG_DEBUG_ERROR=y
CONFIG_DEBUG_FEATURES=y
CONFIG_DEBUG_INFO=y
CONFIG_DEBUG_WARN=y
CONFIG_DEFAULT_TASK_STACKSIZE=4096
CONFIG_DRIVERS_BLUETOOTH=y
CONFIG_DRIVERS_IEEE80211=y
CONFIG_DRIVERS_WIRELESS=y
CONFIG_ESP32S3_BLE=y
CONFIG_ESP32S3_RT_TIMER_TASK_STACK_SIZE=4096
CONFIG_ESP32S3_UART0=y
CONFIG_FS_LARGEFILE=y
CONFIG_FS_PROCFS=y
CONFIG_HAVE_CXX=y
CONFIG_HAVE_CXXINITIALIZE=y
CONFIG_HOST_MACOS=y
CONFIG_IDLETHREAD_STACKSIZE=3072
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INIT_STACKSIZE=8192
CONFIG_INTELHEX_BINARY=y
CONFIG_NAME_MAX=48
CONFIG_NDEBUG=y
CONFIG_NETDEV_LATEINIT=y
CONFIG_NET_BLUETOOTH=y
CONFIG_NET_ETH_PKTSIZE=1514
CONFIG_NET_SOCKOPTS=y
CONFIG_NET_TCP=y
CONFIG_NIMBLE=y
CONFIG_NIMBLE_BLE_SM_SC_ONLY=y
CONFIG_NIMBLE_BLE_WHITELIST=y
CONFIG_NIMBLE_DEBUG_ERROR=y
CONFIG_NIMBLE_DEBUG_INFO=y
CONFIG_NIMBLE_DEBUG_WARN=y
CONFIG_NIMBLE_HS_FLOW_CTRL=y
CONFIG_NIMBLE_ROLE_CENTRAL=y
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_LINELEN=64
[... 151 lines cut ...]
  rc = ble_hs_id_infer_auto(0, &g_own_addr_type);
  assert(rc == 0);

  scan();
}

void nimble_host_task(void *param)
{
  int rc;

  ble_hs_cfg.sync_cb = app_ble_sync_cb;

  /* Initialize data structures to track connected peers. */
  rc = peer_init(MYNEWT_VAL(BLE_MAX_CONNECTIONS), 64, 64, 64);
  assert(rc == 0);

  rc = ble_svc_gap_device_name_set(gap_name);
  assert(rc == 0);

  nimble_port_run();
}
```

I'm not sure if I can share more parts of the application here, but I hope that this should be enough for you to move forward.
Also I haven't tried my defconfig with the latest nuttx master, so I'm not sure of defconfig can be used "as is".

maxikrie, 2024-10-13:
@pkarashchenko Awesome, thanks a lot!

How does your scan function look like or is it similar to the the mynewt-nimble blecent example?

pkarashchenko, 2024-10-13:
@maxikrie I think I can share `scan`
```
static void
scan(void)
{
  int rc;

  /* set scan parameters:
      - scan interval in 0.625ms units
      - scan window in 0.625ms units
      - filter policy - 0 if whitelisting not used
      - limited - should limited discovery be used
      - passive - should passive scan be used
      - filter duplicates - 1 enables filtering duplicated advertisements */
  const struct ble_gap_disc_params scan_params = {0, 0, 0, 0, 0, 1};

  /* performs discovery procedure */
  rc = ble_gap_disc(g_own_addr_type, BLE_HS_FOREVER, &scan_params,
                    gap_event, NULL);
  assert(rc == 0);
}
```

tmedicci, 2024-10-28:
Is there anything else pending for this issue? 

@tommasoclini , can you close the issue, please?

maxikrie, 2024-10-30:
@acassis @pkarashchenko @tommasoclini 
I have created a pull request on the nuttx-apps repository, which adds nimble_bleprph and nimble_blecent working in companion. So far I have tested this only with two nrf52840-dk boards, but I will also try with the ESP32 in the coming days. These are the pull requests for apps (https://github.com/apache/nuttx-apps/pull/2807) and board configs (https://github.com/apache/nuttx/pull/14566)

tmedicci, 2024-10-31:
Thanks @maxikrie , let's us know when you are able to test on ESP32 devices and we can add this test on our internal CI.